silvanmelchior/RPi_Cam_Web_Interface #23: Resulting video missing part of the motion activity - implementing a circular buffer?

spikedrba, 2015-02-03:
Silvan,

thanks a ton for all your great work, this is fantastic.

One thing that is being a challenge however is the motion detection part. Basically when motion sends a ca 1 signal, if the object is moving fast enough through the frame, good chances are you'll miss it or miss at least half of the action. This is especially a problem with cars.

Discussing the issue on this thread:

http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=97734&p=680700#p680700

it seems that newer versions of raspvid support recording to a circular buffer, which would allow us to prepend a number of seconds to the final video capture. I've tried to look at this myself and see if I could port those changes to raspimjpeg, but it's unfortunately way over my head.

Any chance you had any planned work in this direction?

thanks,

Spike

silvanmelchior, 2015-02-11:
It sounds very interesting and I'd like to implement it if I find the time, but to be honest, this won't be in the next few months, because I have to finish other projects first, sorry :). Maybe someone else in the forum might try it, so why not asking there?

spikedrba, 2015-02-13:
@silvanmelchior thanks for being upfront about your commitment, I totally understand, I wish I could do more myself.

Yeah I've already asked on the forums, no takers, best I got was a note that this was implemented in raspivid in the new versions, but that's it. I tried to compare the two codebases to see if I could implement myself, but failed. The toolchain is also kind of an issue, every time I need to recompile the whole thing for a little test it takes me forever.

Maybe you could try to give me some input and I'll give it another go? but not sure if that in itself would cost you more time than you have.

Thanks

silvanmelchior, 2015-04-11:
Implemented

closed 2015-04-11 by silvanmelchior

spikedrba, 2015-04-11:
fantastic, thanks!

could you share a little about its defaults and/or how we can set how big the circular buffer is and how much will be prepended to the video? I checked https://github.com/silvanmelchior/userland/commits/master/host_applications/linux/apps/raspicam/RaspiMJPEG.c but I don't see any push to that branch and I couldn't find a new parameter in the config that says for example "append 5s", kinda like motion does it.

thanks again,

roberttidey, 2015-04-11:
When silvan and I merged things the main installer web code etc ends up in this repo. I pushed all my previous stuff back here.This just includes the raspimjpeg binary.

Previously the raspimjpeg was built in silvan's userland but I had forked a much more recent userland from Raspberry foundation in order to update annotation etc so it was easier to start from that one and silvan has put his circular buffer into there.

So installer, web code etc for most users is here. Userland for raspimjpeg source build is at
https://github.com/roberttidey/userland

There is a new entry /etc/raspimjpeg video_buffer that the new circular buffer code uses to control the buffer. This is nominally in mSec units but this is just an estimate as it depends on video compression. You can set it from the main web page under Camera Settings (Buffer)