silvanmelchior/RPi_Cam_Web_Interface #28: Opening the stream in OpenCV

lukehsiao, 2015-02-20:
Is there a way to open the video stream using a program like OpenCV?  For example, with MJPG-streamer, I could create a VideoCapture object in OpenCV, then open it using this code:

```
	//video capture object to acquire webcam feed
	const string videoStreamAddress = "http://192.168.1.126:8080/?action=stream?dummy=param.mjpg";
	VideoCapture capture;

	capture.open(videoStreamAddress);
```

Is there a way to access the feed in this way using RPi Cam Web Interface?

silvanmelchior, 2015-04-12:
Unfortunately I don't have any experience in OpenCV. Under <IP-of-RPI>/cam_pic.php is a jpg-image that gets refreshed every few miliseconds, maybe it's possible to read this file. However, it's not a correct MJPEG-Stream, just an updated JPEG.

You might consider asking in the forum: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=63276

closed 2015-04-12 by silvanmelchior

spikedrba, 2015-04-13:
@lukehsiao fwiw there is a push request for support "proper mjpeg streaming" which may solve your problem. See #49

lukehsiao, 2015-04-13:
Perfect. Thank you for your response and for all the work you've done on this great program!