silvanmelchior/RPi_Cam_Web_Interface #32: raspimjpeg configuration

StNNNx, 2015-03-31:
Hi,

I am developing a little project which uses the raspimjpeg stream.

I have some noobish doubts, here they go:

1) When I send a change on the settings (i.e. Contrast 50) it works fine, but I dont see the changes on the /etc/raspimjpeg file. Where are these changes going to?

2) I don't understand the PIPE thing. I can see that when you send a change it writes on the PIPE file, but how does it work? and where is the parser of those commands? 

3) I try to run the raspistill command when the raspimjpeg is active and it doesn't work. Do you know how could I fix it? Or another command to be able to take a snapshot while the raspimjpeg is running?

4) Which command does the "Record Image" button execute? Does it go through the PIPe also?

Thanks in advance,

StNNNx!

spikedrba, 2015-03-31:
Hi @StNNNx ,

what project are you working on?

to respond to your questions:

1) the changes goes to the running process via the pipe, in other words it's entirely temporary and won't survive a reboot or even a process restart

2) the PIPE like anything in unix is a file, something writes to it, something reads from it. In our case the web interface writes to it while the raspimjpeg process reads from it. Does that make sense? It's really just like a physical pipe with a reader and a writer at the two ends of it.

3) no more than one process can access the camera at the same time. This means that if you have raspimjpeg running you cannot use raspstill at the same time. The same would be true about raspivid and raspistill. In that sense there's nothing to fix, it's not broken. The solution is to use raspimjpeg itself, which has support to take stills, just send "im" through the pipe and it'll do the right thing

4) yes, see 3 and read the docs, all the pipe commands are listed in the raspimjpeg readme here https://github.com/silvanmelchior/userland/blob/master/host_applications/linux/apps/raspicam/README_RaspiMJPEG.md

take care,

Spike

StNNNx, 2015-03-31:
Thanks for the fast answer @spikedrba !

I'm trying to develope my own security system for my home. I just tested your stream and it works so much faster than the motion stream. I just want to make some changes (like e-mail advertisement and a mobile app to show the stream and change configs etc...) 

I guess that it's not a problem for you right?

Thanks for your answers, just a minute ago I read the RaspiMJPEG.md and it clarified lots of things to me! 

Thanks for your help

StNNNx!

spikedrba, 2015-03-31:
@StNNNx , it's not my project, it's @silvanmelchior 's project :) , just a user here.

Raspimjpeg is indeed several order of magnitudes faster than Motion, I was trying to setup home security just like you using that and it was useless. There's a motion-mmal patch that helps, but still find raspimjpeg far superior. The biggest issue right now is the lack of a circular buffer: if an object moves fast across the frame by the time motion triggers and the camera starts recording you will only catch the tail of it.

For the mobile app, there's already a few that are compatible with MJPEG streams, I'm using this on iOS:

https://itunes.apple.com/us/app/ip-cam-viewer-lite/id397322405?mt=8

it's also available on android:

https://play.google.com/store/apps/details?id=com.rcreations.ipcamviewer&hl=en

At to email notifcatoins (I guess that's what you meant with advertisement), do you mean upon motion? in which cases probably the best option is to modify motion's script to send your thing a signal as well (or even just have motion send that email).

hope that helps,

Spike

closed 2015-04-09 by silvanmelchior