silvanmelchior/RPi_Cam_Web_Interface #34: Password

silvanmelchior, 2015-04-09:
Idea: Protect settings or whole interface with a password

To do this, many other security holes needs to be fixed

spikedrba, 2015-04-09:
@silvanmelchior I was thinking about the same. Would you accept a simple patch that splits out the second "settings" part into its own page and protect that with an .htaccess? that would be pretty easy/quick to implement, happy to send a patch over.

roberttidey, 2015-04-09:
It depends a bit on use case here. Is password to hide / prevent simple access to settings to somebody who already has main access to the site via an overall password or is to provide very secure access to the settings system? The latter would need a bit more thought and a lot of the other support php files would need to be locked down. cmd_pipe.php, for example, allows one to do almost anything to the camera settings.

silvanmelchior, 2015-04-10:
It thought of some settings in the config file or any other place to set a password for the whole interface or just the settings.

As robert said, this would involve some more things to do:
- cmd_pipe needs to check privilegs
- all other files too
- download_file.php should be make secure
- warning for user to keep system up-to-date and change std-password should be added
- ...

LeonardoGentile, 2017-02-08:
any update on this?

roberttidey, 2017-02-08:
Check out this work by quavoce

https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=63276&p=1072602&hilit=RPi+Cam+password#p1072602