silvanmelchior/RPi_Cam_Web_Interface #36: Webserver nginx instead of Apache

blackjack4494, 2015-04-10:
Hey guys,
what is about using nginx instead of Apache. Because Apache is using significantly more ressources compared to nginx. It is a lightweight webserver with all features of Apache.

slabua, 2015-04-10:
You can have a look at my branch if you want https://github.com/slabua/RPi_Cam_Web_Interface/tree/nginx

blackjack4494, 2015-05-21:
okay. but will it be integrated to the official branch?

roberttidey, 2015-05-21:
I don't have a problem in principle with merging into official branch as the nginx was done as separate install verbs which doesn't affect those who continue to use Apache.

However, this version of the nginx was done a while ago and there have been quite a few changes to the standard Apache one. So it is not just a merge but bringing the nginx one up to date with latest one. I don't mind doing that at some point but I am working on other stuff. 

If somebody wants to update the nginx one to align it and make it merge friendly then that would get it done quicker.

roberttidey, 2015-06-01:
Added and updated in our installer but untested as yet.