silvanmelchior/RPi_Cam_Web_Interface #37: v2 Annotations

silvanmelchior, 2015-04-11:
Is there a special reason to support v2 and v3 annotations? Otherwise I would suggest to remove v2 or at least make v3 the standard

roberttidey, 2015-04-11:
Yes. I think we can probably remove v2 now. I had left it in initially as there were some v3 issues  and it relied on a updated  startx_elf but that is now all merged so that an rpi-update ensures all is OK.

It will need Annotate version removed from index.php (and labels losing v3) and raspimjpeg losing the anno_version config and always using cam_set_annotationV3

How do we want to do it? Shall I go ahead and make the changes?

roberttidey, 2015-04-11:
Changes made. Pushed to both repos

silvanmelchior, 2015-04-11:
Thanks, great!

closed 2015-04-11 by silvanmelchior