silvanmelchior/RPi_Cam_Web_Interface #38: Boxing

silvanmelchior, 2015-04-11:
Is there a reason to support both boxing inline and as a separate process? Otherwise I would suggest to remove the inline-boxing, your solution is way more comfortable.

roberttidey, 2015-04-11:
Main initial reason was to ensure that background mode wasn't breaking anything. However, it might be worth retaining at least for a bit. With the current background code it is possible to get multiple MPBox tasks launched if several video captures are close together. That in itself isn't a problem but I think MPBox uses a fair amount of memory and that can then get challenging on a Model A. 

I had thought an improved scheme was to add a cmd line queue into raspimjpeg for this so that the stop video would add the command to the queue and a launcher (inside main loop) could execute them one at a time. Most of the time it would operate like now, but it would then be safe to always run that way.

silvanmelchior, 2015-04-11:
Well, the perfect solution to this in my opinion would be a separate process with a queue as you suggested, but this process could box as many files as the ressources (cpu + ram) allow without making the interface slowly.

But to be honest, this isn't so necessairy, maybe your solution with the choose between inline and all at once is good enough.

closed 2015-04-15 by silvanmelchior