silvanmelchior/RPi_Cam_Web_Interface #39: Scheduler

silvanmelchior, 2015-04-11:
I just studied the scheduler, thanks for this impressive work! There are so many settings there, I love it! Just some questions:

With the default settings, motion detection gets startet as soon as my pi boots up, because of the period-start-command "md 0;em auto;md 1;" in the Day-Period. But I think if somebody installs the interface new on his pi, motion detection should not be activated until he wants so. I think a good default-setup would be:
- The scheduler starts automatically (as it is now)
- The scheduler doesn't start motion detection right at the beginning
- Motion Detection works everytime the user turns it on (now the scheduler ignores it if his not in the day-period).
So somethink like the all-day mode without a period-start command could be a good default-setting in my opinion. What do you think?

- Maybe is 30 seconds for the Max_Capture-value a bit to short, or what do you think? If somebody doesn't check this settings and just uses motion-detection "the simple way", then he might not record the whole motion because it stops after 30 seconds. Somethink like 0 (deactivate) here would be great i suppose. Or what is your suggestion?

roberttidey, 2015-04-11:
Yes. Those defaults could be improved for a clean install. They started off just being mine!

DayMode AllDay, Motion start ca 1 Motion stop ca 0 Period command just say an em auto in Day mode but no motion commands.

My thinking on including em auto is so I can put em night in other periods so that if / when somebody looks at expanding the periods then there is something there to start from. To show that multiple commands can be put in I could include an md 0 for dusk and an md 1 for dawn. They obviously have no effect unless one activates sun based settings.

Max video 0 is already no limit so I'll just change the default to that

silvanmelchior, 2015-04-12:
Yes this is great.

One detail: I think the "em auto" is not so good because this way the exposure-mode setting is lost everytime the Pi reboots and this could confuse the users. If you want the user to have something to start from, why not making two settings for Day Period Start in All Day mode and in Su based mode?

roberttidey, 2015-04-12:
Yes. I did think about that.

It's slightly tricky if I want to keep backwards compatibility with any pre-existing schedule.json as these are just arrays with the first 4 being Night, Dawn, Day, Dusk and the next six being the time ones.

I'll give it some more thought.

roberttidey, 2015-04-13:
Added an AllDay row separate from Day (Sun based). Clean defaults to just ca 1 ca 0. If old settings already present then copies the Day settings into AllDay to keep functionality the same.

silvanmelchior, 2015-04-13:
Great, thanks!

closed 2015-04-13 by silvanmelchior