silvanmelchior/RPi_Cam_Web_Interface #40: IFrame Logging error

roberttidey, 2015-04-11:
Somehow my raspimjpeg flipped into logging lots of IFrame errors

{2015/04/11 20:18:43} Error: Error in iframe list

About 1600 per second so raspberry slowed down and log got big very quickly

Stopped and started raspimjpeg and all OK again for the moment.

roberttidey, 2015-04-12:
I seem to be getting this fairly regularly. Will run for a while and then flips into iFrame error mode.

I have moved the logging outside the check loop just to try to slow the frequency down but obviously the root cause remains. Maybe we need to restart the buffering if we catch this type of error?

silvanmelchior, 2015-04-12:
Ok. thanks, I'm trying to reproduce the error and then find out the reason

silvanmelchior, 2015-04-12:
Sorry, I wasn't able to reproduce the bug, even after many changes in the buffer and many captures etc. Could you check wheter the error still occurs with the newest version and if so, explain me how to reproduce it?

Thanks

roberttidey, 2015-04-12:
I hadn't seen a particular trigger to start this yet. It would run for a while, do captures etc OK and then flip into log error mode. Moving error outside loop reduced it down to 20 / sec which made it easier.

I have put latest binary on, and will monitor it.

roberttidey, 2015-04-12:
Unfortunately still happened with latest binary. Lasted about 90 minutes and then flipped. No other activity logged around that time. Previous capture was 60 minutes earlier.

I had also experienced a capture where the video jumped back making it look like the iframe pointers are getting out of sync somehow and I wonder whether that is related. The video started with a timestamp near the end and then jumped back to the beginning and hesitated for a couple of seconds.

silvanmelchior, 2015-04-12:
Sounds not good, I'm trying to find the problem tomorrow

roberttidey, 2015-04-13:
Don't know if this may help. Started errors, can't see any direct cause. A capture request was then handled, but resulted in a bad video. The capture stop fixed the errors and next capture was OK. I think your latest code clearing the buffers on a stop could be be why it recovered. Perhaps detected errors should do this anyway.

[2015/04/13 02:54:28] Scheduled management tasks
[2015/04/13 03:08:08] New period detected 1
[2015/04/13 03:08:08] Send md 0
{2015/04/13 03:08:08} Motion detection stopped
[2015/04/13 03:08:10] Send em night
{2015/04/13 03:08:10} Change: exposure_mode = night
[2015/04/13 03:55:28] Scheduled management tasks
[2015/04/13 04:56:25] Scheduled management tasks
{2015/04/13 05:47:16} Error: Error in iframe list
{2015/04/13 05:47:16} Error: Error in iframe list

repeated about 50 per sec until

{2015/04/13 06:08:10} Error: Error in iframe list
[2015/04/13 06:08:10] Start capture requested
[2015/04/13 06:08:10] Send ca 1
{2015/04/13 06:08:10} Error: Error in iframe list
{2015/04/13 06:08:10} Error: Error in iframe list
{2015/04/13 06:08:10} Error: Error in iframe list
{2015/04/13 06:08:10} Error: Error in iframe list
{2015/04/13 06:08:10} Capturing started
[2015/04/13 06:08:16] Stop capture requested
[2015/04/13 06:08:16] Send ca 0     (BAD VIDEO)
{2015/04/13 06:08:16} Error: Error in iframe list
{2015/04/13 06:08:16} Error: Error in iframe list
{2015/04/13 06:08:16} Capturing stopped
{2015/04/13 06:08:16} Boxing in background
[2015/04/13 06:58:47] Scheduled management tasks
[2015/04/13 07:35:21] Start capture requested
[2015/04/13 07:35:21] Send ca 1
{2015/04/13 07:35:21} Capturing started
[2015/04/13 07:35:27] Stop capture requested
[2015/04/13 07:35:27] Send ca 0 (GOOD VIDEO)
{2015/04/13 07:35:27} Capturing stopped
{2015/04/13 07:35:27} Boxing in background

silvanmelchior, 2015-04-13:
Ok, this is a good hint. I'm still trying to reproduce the error, the interface is running for 2 hours now without logging the iframe-problem, pretty tricky to debug unfortunately.

roberttidey, 2015-04-13:
Left running and not tripped up today for 10 hours so far which obviously makes it tricky to find. However, I have had one capture with video that jumps around a bit so I think there is definitely somethingin the circular code that is not wrapping around correctly all the time and that might be associated with it.

For the suspect video the time stamps secs go 29,30,31,32,33,34,34,38,39,35,36,37,38,39,40,41,42

Where the double at 34 indicates a hesitation and a bit of video break up. The thumbnail has a timestamp of 34 showing this is where the video capture trigger started. So its a bit like a bit of the buffer gets added twice.

silvanmelchior, 2015-04-13:
I  discovered something new: The error is very likely to occur, if the buffer is small (1 or 2 seconds). The problem there is that the h264-stream doesn't contain enough iframes and so there is often a buffer without iframes in it. This is bad because without iframes are the other frames in the buffer useless. I made some changes in the source-code, now overwritten iframes are recognized, even it they are the last one. This way, the iframe-error should never happen again.
The question is now, why does this happen in large buffers? I read that raspivid emits an iframe every 2.4 seconds or 60 frames, so there should always be an iframe in it. I'm running a little test now to see if the same happens in a large buffer.

roberttidey, 2015-04-13:
OK That sounds promising. I had a buffer of 2000. It lasted 14 hours today but then flipped.

On a related topic I find that with buffer on the preview update behaviour with exposure compensation is different. With buffer off (as before) switching to night mode gives a light preview but captured video was dark with the 25fps frame rate. This was as before and something to do with frame rate overriding the em mode when video was activated. With buffering on the video is activated even outside of captures so now the preview goes dark and doesn't benefit from em night. 

I'm not sure there is much that can be done about that unless we work out how to set up effective exposure modes when capturing video.

I can of course set buffering to 0 in scheduler when using night mode and back to say 3000 in day mode.

spikedrba, 2015-04-13:
@roberttidey fwiw to me that's better (with the buffer and darker preview). Maybe I'm biased because I'm working on a security camera project, but having the preview bright and the recording dark is just flat out misleading, I think consistency is critical here.

Regarding effective exposure modes, that's what my other issue is about  #44 , I found that setting to night mode isn't in itself great, adjusting brightness and contrast helps a lot to squeeze out a few extra fps, not to mention that with an IR light that I'm expecting most people doing sec cam to have, you can still push at least 10fps.

roberttidey, 2015-04-13:
Yes. I don't want inconsistency either. My ideal is that the exposure mode, for example, has the same effectiveness in video recording with the automatic selection of appropriate parameters. The night mode for me extends good viewability in preview a lot and I can use time lapse images to give equivalence, but obviously nicer to get comparable videos.

silvanmelchior, 2015-04-14:
Did you ever get an iframe-error with a buffer of 3 seconds or more? I suppose your test with 2 seconds was good at first because even if there is only every 2.4 seconds an iframe, the buffer is normally large enough to hold at least one of them. (Because the 2 seconds are in the regular case more than 2 seconds).
But then I guess the bitrate of the video increased slightly (morr movement/more colors) and this lead to the iframe error.

silvanmelchior, 2015-04-14:
Just found this in the MMAL-documentation:

MMAL_PARAMETER_VIDEO_REQUEST_I_FRAME

This might be a solution to support smaller buffers

roberttidey, 2015-04-14:
That sounds interesting. I don't see much harm in running permanently with a smaller iframe period or making it dependent on buffer size. I guess we could enforce a minimum non zero buffer size as well.

I'm now running with buffer 3500 to see how that goes.

spikedrba, 2015-04-14:
@roberttidey if I understand it correctly, depending on the application, high number of i-frames could be a concern in relation to file size. I'm not sure how much it would grow, but I believe I read an i-frame is 2x a P-frame, so the impact could be noticeable depending how small you make that. It would be great to have it as a config option with a sane default tho, maybe as you suggested that can be dependent on buffer size.

silvanmelchior, 2015-04-14:
I'm running some tests as soon as I find out how to change the frequency of
iframes. But I think if I force additional iframes just in the buffer if
there is not enought space and stop this as soon as the capture starts,
there shouldn't be a problem with the filesize.

spikedrba, 2015-04-14:
fwiw I've just got an iframe error running with 3s buffer (had many with smaller buffers and at 3s only one line got logged, not the usual flood).

roberttidey, 2015-04-14:
So far been running for 11 hours at 3500 buffer with no iframe errors. Will keep it at this value for the moment.

silvanmelchior, 2015-04-14:
So I'm going to implement the following tomorrow: The buffer counts how
long it's been since the last iframe and if this is more than 25% of the
buffer-site, it requests a new one. This way, there is:
- always at least 75% of the buffer usable
- nearly the same size for the video, just the first few seconds might need
slighlty more space

Or does anyone see a better solution?

roberttidey, 2015-04-14:
Sounds good to me. For a lot of uses the buffer seems to hold significantly more than calculated which I guess is because with motion trigger it is very static for quite a bit of lead-in.

silvanmelchior, 2015-04-15:
I uploaded a new version of raspimjpeg, it should now be possible to use small buffers (>=1000, but I made a short test with 500 and it worked too).

I made the following changes:
- raspimjpeg calculates the intraperiod for a new buffer and changes it. While recording, the intraperiod gets changed back, so the resulting video is just a tiny bit larger
- Additionally, a counter checks if there are enough iframes in the buffer. Otherwise, he requests a new one.

So I didn't do a long-time test yet, but I hope this solves the problem

roberttidey, 2015-04-15:
Great. I have changed down to 2000 and will see how that goes long term.

spikedrba, 2015-04-17:
been testing on 2000 and just got a bunch of iframe errors again. seems the problem is still there.

spikedrba, 2015-04-23:
just fyi this is still happening, got a bunch more last night, I don't see a pattern in when they happen right now. Buffer is set ot 2K. This is a breakdown of error count throughout the day per hour of the day:
```
 131530 00
 100853 01
  92582 02
  92806 03
  93010 04
  93152 05
 112563 06
  92382 07
  92322 08
  92234 09
  91677 10
  92715 11
  93028 12
  93148 13
  72091 14
  37037 15
  37122 16
  11937 17
  19498 21
 146387 22
 145475 23
```

silvanmelchior, 2015-04-23:
You got about 10 errors per second? Does anybody else get so many errors?

spikedrba, 2015-04-23:
@silvanmelchior sorry,no, those are aggregated over a few days of tests, just meant to show the lack of a pattern in terms of conditions. This is a breakdown including the day:

```
  37706 2015/04/20 22
  36872 2015/04/20 23
  36862 2015/04/21 00
  36810 2015/04/21 01
  36808 2015/04/21 02
  36790 2015/04/21 03
  36760 2015/04/21 04
  37053 2015/04/21 05
  42818 2015/04/21 06
  36844 2015/04/21 07
  36856 2015/04/21 08
  36906 2015/04/21 09
  36972 2015/04/21 10
  37440 2015/04/21 11
  37712 2015/04/21 12
  37798 2015/04/21 13
  37369 2015/04/21 14
  37037 2015/04/21 15
  37122 2015/04/21 16
  11937 2015/04/21 17
  11929 2015/04/21 21
 108681 2015/04/21 22
 108603 2015/04/21 23
  94668 2015/04/22 00
  64043 2015/04/22 01
  55774 2015/04/22 02
  56016 2015/04/22 03
  56250 2015/04/22 04
  55185 2015/04/22 05
  66606 2015/04/22 06
  55538 2015/04/22 07
  55466 2015/04/22 08
  55328 2015/04/22 09
  54704 2015/04/22 10
  55275 2015/04/22 11
  55316 2015/04/22 12
  55350 2015/04/22 13
  34722 2015/04/22 14
    914 2015/04/23 05
   1918 2015/04/23 06
```

that said, highest count is still way higher than 10/s. Highest report is * 108681 2015/04/21 22*, so that's 108681 lines of *{2015/04/21 22:xx:xx} Error: Error in iframe list* during 9pm on the 21st. 108681/60*60 = ~30.

This is the bash line I'm using to generate that report just in case you wanna test it or double check it:
```
grep "Error in iframe" rpicam-2537-schedule.log | sed 's/{\([^:]*\):.*}.*/\1/' | sort | uniq -c
```
finds all the lines with errors, extracts the dates between {} grabbing the time up to the first : (so just gets the hour) and then sorts them before doing a count using uniq

silvanmelchior, 2015-04-23:
Sorry, I still don't really understand, so you get the error every hour
between 10'000-100'000 times? Are you using the version where the
intraperiod is adjusted?

spikedrba, 2015-04-23:
apologies if I'm not being clear. I've been getting errors almost every hour, but not all (you can see gaps in the afternoon yesterday for example all the way until this morning at 5am.

That said, I thought I had upgraded all cameras to latest, but this particular one was up to date to the 14th and changelog seems to suggest the dynamic frame adjustment was made on the 15th. Apologies for the noise, I've upgraded and cleared out the log, will give it a day or two of testing with buffer set to 2K and report back.

thanks,