silvanmelchior/RPi_Cam_Web_Interface #44: settings lock paired with scheduler

spikedrba, 2015-04-12:
Hi,

so excited about the recent merge and the whole new host of features, thank you so much @roberttidey.

One thing I was thinking about is locking stuff like awb/exposure because I'm noticing substantial fluctuations in color and light with the pinoir on my security camera prototype if there's any movement in the scene.

Playing with the python picamera for faster prototyping I seem to be getting decent results by settings and releasing awb and exposure_mode at short intervals to let it adjust to the new light conditions before locking it again (computing new settings manually doesn't seem feasible/accurate).

Do you think this makes sense and could be included in RPI_CWI?

thanks,

Spike

roberttidey, 2015-04-12:
Can you clarify a couple of things on this? When you say set and release do you mean alternate between auto and some other mode? Plus what sort of time interval are we talking about?

spikedrba, 2015-04-13:
@roberttidey I mean auto Vs manual settings. So basically set awb_mode to auto for example and give it a few ms to level out the gains, save them and then set it to off and assigned the saved gains. Same for exposure_mode and exposure_speed. Repeat this process maybe ever 5-10 mins, considering light might change quite a bit over that time period.

This is an example with picamera: http://picamera.readthedocs.org/en/latest/recipes1.html#capturing-consistent-images

lemme know if that makes enough sense now. Thanks.

roberttidey, 2015-04-13:
You could sort of do this today using schedule management_command, at least as a test to check feasibility.
1. Set management_interval to 600 for 10 minute check
2. Write script to echo -n 'wb auto' >/var/www/FIFO sleep for say 1 sec then echo -n 'wb off' >/var/www/FIFO
3. Make script executable
4. Set management_command to execute script

spikedrba, 2015-04-13:
not sure that will work, maybe raspimjpeg is different, but as far as I understand once you set awb_mode to off you're on your own, meaning you need to set the various gains yourself and I've no idea how to go about calculating those myself so I'd have to read them out after sleeping for 1sec, but I don't know that that is possible through the pipe, is it?