silvanmelchior/RPi_Cam_Web_Interface #46: Preview

silvanmelchior, 2015-04-12:
I'm testing your new implemented preview-features and I think they're great, thanks. Just some things:

- The zip-donwload doesn't seem to work for me: If I select a few files ans press download zip, there is just an empy rectangle appearing, does it work for you or did it break somewhere and nobody noticed?

- I think there should be a security-question for the delete, delete-sel and delete all button. Especially the delete-all button is very dangerous right now. The question could easily be implemented with a bootstrap-modal: http://getbootstrap.com/javascript/#modals . What do you think?

- The conversion from a timelapse into a video doesn't work for me. If I press the button, I get a new picture, labeled as a video, and if I download this new picture, there is still just a picture I get. Is this the same for you with the actual version?

silvanmelchior, 2015-04-12:
Update: the theme-change didn't work because of the cache in my browser, so this isn't a big problem to fix

roberttidey, 2015-04-12:
Zip is still working for me. The rectangle is a progress box which steps along for each zip added but doesn't update until after the first one is added so if zipping large videos that can take a while which can be confusing. I could change that so that it updates just prior to each one but I'd need to ge the last one showing a bit less than 100% else that would be confusing.. The download after it completes relies on it still being on the preview when it completes so navigating away breaks this. . I find a Pi 1 takes about 0.7 sec per 1MB which is pretty slow using the standard zip package. Try taking 3 or 4 image snaps and zipping them. The styling of the progress is not great. I did spend a bit of time trying to improve that but got tangled in different browser behaviours.

Yes. A confirm on at least Delete All would be good. Maybe on select as well. Not so sure about individual deletes as that can get irritating and its less dangerous.

Conversion on timelapse does work for me but it is painfully slow particularly on a Pi 1. It creates a video icon straight away but that is just the thumbnail and the production of the video should then be going on in the background with the video icon showing Busy. When it completes then a preview refresh shows a completed video present. If normal high res images are used I am finding about 5 seconds per image which can be improved quite a bit by capturing lower res images as source.

Apart from any ideas on fundamentally speeding it up it may be possible to launch it via a script and get some sort of progress indicator back on to the icon.

I have noticed the style selector (day/night) can be a bit problematic now and is fighting the cache somehow. It was working totally reliably when I first added it so something has changed and I was planning to check it out.

roberttidey, 2015-04-13:
Changed the way selected extrastyle is fed back to avoid cacheing issues.

roberttidey, 2015-04-14:
Added just simple javascript confirm to Delete Sel and Delete All for the moment until something fancier can be done.

silvanmelchior, 2015-04-15:
Ok, I think this is enough at the moment, thanks

closed 2015-04-15 by silvanmelchior