silvanmelchior/RPi_Cam_Web_Interface #47: pipan changes

roberttidey, 2015-04-12:
Not a pipan user myself (although just been given one!) but just put the original stuff in to avoid update issues.
Saw the further update and I'm not sure I understand where init_pt is used now. That was apparently needed to avoid re-centring on page refreshes. Should it be an option?

silvanmelchior, 2015-04-13:
To be honest, I just removed the command because it gave me a php-error and then I forgot to add it again, my fault. I'll undo it as soon as I resolve the problem with the circular buffer.

silvanmelchior, 2015-04-15:
Sorry, I don't get it: why do you try to read from the FIFO-pipe in the php-script? As long as nobody writes into the pipe, there shouldn't be the possibility to get the current position from it, or do I understand your code wrong?

Thanks

roberttidey, 2015-04-15:
I just copied th eoriginal mod to preserve the pan / tilt as originally posted and I haven't dug into this yet but pipan.php seems to be writing stuff into that FIFO but as far as I can see this is just internal and doesn't involve the python side. 

It could be that there is a better way than using a FIFO for this part as it's seems to be just a way of remembering the last values. I don't think I would have done it that way myself.

silvanmelchior, 2015-04-15:
Ah ok, then somebody changed my code in the wiki and I didn't notice, I was a bit confused and thought, maybe you added the new lines.

I modified the php, it writes now the current position into a file to remember them.

closed 2015-04-15 by silvanmelchior