silvanmelchior/RPi_Cam_Web_Interface #48: Feature Request: Annotations containing GPS location data from gpsd

WebMaka, 2015-04-13:
I have a feature to request, which will make RPCWI great for things like car dashcams:

I'd love to be able to annotate videos with wildcards for latitude, longitude, speed, heading, and altitude based on GPS readings taken by gpsd.

I'll experiment with adding these myself, but since I only just started tinkering with this suite, someone with greater familiarity with the codebase will likely be able to whip up a bridge for GPS location data before I could.

roberttidey, 2015-04-13:
Two things.
One can change the annotation template dynamically today using the 'an' command so an externally program could change that string to whatever it wanted at regular intervals or when a change has occurred.

Currently the annotation string has variables for date / time / indexes. We could contemplate a user string like %u that could be included in the template. This could then be changed by a different command. Achieves a similar effect  but without having to worry about the overall template.

WebMaka, 2015-04-14:
Thanks for the 'an' command tip - that quickly led me to discover the documentation's mention of echoing commands into /var/www/FIFO. A quick Googling turned up a python script that reads GPS info from gpsd and displays it, and a little quick hacking on my part made said script dynamically change the annotation into a location/speed readout.

roberttidey, 2015-04-14:
I think the 31 character limit might have been part of v2 annotation. We now use v3, not sure what the limit is at the moment but I've just done a 48 character one with no problem.

We will update the docs / page accordingly.

WebMaka, 2015-04-14:
Interesting. I was feeding roughly 70 characters into the annotation but only having about 45 or so characters displayed and seeing a question mark at the end, so I am wondering if there's a unicode issue afoot. I'll have to mess about with this further.

At any rate, I worked around the situation by simply having the script toggle showing coordinates and speed while keeping the date and time visible. Would prefer to have all of the data on all of the time, but this works in the interim.

roberttidey, 2015-04-14:
Yes. I tried various sizes and behaviour above 60 looks suspicious, sometimes you get a ?, sometimes it disappears altogether. I got the 255 from the MMAL structure which allows the interface string to be up to that and there is no limit or processing being applied by the front end software. So it looks like there may be some other internal limits in the camera drivers.

closed 2015-04-15 by roberttidey