silvanmelchior/RPi_Cam_Web_Interface #52: Annotation & night record

sh2515, 2015-04-14:
I have just installed the latest version and I have a couple of questions:
1) the text on the videos just has a question mark even with RPi Cam %Y.%M.%D_%h:%m:%s as the setting
2) is there a way to automatically have night fps and day fps to keep the night preview settings when recording, as they are crystal clear. However they change back to standard fps when recording.

roberttidey, 2015-04-15:
1) Haven't seen that annotation problem before unless string got longer than 60 chars. Have you pressed Default to make sure it is clean?

2) You can put any combination of commands in the scheduler Period Start settings. So px allows you to change resolutions / fps at different times. E.g. px 1296 976 2 2 2592 1944 would set a 2fps for recording 1296x976 video. Likewise you can restore it back to 25fps with similar command.

sh2515, 2015-04-15:
Hi
1) I set it as default and it didn't work. Then I did reset settings and no text at all came up. Then did a default and still no more text is on.  The log on the webpage showed me what I had done and no errors.  Is there another log I can check?

2) brilliant will do that.

Thanks

roberttidey, 2015-04-15:
The scheduler log (which you can view from schedule screen) includes the output from raspimjpeg when it changes settings. So when you change it you should see something like 
Change: annotation = Rob Cam %Y.%M.%D_%h:%m:%s

These changes from defaults also get saved in /var/www/uconfig which you can open with text editor (nano). If that file has got something wacky in it then that might cause a problem. You can just delete it (or Reset Settings from Web) to revert to default settings.