silvanmelchior/RPi_Cam_Web_Interface #54: Firefox can't change settings properly

sh2515, 2015-04-15:
Hi
In firefox I am unable to change settings as the delete buttons don't delete previous settings.  In Chrome they do work and I can successfully change the settings.

sh2515, 2015-04-15:
I have just put Firefox into debugger mode to see if I can find the fault, at this point it then starts working.  Keep this open at least till friday and i will try again.

roberttidey, 2015-04-15:
Have you updated recently as we put in a style change that had been preventing entry of new settings in some browsers (Safari / Firefox).

sh2515, 2015-04-16:
I am using firefox 37.0.1.  I have had a look at the console and found the following and a slow refresh.
```
Unknown pseudo-class or pseudo-element '-webkit-inner-spin-button'.  Ruleset ignored due to bad selector. style_minified.css:1:1443
Unknown pseudo-class or pseudo-element '-webkit-search-cancel-button'.  Ruleset ignored due to bad selector. style_minified.css:1:1683
Unknown property '-moz-osx-font-smoothing'.  Declaration dropped. style_minified.css:1:3453
Expected colour but found 'auto'.  Expected colour but found '-webkit-focus-ring-color'.  Expected end of value but found '-webkit-focus-ring-color'.  Error in parsing value for 'outline'.  Declaration dropped. style_minified.css:1:12731
Expected end of value but found '\9 '.  Error in parsing value for 'width'.  Declaration dropped. style_minified.css:1:12950
Expected end of value but found '\9 '.  Error in parsing value for 'width'.  Declaration dropped. style_minified.css:1:13259
Expected end of value but found '\9 '.  Error in parsing value for 'margin-top'.  Declaration dropped. style_minified.css:1:32670
Expected colour but found 'auto'.  Expected colour but found '-webkit-focus-ring-color'.  Expected end of value but found '-webkit-focus-ring-color'.  Error in parsing value for 'outline'.  Declaration dropped. style_minified.css:1:32924
Unknown pseudo-class or pseudo-element '-ms-input-placeholder'.  Ruleset ignored due to bad selector. style_minified.css:1:33862
Unknown pseudo-class or pseudo-element '-webkit-input-placeholder'.  Ruleset ignored due to bad selector. style_minified.css:1:33910
Expected end of value but found '�'.  Error in parsing value for 'line-height'.  Declaration dropped. style_minified.css:1:34285
Expected end of value but found '\9 '.  Error in parsing value for 'margin-top'.  Declaration dropped. style_minified.css:1:34981
Unknown property 'user-select'.  Declaration dropped. style_minified.css:1:40658
Expected colour but found 'auto'.  Expected colour but found '-webkit-focus-ring-color'.  Expected end of value but found '-webkit-focus-ring-color'.  Error in parsing value for 'outline'.  Declaration dropped. style_minified.css:1:40743
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:41112
Expected 'none', URL, or filter function but found 'progid'.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:49301
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:53741
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:77298
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:77494
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:78406
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:78602
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:79062
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:79258
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:79724
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:79920
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:80384
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:80580
Unknown property 'zoom'.  Declaration dropped. style_minified.css:1:80981
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:95428
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:95537
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:96672
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:96726
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:97637
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:97684
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:101624
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:101764
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:101855
Expected 'none', URL, or filter function but found 'progid'.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:102045
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:102215
Error in parsing value for 'background-image'.  Declaration dropped. style_minified.css:1:102306
Expected 'none', URL, or filter function but found 'progid'.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:102496
[... cut ...]
Expected colour but found 'auto'.  Expected colour but found '-webkit-focus-ring-color'.  Expected end of value but found '-webkit-focus-ring-color'.  Error in parsing value for 'outline'.  Declaration dropped. style_minified.css:1:109781
Expected 'none', URL, or filter function but found 'alpha('.  Error in parsing value for 'filter'.  Declaration dropped. style_minified.css:1:110262
The character encoding of the HTML document was not declared. The document will render with garbled text in some browser configurations if the document contains characters from outside the US-ASCII range. The character encoding of the page must be declared in the document or in the transfer protocol.
```

roberttidey, 2015-04-16:
I've tested with Firefox 37.01 with latest Rpi Cam. I do get all those style errors which is due to generic style sheet. but it works fine. You have to use Default streaming and my settings updateok now I have taken out a specific mozilla style which was blocking text input.

closed 2015-05-02 by roberttidey