silvanmelchior/RPi_Cam_Web_Interface #55: Documentation

silvanmelchior, 2015-04-15:
I think there's a bit of a mess in the documentation. We have:
- userland README-RaspiMJPEG
- userland README-RaspiMJPEG-RJT
- RPiCam.pdf
- Wiki
- Help-Text in scheduler

I think it would be good to merge all this documents into one, big documentation. If we use the wiki for example, anybody could help to complete it. What do you think?

If we took the wiki, I'd suggest the following structure:

1 Installation Instructions
1.1 Basic Installation
1.2 Startup Behaviour
1.3 Links
1.4 Special thanks

2 Usage
2.1 Basic usage (Preview, Settings, Catpures, Download)
2.2 Scheduler
2.3 Motion

3 Configuration
3.1 Configuration Scheme
3.2 Naming Scheme

4 Additions
4.1 Pipan
4.2 ...

5 Architecture
5.1 Data Flow
5.2 Control Flow
5.3 RaspiMJPEG
5.3.1 Components
5.3.2 Pipe

roberttidey, 2015-04-15:
OK I'd be happy to move stuff in there. 
How do you want to do this? Maybe you change the structure and I move existing material as a first step?

silvanmelchior, 2015-04-16:
I changed the structure, so you could add the existing parts, afterwards we can ask in the forum for additional help

silvanmelchior, 2015-04-16:
I wait with the closing until we removed the existing documentation, so we won't forget it

roberttidey, 2015-04-16:
Made a start merging some of the PDF stuff in. Will do some more later.

silvanmelchior, 2015-04-16:
Looks great, thanks a lot!

closed 2015-05-02 by roberttidey