silvanmelchior/RPi_Cam_Web_Interface #56: Break away from Motion?

spikedrba, 2015-04-15:
Hi,

provided that I don't fully understand the details and therefore I might be misunderstanding the complexity of doing this - but wouldn't it make sense to bake motion detection inside raspimjpeg?

While I'm grateful to have motion detection and for all the effort put into the Motion software, it seems to me not to be very effective at its job: based on my tests it takes a lot of cpu to do motion detection at really low framerate (2fps) due to its implementation relying on the CPU. There's a -mmal branch that performs better, but is not mainstream and still doesn't seem to be able to push more than 10fps (which is plenty for most applications, mind you).

Looking at picamera and some python implementations, it seems that it's possible to make this work relatively easily by using the motion vectors provided by the h264 encoder. Here's a walkthrough of the process:

http://bits.citrusbyte.com/motion-detection-with-raspberry-pi/ (look at Method 2, mid page)

Here's an original article from the foundation talking about the motion vectors:
https://www.raspberrypi.org/vectors-from-coarse-motion-estimation/

I'm still pretty much a noob around C++ and MMAL and can't quite send a patch, but thought I'd bring this up anyway for consideration.

thanks,

Spike

PS: could you guys share any quick way to change code/compile/test on a pi? recompiling the userland each time on the pi is really time consuming. Any article you could point me to to make the testing cycle faster would be great for me to try things out.

spikedrba, 2015-04-15:
and just to comment on the benefits/pov: this would save a lot of CPU making it possible to run raspimjpeg on an A+, not to mention simpler code/leaner dependencies (since motion does a ton of other things we don't use) not to mention that an implementation using the GPU can work at higher framerates and help with things like license plate detection for example.

silvanmelchior, 2015-04-16:
Sounds pretty interesting to me because I don't really like motion for motion detection, I think it's very slow and sometimes doesn't react at all. And as you said, the CPU-ussage and more important, the higher framerate would allow a much better detection.

silvanmelchior, 2015-04-16:
I'm going to do some changes to show the motion-vectors from the h264-encoder on the motion-page on the webinterface, this way we can easily decide whether or not we want to replace motion with this one. I let you know as soon as it works.

roberttidey, 2015-04-16:
This sounds exciting both being more effective, lowering load and simplifying the overall scheme.

I think the key parameters that would be nice to replicate here would be : vector Value change (like Noise), Number of vectors (like threshold), Mask area of image to include, number of changed frames to trigger start, number of still frames to trigger stop (Gap)

spikedrba, 2015-04-17:
on top of what has been discussed so far, I've been doing a lot of tests at night and it seems to me motion isn't really stable in those conditions, or at least something communicating with it isn't: I've had two cases where the itnerface got stuck "recording", two different pis, same config.

silvanmelchior, 2015-04-17:
I made some tests yesterday with the motion vectors and could extract them from the h264-encoder, I'm going to try to analyze the data now. My idea is to provide a live-preview of the motion vectors on the web interface, this way it would be very easy to calibrate the parameters and so on. Additonaly, it would look pretty cool ;)

silvanmelchior, 2015-04-17:
After a few more tests I have come to the conclusion that I need to process the data on the GPU itself. I think the best would be to convert the motion-vectors into yuv-pixeldata and send to the preview-imagedecoder. On the webinterface could be an option to change between image-preview and motion-preview. I tried to send the vectors to the browser directly and show with html canvas and other things, but the browser is to slow to convert the stream into a live-picture.

silvanmelchior, 2015-04-17:
Unfortunately no success yet after hours of debugging. I try to convert the motion data into YUV-Frames and send them to the image converter for the live-preview, so it would be easy for the users to check what the motion detection reqognizes.
I added a new brach to the userland named MotionDetection, so if anybody has an idea what could be wrong, I would be very happy.

spikedrba, 2015-04-17:
@silvanmelchior I don't see any branch here (or the circular buffer for that matter):

https://github.com/silvanmelchior/userland

where are you pushing this stuff to?

silvanmelchior, 2015-04-17:
To roberts fork of the userland, my repo is out-of-date

silvanmelchior, 2015-04-18:
FINALLY, a first success. I was able to convert the motion data in YUV420P-Format and then the GPU converts it into JPG for the live-preview. A little demonstration is here:
https://www.youtube.com/watch?v=5Yw94-f3GnY

If you want to try it, here is the binary: http://grustu.ch/various/raspimjpeg
You need to have the buffer enabled and the resolution needs to be set to standard at the moment.

Now there is one big question: How is it possible to detect motion without buffering/recording. Because this is important for the night preview and so on. One idea would be to split the preview-output of the cam and then simultaneously convert it to jpg for the preview and h264 for the motion detection part. What do you guys think?

spikedrba, 2015-04-18:
woohoo :+1: 

this might end up being a design choice more than a "how do I do this". To me the whole idea that the preview is not the same as the recording isn't helpful, in fact the opposite, it's misleading. As a user I expect first and foremost consistency. In that sense to me recording/buffering all the time and showing that rather than the brighter preview would be the correct behavior. At that point this problem no longer exists. Again being in mind my use case, I'm working to ward a security camera system, thinking I have clear recording and then when something happens I go watch the footage and it's too dark would potentially costs me a lot. Literally.

fwiw in picamera this is commonly done by using a different splitter port and recording while throwing the recording away and just keeping the motion vectors. This also has the benefit of being able to use a resized/sampled down stream. See this for an implementation I was prototyping:

https://gist.github.com/spikedrba/aecb82d8b51e991dbd01

This works very well for me. Before deciding to use raspimjpeg I was going to add another start_recording, throw away the video and just pass a sampled down version to a custom output to spit our a jpeg to the ramdisk like raspimjpeg does.

roberttidey, 2015-04-18:
Great news

However, I'm not able to run it successfully at the moment plus few questions to make sure I understand this so I can make some sensible input.

1) Tried this out both from the your binary plus one I built. (They are slightly different sizes but give similar results). I have default 1920x1080 video settings. If buffering is on I get the live preview motion vector image for a few seconds then it dies with a Error: Could not get buffer for motion data. Sometimes it dies straight-away but always within 10 seconds. If I set buffering to 0 then it doesn't crash but starting video recording isn't updating cam.jpg like I would expect.

2) As I understand it the h264 call back is getting the motion data. This can be written to a file if motion_file is configured but is not used in the demo. The test flag sets it up to encode the motion data to cam.jpg so that the result is visible in normal preview. Does that sound right?

3) In terms of integrating this in, in the short term into the current environment, I can see a few possibilities. We could ignore the preview compensation issue for the moment as it is also out of alignment with the actual video recording as spikedrba says. So we then add commands to give user a choice between normal view and motion (mainly just for fun / set up), and we add in code to raspimjpeg to process the motion data into start stop detection, If that is built in then I assume we don't need the motion_file export? We would then need some extra commands to pass in appropriate detection threshold parameters like I mentioned earlier.

4) For initial integration we could just pipe out the detection start stop so they go through the same process as today. This is not maybe a clean as one would like but avoids making too many changes at once. On the web side the motion set up would need rework and we could just abandon that page and put the internal motion set up parameters and command calls on the main page under a new accordion.

roberttidey, 2015-04-18:
I added a bit of tentative motion support material to the branch. I have not integrated in or touched RaspiMCam.c There is a new module RaspiMMotion.c intended to  contain the motion detection material. The actual algorithms processing the vectors are not there just initial parameter and command support to mimic the existing motion set up and frame processing, although I realise we may want to change and enhance that. The citrusbyte method mentioned earlier seemed to be similar in using x^2 + y^2 as a noise measure and then using a threshold count similar to motion. 

I am not sure about the vector buffer structure. I know it is 4 bytes per block but some references seem to imply sad + x + y and others x + y then sad.

I also have an index page with these parameters / commands under a new accordion below camera settings and the old motion settings button gone. 

I am still getting crashes (Could not get buffer for motion data) even with larger buffer. It seems to last about 3 or 4 minutes but I don't see any pattern.

silvanmelchior, 2015-04-19:
1) Strange, I runned mine for 30 minutes without any problems, I'm going to do more tests. I'm not at home right now, but I should have some time this evening.

2) Yes thats true

3) The problem is that if we force the recording while motion detection is active, we disable the option to just take images in the night, triggered by motion detection. And the motion_file option could be useful for further projects I thought.

4) Sounds good

Well, my first goal is to switch between live-preview and motion-vectors on the webinterface, even if there is no capturing/buffering active. I try to add a splitter to the preview output and connect the h264-encoder either to the splitter or to the capturing outputof the camera, depending on the state. Or do you have a better idea?
Of course you can start with the processing of the vectors as you think it would make sense.

roberttidey, 2015-04-19:
I think we can pursue both. My idea was to get a replacement for motion in place quickly even if we then wanted to add more sophistication later. That way we can put this out and remove motion altogether. Even just using the noise / threshold counting parameters I think this is likely to do a much better job than motion itself as it should be less sensitive to light changes.

Most of the infrastructure for the motion detect is in place although obviously untested. It is going to need calls to setup_motiondetect to init or when major parameters change, plus calls to set_motion_state (0 / 1) to enable disable. It will then need the analyse_vectors to be called when new vector buffers arrive. I was planning just to start with the x^2 + y^2 calculation for noise counting to start with.

It would be nice to get the sizing of the vector buffer dynamic but I think from your TODO's you are planning on that.

I have temporarily parked a replacement index.php (named as indexmb.php under the www-src) rather than create a separate branch at the moment. That has the new accordion for the motion parameters, although I left the existing motion set up in as well. I have been running with that and it seems OK.

There is an added cmd vm intended to be used for switching between a vector display and normal but it is just a cmd at the moment and does nothing.

silvanmelchior, 2015-04-19:
Thanks a lot. I moved it to a new branch, so we can easily check and merge at the end.

silvanmelchior, 2015-04-19:
I updated raspimjpeg:
- choose live-preview or motion-vectors from web-interface
- bigger preview-picture for vectors
- support different resolutions than 1080p
- I added a counter to check if the "Could not get buffer for motion data"-error happens just sometimes or all the time. If it happens just sometimes, it could be that the image-encoder is to slow to show every single frame, so with the counter it just droppes a few frames and gives an error if this happens to often. Could you check if you still get the error with the update?

My next step would be to make motion detection without capturing/buffering possible, I'm pretty busy next week, but I'm working on it as much as possible

roberttidey, 2015-04-19:
Those changes sound good, but I am struggling to make them work at the moment.

In motion vector preview off mode then it is working normally. If I switch it on then preview image freezes. raspimjpeg is still running and I can set the motion vector preview back to off and the normal image resumes.

When the motion vector preview is on the the shm/mjpeg folder has a frozen cam.jpg plus a file called cam.jpg.part of length 0 and also frozen.

Logging also shows that when I switch to motion vector on it starts logging 'image' for every frame. Setting it back to off then this logging of 'image' continues even though the display is back to normal.

silvanmelchior, 2015-04-19:
Hmmm, sounds strange, my rpi is working for 3 hours now in motion vector
preview-mode without a problem. Does your config-file contain all new
options?

roberttidey, 2015-04-19:
Yes. To start with I just copied in the raspimjpeg.1 into /etc and used your binary. I get similar results with a binary I have built.

The 'image' logging is weird because there is no printLog or error in raspimjpeg with that but then I noticed it is coming from web logging not raspimjpeg but only when Motion preview is turned on. I am continuing to look into that.

I'll also try a full clone rather than patching my existing set up just in case there is something I missed. I had updated raspimjpeg, its config, index.php and config.php

roberttidey, 2015-04-19:
I fixed the logging issue was some local debug I had that was getting triggered. Rest of the behaviour remains the same, however, including the cam.jpg.part.

Update: I did a full upgrade rpi-update reboot etc. Still didn't seem to be working but then I upped the buffer from the 2000 to 10000 and it sprung into life. So maybe there is something still dependent on the buffer size.

silvanmelchior, 2015-04-20:
Great, I try to find out why small buffers don't work.

One question: What is the idea behind the vector_mode-parameter you introduced?

roberttidey, 2015-04-20:
It was a placeholder for switching to vector preview so is now redundant. vector_preview is a better name.

silvanmelchior, 2015-04-20:
I updated raspimjpeg. Attention, nearly nothing works right now, but I share it with you guys because I don't have time to continue my work today, so maybe you find some problems.
In the new version is a splitter behind the preview-port of the camera. The jpeg-encoder for the live-preview is connected to this splitter so generate the preview as normal. Additionally is the h264-encoder connected to the splitter if no buffering/capturing is happening. THe h264-buffer get reconnected to the video-output of the camera if needed. Some points for the new version:
- I recognized a higher latency for the live-preview. This could be because with the splitter is an additional component between the camera and the image encoder. Do you think this delay is a problem?
- If you want to try the motion-vectors, you can change your config files to vector_preview 1 and no buffer, then start raspimjpeg. The switching on the website doesn't work at the moment. There are also a lot of errors with the buffer, but as I said, I can't fix them right now.
- Capturing/buffering doesn't work at the moment, I need to fix this.

So if somebody of you finds a bug, please report ;). Otherwise I try to continue tomorrow.

roberttidey, 2015-04-20:
Yes. Definitely having some problems trying to use this one.

I get a  mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x1302f90 (EINVAL) on start up

raspimjpeg is still running and updating cam.jpg 

I was also getting please restart raspimjpeg from web site even though raspimjpeg was running. That seems to have sorted itself out. I guess the status file had got out of sync somehow.

spikedrba, 2015-04-20:
Hi there, sorry for the absence over the last couple days, had to deal with some stuff.

Just tried out the branch and getting same errors as @roberttidey 

```
RaspiMJPEG Version 5.2.0
Error: Could not get buffer for motion data
Error: Could not get buffer for motion data
Error: Could not get buffer for motion data
Error: Could not get buffer for motion data
Error: Could not get buffer for motion data
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x9c1a70 (EINVAL)
Could not set bitrate
mmal: mmal_vc_port_info_set: failed to set port info (2:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x9c1750 (EINVAL)
mmal: mmal_connection_create: format not set on input port
mmal: mmal_connection_destroy_internal: connection vc.ril.camera:out:1/vc.ril.video_encode:in:0 could not be cleared
Error: Could not create connection camera -> video converter
MJPEG streaming, ready to receive commands
```
with motion detection disabled the cam works just fine.

fyi: I normally just got a black image except for once where I got a bunch of mixed pixels, not sure why, couldn't reproduce. Attaching it here just in case it helps:
[image]

I'm trying to figure out cmake and the rest of the toolchain to see if I can help any. Please feel free to share any tips/guidelines on this, especially docs around mmal that you think could help me to come up to speed.

thanks,

Spike

roberttidey, 2015-04-21:
Here is a tiny clip taking using the previous non-split version which seems to be working pretty well as long as I set buffer to 10000
https://dl.dropboxusercontent.com/u/18838174/Vectors2.avi

For changing / compiling directly on Raspberry it's very easy you just sudo apt-get install cmake ,then clone this userland, cd into it and run ./buildme. First build takes a while but incremental builds after that are much quicker. On a Pi2 an incremental takes about 15 seconds. The only files of interest are those beginning RaspiM. If you want to build on an external platform then there is some guidance on the raspberry userland github on doing that on linux.

userlandmb/host_applications/linux/apps/raspicam

after a build a new version ends up in build/bin

silvanmelchior, 2015-04-22:
New progress: it is now possible to change between motion-vectors and live-preview on the webinterface without a buffer. Depending on the preview the setup is

camera -> splitter
                       |----> resizer -> jpg-encoder -> webpage
                       |----> h264-encoder -> motion detection

or

camera -> splitter
                       |----> null-sink
                       |----> h264-encoder -> motion detection -> jpg-encoder -> webpage

At the moment, capturing or buffering is not possible. If you press capture, the h264-encoder is disconnected from the splitter and connects directly with the video-output of the camera. This works and a file is generated, however MP4Box can't convert this file, but I can't find the bug right now.

Please test it a bit and say if it works for you. And if anybody can find the problem with the recording, I would be very glad.

roberttidey, 2015-04-22:
Runs providing buffer set to 0. I think you say this is expected.
Starts up with
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0xdf40a0 (EINVAL)
Will do more testing tomorrow.

spikedrba, 2015-04-23:
same here (same error, same thing with the buffer, but that was expected per previous comment).

As an aside, I was looking at the motion video and kept seeing blips of light on the background and then I thought about my settings lock issue ( #44 ): if I set exposure mode to off the motion video is **way more accurate**: no more flashy background and random pixels deriving from the camera flickering as it keeps re-adjusting itself. I think it becomes especially important for this motion part to figure out settings locking (I put more info about fps 0, shutter speed and gains in the other thread on dark preview).

spikedrba, 2015-04-23:
fwiw, I've also been keeping an eye on cpu usage and performances and so far on the MotionDetection branch. I get consistent number with raspimjpeg running at about 8% cpu as it used to do in normal video mode and ramp up to 14% when vector preview is turned on. I'm unclear at this point what's taking the cpu since the vector computation is on the GPU, but I've yet to look at the implementation and understand the code.

silvanmelchior, 2015-04-23:
I think the converting of the vector-data to a yuv420p-frame for the
jpg-converter is responsible because this runs on the cpu and is executed
for every single frame. But I think this isn't a real problem because
you're normally not interested in the vector preview itself.

roberttidey, 2015-04-23:
This may be a stupid thought but I'll go for it anyway.

What if buffering were always on in the background with either user setting or some minimal level to make things like vectors work? Then if user has buffer set to 0 then we ignore the contents of the capture buffer for the purpose of building the capture file. I.e. we just do the normal h264 streaming into the file. 

I am still trying to understand the splitter piece, but I'm not sure I understand the connection back to the camera during capture. Haven't we got to keep the motion vectors going during capture as well so we can detect stops?

spikedrba, 2015-04-23:
wfm. fwiw to me that's also worthwhile to resolve the discrepancy between preview and actual recording (re convo on brightness of preview and fps).

silvanmelchior, 2015-04-24:
The problem with this approach is that it's not possible to detect motion
at night and then take a snapshot for example or something else. This is
possible with motion and I don't want to publish an "upgraded, better"
version of raspimjpeg which doesn't support things that were possible
before.

If this inconsistency between preview and motion is a problem for you, you
could still enable the buffer, this way the h264-encoder is connected to
the video-port again as it was usual until now.

spikedrba, 2015-04-24:
provided that my case should not in any way dictate what is useful to the larger community - you'll never hear anything from me like that - I don't understand why it's not possible to detect motion at night if the buffer is on. If the issue is high fps that would be honored if recording, therefore forcing a darker preview that is currently split to feed motion, then I'd argue you have a setup problem. Especially since there's support for day/night periods in the scheduler, when night hits, fps setting should lowered anyway and you should have an IR light. The sensor is rated as 0.67 lux which is almost full moon (1 lux). In anything less than that my direct experience is that you can't basically make anything useful out from the image. In my current night period I switch to 10-15 frames, raise brightness to 70 and contrast to 60 and I can get a pretty decent preview that's consistent with the recorded video.

again, I totally understand your reasoning, just trying to share a different point of view, but maybe I'm simply not understanding you.

silvanmelchior, 2015-04-24:
To be honest, I don't understand why you don't like the version where
motion detection is possible with and without buffering. This way you can
either use raspimjpeg as usual or use buffering if you don't like the
inconcistency between preview and capture.

spikedrba, 2015-04-24:
@silvanmelchior, thank you for discussing this with me. It's not that I don't like to have the choice between motion with or without buffering, to me it comes down to safe defaults and protecting the user. Again I might be wrong, but I expect most people will be thrown off by the inconsistency between preview and videos in low light, with some maybe only finding out when it's too late. That being the case I'd rather have it on by default, explain the situation, and then let an aware user make the choice, while the unaware one gets no surprises out of the box.

Personally I'm perfectly fine the way it is now that I understand what's going on and I don't mean to insist on this, that's not the intention.

best,

silvanmelchior, 2015-04-27:
I had some spare time this evening, so I continued my work and updated raspimjpeg. It should be possible now to see the motion-vectors or live-preview while streaming, buffering or capturing. My next step is to implement motion detection itself and then merge the branch back into master.

I would be glad if you could test the new version exactly and report every single bug you've found.

roberttidey, 2015-04-27:
This version seems pretty stable. I got one error Error: Could not send buffer with motion data but that was during a change in video mode. Otherwise I can change settings around and set vector preview on and off without problems.

I do notice a difference in vector display with buffer on or 0. With 0 although I see the vectors of moving objects there is a lot of background 'noise' even with nothing changing. If I turn on buffering even a small value then this noise disappears and I see just motion and if there is no motion then the screen is effectively black.

spikedrba, 2015-04-28:
on a B+ with everything up to date:
- I did a git pull + install, refreshed the browser and it was locked up (I had uconfig enabling motion preview)
- I rebooted/refreshed everything, and things looked ok for about 5 minutes switching between video and motion and back, etc (no recording, 0 buffer). then I started noticing motion video skipping, there was visible lag. A few seconds later it completely locked, would not respond to a kill, had to kill -9 just like @roberttidey reported.
- regardin background noise that @roberttidey mentioned, I get a lot of it, and as I already commented on it previously, for me it goes away as soon as I set exposure mode to off. This goes back to locking settings, seems even more important for motion
- while watching motion vectors, I tried to hit record and saw in console that recording happened and then boxing kicked off and that's it, nothing else happened and motion vectors froze. The boxing completed and I could watch the video from the preview (as an aside, I thought I was gonna record the motion vector video, but instead it was the normal one). Any further change in the web frontend came up in the logs, but raspimjpeg never recovered and the image remained black. Previoiusly when things locked up I saw plenty errors "could not get buffer data", but this time there was nothing.
- with the buffer set to 1000 I got no crashes over 30mins or so of testing
- I tried to lower the buffer to 100 and that was still ok
- I tried to set it to 0 and as soon as I hit OK the whole thing crashed. Interestingly enough first I got a bunch of "Error: Error in iframe list" then " Change: video_buffer = 0" got logger and immediately after that "Error: Could not get buffer for motion data". I had to kill raspimjpeg again. Note that this time it died nicely at I hit C-c.
- I tried to restart it from console with the buffer set to 0 from the previous session and it immediately locked up spitting "Could not get buffer for motion data". This time a C-c would not do and I had to login from another terminal and kill -9

I think I'll leave it at that for now.

two asides:
- I noticed the motion preview shrunk considerably, were you planning on adding options to choose size in the web panel at some point?
- the preview even of normal video seems slightly lagged, but maybe I'm misremembering and that's always been the case, might setup another cam up later to compare with an older version of the code.

silvanmelchior, 2015-04-28:
So it would be good to find an exact scenario where it freezes every time with allowed buffer settings >= 1000 or without buffering. If the buffer is set so 100 or other small values, the intraperiod is set very very low too, so this could be a problem. But I think a buffer smaller than 1 second doesn't make sense, so this shouldn't be a problem.

@spikedrba I shrinked the motion preview because the CPU + GPU were to slow to create an yuv420p-image out of the motion vectors and convert these to jpeg, especially together with the splitter and so on, this was the reasons for the buffer-errors. And the lagging could be a problem with the splitter, my idea is to redesign raspimjpeg again a bit, so the splitter is just used if it is really necessary (live-preview, no buffering and motion detection turned on)

spikedrba, 2015-04-28:
makes sense. I will re-run a batch of tests before EOD with buffer 0 and buffer 2000. Should you then have a check on buffer length to make sure it's >=1000? that seems to make sense if we know it will cause problems and I completely agree that it doesn't make sense to anyway. Maybe the whole parameter could simply becomes expressed in seconds rather than ms.

spikedrba, 2015-04-28:
with 0 buffer:
- enabled vectors, watched it for a while while juggling, no problems
- enabled motion detection recording, nothing happened (I think this is normal, still not triggering recording, right?)
- disabled motion, hit recording, worked just fine
- disabled vectors
- started recording from normal picture and then enabled motion vectors view. it got super laggy for a few moments but then I could see motion for another few seconds and then it segfaulted (in the past it just sat there not responding, this time it literally segfaulted). Nothing in the logs. It also seemed to have left with an incosistent setting state, after restarting the process it showed a black square like it was doing motion vectors and showing the dropdown as off and no motion movement. Had to toggle that and then it went back to normal image.
- I tride to repro and behavior was slightly different: while recording and switching vector on I got no crash, but I did get another segmentation fault when I turned it back to off and this time I also saw "{2015/04/28 10:52:22} Error: Could not send buffer with motion data" in the log. Also after restarting raspimjpeg the web interface was responding and correctly showing motion vectors, BUT, the settings on the webinterface said off.
- as an aside, one time out of 3 when I clicked recording while watching motions the thumbnail picked up was of the motion vectors even tho the recording was of the normal video
- I tried to snap a picture and thumbnail was also of the motion vector
- similar but different, I also got a segfault while trying to stop recording while watching motion vectors. In the log it never shows anything about capture being stopped.

with buffer set to 2K:
- lag went up in normal video
- after taking two snapshots motion vecors image was not responding, but no crashes, had to toggle it and then it came back
- got one error "{2015/04/28 11:01:19} Error: Could not send buffer with motion data"
- I started capturing from normal video, switched to vectors, no crashes however tons of "{2015/04/28 11:04:43} Error: Error in iframe list" errors. When I tried to switch vectors off it went back to normal image, but capturing also stopped tho there's no trace of it in the logs. There is h264 video created and omxplayer will play it. No sign of mp4boxing even tho the web interface says busy,

that's as much as I can test right now, please let me know if there's any more specific test cases you'd like me to run through.

All of this is on a B+ with 128M dedicated to GPU.

Spike

silvanmelchior, 2015-04-28:
Great, thanks a lot! This will help me to develop a new version. I let you
guys know as soon as I'm ready. And of course, feel free to search the bugs
on your own if you have some time left ;).

silvanmelchior, 2015-05-01:
New update, the splitter is now just added if it is really used (motion detection active and live-preview without buffering)

About the changing of the preview while recording: The problem is that it needs a lot of reconnection in raspimjpeg to switch between the previews and this is done by recreating the GPU-structure from scratch, so it is not possible to change between the previews while recording. I also think this isn't really necessary, so I suppose I'm going to disable the html-selection input while recording.

I couldn't fix the other problems yet, so some tests may result in the same behavior.

roberttidey, 2015-05-01:
Built that one and working OK so far.

I don't seem to get the rippling effect on normal 0 buffer image and motion vector is clean as well with 0 buffer. That applies with motion turned on or off.

roberttidey, 2015-05-02:
I did a small tweak to the motion vector scaling. I changed the index.php to scale image display to preview width. This doesn't affect normal view but upscales the motion vectors to same size. Then I removed the 2x2 scaling in raspimjpeg which lowers the load there and also makes the buffer suitable to pass to motion detect.

The live motion correlates with the image and looks suitably blocky as one would expect.

spikedrba, 2015-05-03:
I didn't built anything, should have I? I'm just testing with whatever was in the branch. @roberttidey your upscaling changes look great, thank you!

@silvanmelchior totally makes sense not to be able to change preview while recording and certainly not an issue, thanks.

Here's what I'm seeing.

With buffer 1K:
- The camera started with a 2000 buffer, as I changed it I got about 20 iframe errors right after the change in the same second (runnig at 25fps):
```
{2015/05/03 10:10:11} Change: video_buffer = 1000
{2015/05/03 10:10:12} Error: Error in iframe list
...
{2015/05/03 10:10:12} Error: Error in iframe list
```
- I'm seeing a lot of noise, lots of pixels all over the place constantly flickering, but then the image itself is also noisier. I will double check the hw again, but it's the same setup/noir camera sitting on my desk indoor with the same light conditions so a bit puzzled.
- I actually went to platy with the light and changed FOV back and forth and now most of that noise is gone. quite puzzled by it.
- video recording while watching motion vectors worked fine with no errors, altho snapshot is still a black square of the motion vectors
- after recording I went to watch the video and going back from the preview page yield a stuck black image. refreshing didn't change anything, same frozen black pic. I had to set motion off and it went back to the right normal preview and then on again for it to work again. No errors in the logs
```
{2015/05/03 11:17:49} Add /var/www/media/vi_0012_20150503_111739.mp4 to Box Queue at pos 0
{2015/05/03 11:17:49} Start boxing /var/www/media/vi_0012_20150503_111739.mp4 from Box Queue at pos 0
{2015/05/03 11:17:51} Finished boxing /var/www/media/vi_0012_20150503_111739.mp4 from Box Queue at pos 0
{2015/05/03 11:20:38} Change: vector_preview = 0
{2015/05/03 11:20:51} Change: vector_preview = 1
```

With buffer 0:
- changed buffer to 0 while watching motion vectors, no problems
- same issue about things being stuck when going back to motion vectors stream after watching a recorded video. Setting motion off and on solved again the issue. Do not, setting it to off by itself didn't do it, I had to refresh the page to see the normal image and then set it to on again

I know you said you didn't change anything else but modifying the behavior of the splitter, but I could not get any of the segfaults I used to. I did not try to change mode while recording since we said that will be disabled, but starting/stopping recording while watching motion vectors never segfaulted with or without buffer.

As it stands I'm only seeing minor issues with thumbnails and comign back from preview, but otherwise everything look good to me!

looking forward to motion detection working! very excited!

thanks a ton,

Spike

roberttidey, 2015-05-03:
I've had my first internally triggered motion capture. It is using a fairly crude change count algorithm at the moment just to get the overall process checked out.

As the counts are on the full 25 fps frame rate one wants relatively high changed frame counts to start and even higher ones to stop. I started with noise 20, threshold 200, start frame 5, stop frame 50. I suspect we may want to use time based at least for the stop (like motion gap).

The problem I have is that a motion stop is not clean and throws an error Could not enable connection splitter -> video converter and I sometimes get  Error in header bytes so I haven't pushed the binary at the moment.

spikedrba, 2015-05-03:
\o/

as an aside, could you explain me a little more how noise and thresholds are going to be used? Time based stop condition like gap makes sense to me, also the more we're consistent with motion settings the easier it's going to be for people to transition.

Also as I was trying to improve my night captures and looked more closely to motion I learned more about despeckle and found it quite powerful and useful (as opposed to an image mask). Here's the best explanation I found along with a C implementation that's currently used in Motion:

http://emit.demon.co.uk/motion/
http://emit.demon.co.uk/motion/despeckle.c

Not sure if that's going to be too heavy to be plugged into raspimjpeg, but from the standpoint of a security camera I think it's a really important feature, especially for outdoor installations.

roberttidey, 2015-05-03:
Currently I was using those in a similar way to 'motion' but working on vector data rather than pixel data so should be better. As I said this is crude but is just to get things going.

So for the vector data it goes through counting the vectors that have a value above the 'noise' level. I.e small vector changes have no effect. Then if the counts in any one one frame is above the threshold level (movement) then frame count is increased. If it is below the threshold frame count gets reset to 0. This means that it needs n frames of 'movement' in a row to trigger a motion start.

Inverse mechanism applies to off.

Although the base code for a mask is in there I haven't added that in yet. Even if we make improvements in the base algorithm, a mask is really useful to cut out regions you really don't want to get triggers from. For example, in my case, I have one camera which sees a bit of a road in a corner and I do not want it triggering on passing cars.

I think it would be good to get a version that is at least as good as 'motion' going and then there are lots of room for improvements. For example rather than using a noise and count it might be better just to get an overall frame vector metric. SImilarly, one could be smarter with interframe filters rather than the count and reset method i put in for starters.

I have pushed this binary as I don't think it is any worse in no detect mode and it does seem to work after a fashion.

silvanmelchior, 2015-05-03:
Any ideas on how to prevent the image/video preview to be the motion vectors? A resizing of the large image/videoframe on the gpu would be complex, on the cpu it would be slow.

roberttidey, 2015-05-03:
I guess you mean the thumbnails in the download screen which are 'vectors' whereas the images / videos associated with them are 'normal'.

The way those are captured at the moment is by grabbing the cam.jpg at the start which is a very cost effective way of doing it and has other advantages like showing data which caused a motion trigger whereas converting a frame from video would show nothing if buffering is in use.

I did originally try converting when I first started to put thumbnails on the download screen but it can be painfully slow on the CPU.

The reason why we get the vector thumbnail is because we are sharing the cam.jpg between this thumbnail capture and the live preview. So in vector preview mode it is the vector -> jpeg process which is producing the cam.jpg instead of the resized jpg stream from MMAL.

So this does lead to a possible solution where in motion_preview mode the cam.jpg is produced like it is today from the vector data as that keeps the web interfacing side simple, but we direct the normal jpeg to a separate file name and that is the one grabbed for the thumbnail. 

But my other question would be is this a big deal as I suspect the motion view is really more for set up, or do you see it as useful in its own right?

spikedrba, 2015-05-03:
I'd +1 to split the preview/thumbnail thing to its own issue and make it low priority as I agree with @roberttidey that it's not a big deal - at least personally I don't anticipate having vectors on and recording other than in a setup phase - it just came up on testing as I was trying to be thorough with possible combinations people might ran into.

silvanmelchior, 2015-05-04:
New update: I changed the iframe-handling a bit, I suppose the problem was that the intraperiod-change was not fast enough and so there were iframe-errors until it really changed.
Additionally I made some changes to the webinterface to prevent the change of the preview while recording and I adjusted the motion settings a bit.

I think we are moving to a version where it is possible to merge it back into the master and update the community about the change. So it is important now to test everything again and find possible bugs.

spikedrba, 2015-05-04:
on it! and can't wait for it, I've been literally wasting the morning trying to figure out issues with motion, for whatever reason it's picking up things too late or completely missing the end of the action even tho settings look alright. Sounds like something I'm doing wrong but can't see how. Also since @roberttidey pushed out changes to start processes as www-data I seem to be having some serious problems, but holding off to investigate/report if we can complete this move!

spikedrba, 2015-05-04:
- had motion turned on before the update. went ahead pulled/installed, rebooted, went to the browser, loaded the interface and had a black image, no motion. I had to turn motion off and then back on for it to work. everything was fine after that.
- with motion detection on, I was still able to turn motion preview on, which incidentally triggered a recording. Should this be allowed? It seems to be equivalent to having a record process going almost, which is indeed now not allowing to change mode.
- I'm unclear about usage of the cirular buffer along with motion. I thought it was happended, but that doesn't seem to be the case. I now have 10000ms/10s buffer and recorded myself with a stopwatch counting to 6 secs. the video is 13s long. shouldn't it be ~16?
- tested changing FOV with vectors on and had no issues

fwiw stats wise I saw an avg of 10% cpu utilization and 0.6% mem on a B+, overall the PI is super unloaded (0.43 load 5, most likely from mboxing, 0.14 on load 1). Pretty sure this has a chance to work on an A+ too \o/.

I've also tried to play with noise and threshold and this is where I think we need to improve, but nothing to do with breaking away from motion. Basically it's just too hard simply by watching the video to understand what good values for those should be. Somehow it would be great to have some live preview of the counters so I could wave my hand and say "ok that changes this many vectors by this much so if I want to avoid a moth, but pick up a person A and B are two good settings". Does that make sense?

I'm gonna upgrade the camera outdoor I was going to have problems with and let it run for the rest of the day and see what happens, but manual fiddling is all coming back super positive.

fantastic work @silvanmelchior and @roberttidey , I'm incredibly thankful!

Spike

spikedrba, 2015-05-04:
this stuff showed up on the outside camera with 1s buffer and very low noise/threshold (it's a wide angle and it wouldn't see my motion if I showed up at the edges. With thsoe low settings it picked up a lot of tall grass being blown by the wind, so I turned it down, but in the meantime I noticed this in the log:

```
{2015/05/04 11:56:08} Finished boxing /var/www/media/vi_0317_20150504_115538.mp4 from Box Queue at pos 2
{2015/05/04 11:56:47} send smd 0
[2015/05/04 11:56:47] Stop capture requested
[2015/05/04 11:56:47] Send ca 0
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Error: Error in iframe list
{2015/05/04 11:56:47} Capturing stopped
{2015/05/04 11:56:47} Add /var/www/media/vi_0318_20150504_115604.mp4 to Box Queue at pos 3
{2015/05/04 11:56:47} Start boxing /var/www/media/vi_0318_20150504_115604.mp4 from Box Queue at pos 3
{2015/05/04 11:56:48} send smd 1
[2015/05/04 11:56:49] Start capture requested
[2015/05/04 11:56:49] Send ca 1
{2015/05/04 11:56:49} Capturing started
{2015/05/04 11:56:52} send smd 0
[2015/05/04 11:56:58] Stop capture requested
[2015/05/04 11:56:58] Send ca 0
{2015/05/04 11:56:58} Capturing stopped
{2015/05/04 11:56:59} Error: Error in header bytes
```
never seen that last error in header bytes before.

[58 more replies]