silvanmelchior/RPi_Cam_Web_Interface #58: Stopping motion triggered recording

roberttidey, 2015-04-17:
Currently if a recording is motion initiated most of the buttons are disabled until a motion stop is received. This could lead to a situation where a motion stop is not received or has got lost for some reason and the user has no easy means of recovery.

I did contemplate enabling stop camera and whilst these shuts the main camera operation down the recording is still in limbo in the background and does not start up clean probably due the internal status flags.

A simpler solution would just be to label and enable the stop recording button. This I think is fairly harmless as if a motion stop then gets received it will just be ignored.

I'll try this out.

spikedrba, 2015-04-17:
@roberttidey fwiw I mentioned this in another issue last night, I had a few of these, somehow motion ran away and I had to issue an echo "ca 0" manually to get things back in place.

That said, with the ongoing effort to integrate motion into raspimjpeg maybe any further work to improve Motion isn't worth it?

just a thought

silvanmelchior, 2015-04-18:
As spikedrba said, I would wait to see if we manage to implement motion detection with the h264-encoder. But a button to manually stop would be good in both cases.

closed 2015-05-02 by roberttidey