silvanmelchior/RPi_Cam_Web_Interface #60: Stop Start camera

roberttidey, 2015-04-21:
I think there may be an issue with stop_all not releasing all the resources so that stop_all followed by start_all does not always fully clear out and restart the MMAL interface.

I noticed this because occasionally I can get the system to freeze using parameter commands that use a stop_all start_all to ensure the change takes place. When this happenes manually issuing a stop camera / start camera does not recover the system and raspimjpeg itself has to be stopped and started.

There seems to be some components and connections created that are not destroyed in the stop_all.

silvanmelchior, 2015-04-21:
I updated stop_all in the MotionDetection branch yesterday to include all
resources, I hope this fixes it.

roberttidey, 2015-04-22:
I saw those. Thanks.
I tried to fit a subset of those to the pre-split version. It works but I get the same error as the spliiter version, that a component can't be disabled because it isn't enabled. I am not sure how we test for that if we want want to make the stops dependent on waht state they are in.

silvanmelchior, 2015-04-23:
How does it work with the new update?

roberttidey, 2015-04-23:
I'll retest with changes made last night/today