silvanmelchior/RPi_Cam_Web_Interface #63: Region of interest

roberttidey, 2015-05-02:
I don't think region of interest is working. No big deal as I am not sure what real use this is.

I had noticed a problem in the command handling of the ri command where it wasn't picking the right parameters but even with that corrected the display just breaks up into noise.

silvanmelchior, 2015-05-03:
To be honest, I never used this setting, so we could ask in the forum if
everyone uses it and what's the idea behind it.

roberttidey, 2015-05-03:
OK will do

spikedrba, 2015-05-03:
fwiw never used it either, but my limited understanding of it is to simply crop the image to a smaller area you really care about. So for example if you were filming a street and had houses and trees on the sides/edges that you don't care about because you only want the street in the middle, you could set your region of interest to that. This allows you to keep a wide angle lens to capture a longer stretch of the street while isolating the road from the rest of the image. I can definitely see how this could come useful in some scenario, but as I said haven't used it yet myself.

roberttidey, 2015-05-03:
That sounds what it should be used for but it doesn't work at the moment. If I try to set it big so it has no effect (like the 0,0,65536,65536) then then the camera connection errors. If I set it smaller than video height, width the whole display cracks up.

Either way I think it is a low priority item.

spikedrba, 2015-05-03:
agreed on the low priority

mark-hahn, 2019-10-11:
I need it to do zooming in on image by the user.  I can crop in the browser but then I'm still using the full bandwidth.  Is it still broken?

roberttidey, 2019-10-12:
I'll check the status of this.

roberttidey, 2019-10-12:
The ROI feature does work but it can be confusing how to set the parameters. It is also worth noting that ROI sets the area of the image sensor used as the source of data into the main GPU. It does not directly control the size or resolution of the processed data (see later)

The ROI area is actually set in fractional terms of the image area multiplied by 65536, not in absolute pixel terms.

so the default values of 0,0,65536,65536 represent a starting top left at 0,0 and a width, height of 1,1 which is no cropping.

If you were to set 16384,16384,32768,32768 then represent would be 0.25,0.25,0.5,0.5 and would crop out a half size image starting at a quarter way across and down the screen.

Note that
1) the size of the live data stream sent to the browser is being controlled by the preview parameters and not directly by the ROI. If you want to adjust this these parameters

2) The resolutions of the recordings is still set by the resolution parameters not by the ROI. You can of course adjust these to be appropriate to any ROI you set.

I will add ths to wiki to help others.