silvanmelchior/RPi_Cam_Web_Interface #64: Installer / wlan0

DonkordE, 2015-05-04:
Hi.

I have a fresh install RASPBIAN Release date:2015-02-16

I have trying to install rpi cam web interface:
sudo apt-get update
sudo apt-get dist-upgrade
sudo rpi-update
reboot
git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
chmod u+x RPi_Cam_Web_Interface_Installer.sh
./RPi_Cam_Web_Interface_Installer.sh install

If eth cable is connected everything works fine install no problem and system reboot and everything is perfect

But if i use instead usb (active powered usb hub) wlan dongle (module 8192cu) install fail of the end. And after reboot system first load, but after less than 1 minute whole system crash.
Can you help me??

-Janne

roberttidey, 2015-05-04:
So when you install over wifi the install fails? In what way?

What happens if you install over cable, then run under wifi? Does it still crash?

Also might be worth setting Preview divider to say 3 (under camera settings while connectet ot cable. This will lower the traffic when streaming the video which might trigger problems if your wifi connection is not fast.

I have used wifi both for successful installs and running normally.

DonkordE, 2015-05-04:
it does not work over wifi more than 20 second.
there are no difference if usb wifi dongle are connected or not. Only thing it matters is eth cable.
Test 1= no eth + wifi connected > install fail > REBOOT > system crash
Test 2 = eth connected + wifi connected > install ok > REBOOT > system ok 
Test 3 = eth connected + no wifi > install ok > REBOOT > no eth + wifi connected > system crash
Test 4 =eth connected + no wifi > install ok > REBOOT > system ok

from another lan raspberrypi:
pi@raspberrypi ~ $ ping webcam
PING webcam (192.168.1.118): 56 data bytes
64 bytes from 192.168.1.118: icmp_seq=0 ttl=64 time=10.891 ms
64 bytes from 192.168.1.118: icmp_seq=1 ttl=64 time=6.552 ms
64 bytes from 192.168.1.118: icmp_seq=2 ttl=64 time=6.191 ms
64 bytes from 192.168.1.118: icmp_seq=3 ttl=64 time=6.571 ms
64 bytes from 192.168.1.118: icmp_seq=4 ttl=64 time=6.554 ms
64 bytes from 192.168.1.118: icmp_seq=5 ttl=64 time=6.599 ms
64 bytes from 192.168.1.118: icmp_seq=6 ttl=64 time=6.615 ms
64 bytes from 192.168.1.118: icmp_seq=7 ttl=64 time=6.545 ms
64 bytes from 192.168.1.118: icmp_seq=8 ttl=64 time=6.491 ms
64 bytes from 192.168.1.118: icmp_seq=9 ttl=64 time=6.167 ms
64 bytes from 192.168.1.118: icmp_seq=10 ttl=64 time=6.472 ms
64 bytes from 192.168.1.118: icmp_seq=11 ttl=64 time=4.802 ms
64 bytes from 192.168.1.118: icmp_seq=12 ttl=64 time=6.557 ms
64 bytes from 192.168.1.118: icmp_seq=13 ttl=64 time=6.550 ms
64 bytes from 192.168.1.118: icmp_seq=14 ttl=64 time=3.684 ms
64 bytes from 192.168.1.118: icmp_seq=15 ttl=64 time=6.541 ms
64 bytes from raspberrypi (192.168.1.127): Destination Host Unreachable
Vr HL TOS  Len   ID Flg  off TTL Pro  cks      Src      Dst Data
 4  5  00 5400 20fa   0 0040  40  01 42bc 192.168.1.127  192.168.1.118
64 bytes from raspberrypi (192.168.1.127): Destination Host Unreachable
Vr HL TOS  Len   ID Flg  off TTL Pro  cks      Src      Dst Data
 4  5  00 5400 7efa   0 0040  40  01 e4bb 192.168.1.127  192.168.1.118
64 bytes from raspberrypi (192.168.1.127): Destination Host Unreachable

DonkordE, 2015-05-04:
root@webcam:/home/pi/src/RPi_Cam_Web_Interface# uname -a
Linux webcam 3.18.12+ #782 PREEMPT Tue Apr 28 19:49:16 BST 2015 armv6l GNU/Linux
root@webcam:/home/pi/src/RPi_Cam_Web_Interface# ./RPi_Cam_Web_Interface_Installer.sh install
raspimjpeg: ei prosesseja
From https://github.com/silvanmelchior/RPi_Cam_Web_Interface
 * branch            master     -> FETCH_HEAD
Already up-to-date.
Luetaan pakettiluetteloita... Valmis
Muodostetaan riippuvuussuhteiden puu
Luetaan tilatiedot... Valmis
Seuraavat ylimääräiset paketit on merkitty asennettaviksi:
  apache2-mpm-prefork apache2-utils apache2.2-bin apache2.2-common ffmpeg gpac-modules-base liba52-0.7.4 libapr1 libaprutil1 libaprutil1-dbd-sqlite3
  libaprutil1-ldap libav-tools libavdevice53 libavfilter2 libavfilter3 libavformat53 libavresample1 libgpac2 libmysqlclient18 libonig2 libopencv-core2.3
  libopencv-imgproc2.3 libpostproc52 libpq5 libqdbm14 lsof mysql-common php5-cli php5-common ssl-cert
...snap...
Adding system user `motion' (UID 107) ...
Adding new user `motion' (UID 107) with group `motion' ...
Not creating home directory `/home/motion'.
Adding user `motion' to group `video' ...
Lisätään käyttäjä motion ryhmään video
Done.
[warn] Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Setting up php5 (5.4.39-0+deb7u2) ...
Setting up php5-cli (5.4.39-0+deb7u2) ...

Creating config file /etc/php5/cli/php.ini with new version
update-alternatives: using /usr/bin/php5 to provide /usr/bin/php (php) in auto mode
Setting up ssl-cert (1.0.32) ...
Setting up zip (3.0-6) ...
Setting up ffmpeg (6:0.8.17-1+rpi1) ...
Processing triggers for menu ...
cp: tiedoston ”/etc/raspimjpeg” tilaa ei voi lukea: Tiedostoa tai hakemistoa ei ole
Installer finished

root@webcam:/home/pi/src/RPi_Cam_Web_Interface# reboot

roberttidey, 2015-05-04:
Few more questions.

What shell are you using to install or have you switched login to root as you seem to have a # prompt? You can install as a regular user like pi providing they are a member of sudoers.

Is their anything written in scheduleLog.txt?

Have you looked in dmesg for any system errors?

Might be worth turning off the autostart of raspimjpeg, checking whether stable after a boot and then manually starting rapspimjpeg to see if that is the cause of the crash. I'm not sur ehow raspimjpeg could do this if it is the case. I've never had a total system crash in months of operation.

spikedrba, 2015-05-04:
@DonkordE could you also expand on what you mean with "system crash"? do you mean the pi goes completely unresponsive and you need to plug the power?

Is this a new pi? have you used it before for a while with other software?

thanks,

Spike

DonkordE, 2015-05-04:
Hello again , and thanks for quick answer. First I have to apologise. After 3 hour test I almost give up. I borrow my friend another RaspberryPI and install Raspbian + dist-upgrade > install RPi Cam Web Interface and everything are perfect now. Uptime now 1 hour with usb wifi dongle. It looks like my original pi had something wrong.

Fritz4all, 2015-06-18:
@DonkordE 

Do u have a Raspberry Pi B+?

Could it be a power problem? have a look at this post : https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=82584

~~~
safe_mode_gpio=4
~~~
AKA : 
~~~
max_usb_current=1
~~~

Normally these settings are applied for USB mass Storage devices, but you never know ;-)