silvanmelchior/RPi_Cam_Web_Interface #65: Scheduler not purging files with thumbnails disabled

federicosalvi, 2015-05-13:
I've noticed that in the raspimjpeg config file it's possible to disable the thumbnails generation.
However if I do this, the scheduler stops purging old files. 

Looking at the code I've noticed that it's because the purgeFiles() function checks the thumbnail to retrieve the file type and then proceeds to delete all the file associated with it, thus if there's no thumbnail nothing gets deleted. 

I was wondering if this should be modified?

roberttidey, 2015-05-13:
The whole basis of the web side operation is based around the use of the thumbnails to navigate through to the capture files. This is used for previewing, downloading, organising time lapse and purging. It also allows capture files to be stored in sub-folders and still be found.

Without the thumbnails It would require a significant rework of all aspects of the web operation, probably using a back-end database to track the capture files.

The purpose of enabling / disabling thumbnail generation in the core raspimjpeg program was to facilitate its use outside of the web program and even mixed mode of operations where only some types of captures are managed by the web interface. For example, a user could turn off time-lapse thumbnail generation and place time-lapse captures in a separate path. Time-lapses would then be hidden from the web interface and the user would then be responsible for managing these files in some other way.

federicosalvi, 2015-05-13:
Understood. Maybe then just add a comment to warn the user in the raspimjpeg file? It took me a while to figure out what was going on.

closed 2015-05-13 by federicosalvi