silvanmelchior/RPi_Cam_Web_Interface #66: Annotation after reboot

Fritz4all, 2015-05-20:
I wanted to create some time-lapse series without annotations in them so i cleared the annotation text box like shown in the picture below : 
[image]

Everything worked like it should until a system reboot. 

When i reboot the system the annotation is back, but the annotation config field is still empty : 

[image]

Clearing the annotation text box even when it is already empty and pressing OK removes the annotations again.

My configuration : 
```
Raspberry Pi Model B (512MB)
 - latest official Rasbian image 
 - Kernel 3.18.11+ #781
 - All other software up to date.
 - RPi Cam Control v5.1.7
The Raspberry Pi camera module 1.3
```

@silvanmelchior great project.

bmedicke, 2015-05-20:
As a workaround you can paste a single `%` into the annotations box (or set it in uconfig).

Fritz4all, 2015-05-20:
@bmedicke tnx, a single ```%``` works.

roberttidey, 2015-05-20:
Thanks for posting the work-around. It looks like there is a an issue when raspimjpeg loads uconfig with totally blank fields. It is accepting the command when doing it live but the reload that occurs when it is restarted or after a bot causes the issue. I'll check into that.

roberttidey, 2015-05-24:
Fixed in latest version

closed 2015-05-24 by roberttidey