silvanmelchior/RPi_Cam_Web_Interface #85: trying to use raspimjpeg "as is" (i prefer separated devices for backend): errors

niegenug, 2015-05-29:
i guess the README 
https://github.com/silvanmelchior/userland/blob/master/host_applications/linux/apps/raspicam/README_RaspiMJPEG.md
is outdated, EVERY OTHER  command suggested there is broken (at least do not work for me even by creating fifo or cheating it with text files).

pi@rpicam ~ $ /opt/vc/bin/raspimjpeg -w 640 -h 360 -d 5 -q 70 -of ./image.jpg
start monitoring for pid: 3008
{2015/05/29 00:03:18} start monitoring for pid: 3008
RaspiMJPEG Version 5.2.1
{2015/05/29 00:03:18} RaspiMJPEG Version 5.2.1
MJPEG idle
{2015/05/29 00:03:18} MJPEG idle
Error: Could not open PIPE
{2015/05/29 00:03:18} Error: Could not open PIPE

My project requires a raspberry pi A(+) powered by solar panel and battery, gotta minimize power drain. It should just perform mjpeg stream, then a different device in touch with wifi will run motion and whatever is cpu intensive. I hope raspimjpeg could be my solution!

roberttidey, 2015-05-29:
The documentation is now maintained on the elinux wiki http://elinux.org/RPi-Cam-Web-Interface

It is normally used after running the RPi_Cam web install but can be used as a standalone executable providing some other actions are taken. I believe other users are indeed using it like this.

First you need to set up the command FIFO pipe yourself. raspimjpeg is configured to read commands from a pipe that is already set up (by default from /var/www/FIFO). Look at the RPI_Cam install script for details.

Second you want to use the config file /etc/raspimjpeg for controlling the set up not command line switches.

Third raspimjpeg controls the camera to provide 2 basic services; a medium resolution preview image cam.jpg continuously refreshed, and then video and image capture facilities controlled from the command pipe. The web side of RPi_Cam uses this to provide mjpeg streaming and web access to all the services.

Maybe the easiest way is to start with the standard web install and then strip it down to what you want for your project. For example, you can ignore the supplied overall web interface and just use the mjpeg stream web call. Similarly you can more easily send in remote pipe commands using the existing web to pipe support.

raspimjpeg has previously used 'motion' for its detection facilities but is now migrating to internal motion detection based on h264 vectors extracted directly from the camera. Either way you could choose to ignore motion detection altogether.

niegenug, 2015-05-29:
Thank you very much for answer and for the job made till now. I did already read the elinux wiki providing an overall understanding, but it really lack in-dept control informations (or i can't find?). I'll follow ur suggestion to read code and deduce raspimjpeg.conf options to get my project done.
Just a point i think would help:
I know this is "work in progress" and many things will change fast, that's why i would suggest to find a state where single elements get frozen to a "elastic" state. RPi_Cam works by combining at least 3 macro elements (raspimjpeg; www; motion detection), i think it would be effective to stay into "Keep It Simple" paradigm to easily combine each element in different manners, both for easy rework the current state (as my needs) or future revamps (as your possible needs)
Consider these points:
Why raspimjpeg lost command line functions for which code has already been written?
Why raspimjpeg MUST require a pipe? Shouldn't it be optional?
Why RPi_Cam shouldn't work out of the box splitted on different devices?
Why raspimjpeg.conf should contain motion directives?
{
 #
# Motion Detection
#
motion_external true
vector_preview false
vector_mode ?
motion_noise 20
motion_threshold 100
motion_image
motion_startframes 5
motion_stopframes 50
motion_pipe /var/www/FIFO1
motion_file 0
}

Someday you'll want to patch a broken code on just a single element and you will want to avoid a complete rewrite. 

Thank you again

roberttidey, 2015-05-29:
Thanks for your comments. This is open source project made available for raspberry camera users, so please feel free to fork, make changes, add pull requests etc on both the userland raspimjpeg, and the RPi_Cam projects. Note though that most users are using the RPI_Cam and so that will be the primary concern.

There is a fair bit of documentation on the wiki around overall operation, commands, naming schemes etc. The wiki itself is open to editing by anyone and we encourage people to add material and improve it further.

 Some of the command line options became redundant or would have needed extra work to support as the program developed, e.g. there used to be ones to set initial video and image indexes, but these are now calculated internally. Everything can be controlled via the config file.

raspimjpeg needs a mechanism to support dynamic commands for capturing and configuration control as it is intended as a service to be controlled by other applications. It is not a one shot application like for instance raspivid. A pipe was a simple standard mechanism for providing this type of control. The other main possibility would have been to have a built in web service and accept commands via an http interface but this is significantly more complex to develop. 

The Rpi_Cam overall app does work pretty much out of the box. That was one of the reasons why it became a popular app for camera use when silvan first published it. I'm not quite sure what work out of the box means for raspimjpeg. It is a component used within RPi_Cam but is made available for other users who wish to integrate into their own projects to match their own particular needs.

The motion settings in raspimjpeg are there to set up and control the internal motion detection built into raspimjpeg as opposed to the external 'motion' program which has its own completely separate configuration. Currently we are supporting both detection mechanisms, switchable from the commands / web interface. The external detection is what was used originally and the internal one is  what we are developing as it offers the potential of high class detection with a much lower overhead as it works from the vector motion information provided by the GPU.

niegenug, 2015-05-29:
ok, i got raspimjpeg to work (aka: just stream via apache) and run a standard 'motion' on netbook. It worked well but noticed cpu load on "raspberry A" around 50%.
A further research showed me a long raspivid record to load cpu around 20-30%, plus downloading the target file via apache more 20% => total load is 50% too. 
So my above opinions was overrated by myself, in fact the internal detection code isn't executed unless u state it on configuration (didn't try it though), and won't hurt any "improper" use of the raspimjpeg outside the rpi_cam.
My concerns was about a lacking brick in the camera software(mjpeg output) and your early job fit it very well. If u will go too far probably i'll have to fork, but as for now it is perfect. 

quote:
" the internal one is what we are developing as it offers the potential of high class detection with a much lower overhead as it works from the vector motion information provided by the GPU"

probably it wont fit my purposes, but it's a damn good project. Keep going! :)