silvanmelchior/RPi_Cam_Web_Interface #86: Annotation behavior broken, prepends string "/var/www/media/"

bp2008, 2015-05-29:
I downloaded your web interface v6.0.1 onto my new Pi 2 today.

I notice the default annotation string is getting "/var/www/media/" appended to the start.

[image]

That text even stays if I set the annotation string to be a single space character:

[image]

If I set an empty string (0 characters) as the annotation, I get "/var/www/media/" followed by some random garbage characters:

[image]

Just to confuse things, if I go from default annotation string to empty annotation string to 1-space-character annotation string, then the entire annotation disappears from the image:

[image]

roberttidey, 2015-05-29:
Yes, as noted in the forums I uploaded the wrong version. I had made a change so that media paths could be relative and the version I uploaded had the annotation using the same call when it should have used a lower level call without the prepend.

Corrected now.

bp2008, 2015-05-29:
Something is still wrong.  Now when I stop & start the service the annotation appears as "?ZZ??" and the annotation text field gets blanked out when I reload the page. If I enter anything into the annotation text field, the annotation just goes away and does not come back until I restart the service.

bp2008, 2015-05-29:
Managed to get it showing what I want by manually editing the **uconfig** file, but I think something is still fishy.

roberttidey, 2015-05-29:
I think there is an issue with a totally blank field now. A single space is OK. Other than that mine is working fine.
I'll check it further as the totally blank was working before the most recent changes.