silvanmelchior/RPi_Cam_Web_Interface #87: Attempting preview width 2592 delivers wrong image height

bp2008, 2015-05-29:
I was happy to discover that this web interface can deliver nearly full resolution jpeg images and mjpeg streams at high resolution, by setting the preview width to 2592.  However the delivered images are 2592x1936 instead of the full snapshot resolution 2592x1944.

Am I doing this wrong (is there another way to pull full resolution jpeg/mjpeg?), or where exactly does the bug lie?

roberttidey, 2015-05-29:
That maybe a rounding error somewhere. The preview width is set up and the height is calculated from it by scaling against the aspect ratio of the video height and width. I am slightly surprised that it works with a preview_width of 2592, Have you increased the GPU memory allocation? Is this on a Pi 2?

bp2008, 2015-05-29:
I have not increased the GPU memory allocation, but it is on a Pi 2.  I think the allocation is 128 MB.

roberttidey, 2015-05-29:
The 1932 is coming from the image resizer that creates the previews. This calculates like I said based on aspect ratio. The first part of this calculation does produce height 1944 but then this is made to be a multiple of 16 and that's what makes it 1932. The resizer is being done by the GPU and I'm not sure if the multiple of 16 is a requirement of that.

The Pi 2 is probably why it is working. I suspect the Pi B may have problems at this preview resolution.

bp2008, 2015-05-29:
Makes sense.