silvanmelchior/RPi_Cam_Web_Interface #94: RPi_Cam_Web_Interface_Installer.sh install / seems to fail

rmjspa, 2015-06-08:
./RPi_Cam_Web_Interface_Installer.sh: Zeile 271: Syntaxfehler beim unerwarteten Wort `fi'
./RPi_Cam_Web_Interface_Installer.sh: Zeile 271: `        fi'

Installer quits with error in Line 271 Syntaxerror unexpected word "fi"

roberttidey, 2015-06-09:
Thanks. I think that was caused when I commented out an unnecessary line and the if doesn't like being empty. Should be OK now

rmjspa, 2015-06-09:
Dear Robert,

Thank you for your quick response.
I still get errors during install. Sorry, it is in german. If you have any question, please don’t hesitate to ask.

regards, Ronald

pi@raspiTest ~ $ cd RPi_Cam_Web_Interface
pi@raspiTest ~/RPi_Cam_Web_Interface $ chmod u+x RPi_Cam_Web_Interface_Installer.sh
pi@raspiTest ~/RPi_Cam_Web_Interface $ ./RPi_Cam_Web_Interface_Installer.sh install
raspimjpeg: Kein Prozess gefunden
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
apache2 ist schon die neueste Version.
libapache2-mod-php5 ist schon die neueste Version.
Die folgenden zusätzlichen Pakete werden installiert:
  ffmpeg gpac-modules-base liba52-0.7.4 libav-tools libavdevice53 libavfilter2
  libavfilter3 libavformat53 libavresample1 libgpac2 libmysqlclient18
  libopencv-core2.3 libopencv-imgproc2.3 libpostproc52 libpq5 mysql-common
Vorgeschlagene Pakete:
  frei0r-plugins mysql-client postgresql-client
Die folgenden NEUEN Pakete werden installiert:
  ffmpeg gpac gpac-modules-base liba52-0.7.4 libav-tools libavdevice53
  libavfilter2 libavfilter3 libavformat53 libavresample1 libgpac2
  libmysqlclient18 libopencv-core2.3 libopencv-imgproc2.3 libpostproc52 libpq5
  motion mysql-common php5 zip
0 aktualisiert, 20 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 9.005 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 24,3 MB Plattenplatz zusätzlich benutzt.
Holen: 1 http://archive.raspberrypi.org/debian/ wheezy/main libavdevice53 armhf 6:9.14-1rpi1rpi1 [86,6 kB]
Holen: 2 http://archive.raspberrypi.org/debian/ wheezy/main libavresample1 armhf 6:9.14-1rpi1rpi1 [84,0 kB]
Holen: 3 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libgpac2 armhf 0.5.0~dfsg0-1 [1.438 kB]
Holen: 4 http://archive.raspberrypi.org/debian/ wheezy/main libavfilter3 armhf 6:9.14-1rpi1rpi1 [162 kB]
Holen: 5 http://archive.raspberrypi.org/debian/ wheezy/main libav-tools armhf 6:9.14-1rpi1rpi1 [3.198 kB]
Holen: 6 http://mirrordirector.raspbian.org/raspbian/ wheezy/main liba52-0.7.4 armhf 0.7.4-16 [31,4 kB]
Holen: 7 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libavformat53 armhf 6:0.8.17-1+rpi1 [413 kB]
Fehl http://mirrordirector.raspbian.org/raspbian/ wheezy/main php5 all 5.4.39-0+deb7u2
  404  Not Found
Holen: 8 http://mirrordirector.raspbian.org/raspbian/ wheezy/main gpac-modules-base armhf 0.5.0~dfsg0-1 [277 kB]
Holen: 9 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libopencv-core2.3 armhf 2.3.1-11 [707 kB]
Holen: 10 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libopencv-imgproc2.3 armhf 2.3.1-11 [672 kB]
Holen: 11 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libavfilter2 armhf 6:0.8.17-1+rpi1 [114 kB]
Holen: 12 http://mirrordirector.raspbian.org/raspbian/ wheezy/main mysql-common all 5.5.43-0+deb7u1 [73,4 kB]
Holen: 13 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libmysqlclient18 armhf 5.5.43-0+deb7u1 [618 kB]
Holen: 14 http://archive.raspberrypi.org/debian/ wheezy/main libpostproc52 armhf 6:0.git20120821-4rpi1 [29,5 kB]
Holen: 15 http://mirrordirector.raspbian.org/raspbian/ wheezy/main gpac armhf 0.5.0~dfsg0-1 [232 kB]
Holen: 16 http://mirrordirector.raspbian.org/raspbian/ wheezy/main libpq5 armhf 9.1.16-0+deb7u2 [125 kB]
Holen: 17 http://mirrordirector.raspbian.org/raspbian/ wheezy/main motion armhf 3.2.12-3.4 [259 kB]
Holen: 18 http://mirrordirector.raspbian.org/raspbian/ wheezy/main zip armhf 3.0-6 [334 kB]
Holen: 19 http://mirrordirector.raspbian.org/raspbian/ wheezy/main ffmpeg armhf 6:0.8.17-1+rpi1 [149 kB]
Es wurden 9.004 kB in 7 s geholt (1.160 kB/s).                                 
Fehlschlag beim Holen von http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5_5.4.39-0+deb7u2_all.deb  404  Not Found
E: Einige Archive konnten nicht heruntergeladen werden; vielleicht »apt-get update« ausführen oder mit »--fix-missing« probieren?
Where do you want to install? Please enter subfolder name or press enter for www-root.
rcam
"Install directory is /var/www/rcam"
Currently webserver is using port "80"
Do you want to change it? <y/n> n

cp: Aufruf von stat für „/etc/raspimjpeg“ nicht möglich: Datei oder Verzeichnis nicht gefunden
cp: reguläre Datei „/etc/motion/“ kann nicht angelegt werden: Ist kein Verzeichnis
chgrp: Zugriff auf „/etc/motion/motion.conf“ nicht möglich: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf „/etc/motion/motion.conf“ nicht möglich: Datei oder Verzeichnis nicht gefunden
Installer finished
pi@raspiTest ~/RPi_Cam_Web_Interface $ cd

Postanschrift:

Ronald Spahn

Rumpenheimer Str. 123 A

D-63075 Offenbach

Germany

Phone -49-69-175548470  

Fax	   -49-69-175548479

Mobile -49-172-6700764

Email: 

RONSPAHN@ICLOUD.COM

Skype Internet phone: rmjspa

miraaz, 2015-06-09:
You runned
./RPi_Cam_Web_Interface_Installer.sh update
first?
After that try again 
./RPi_Cam_Web_Interface_Installer.sh install

rmjspa, 2015-06-09:
I did 

./RPi_Cam_Web_Interface_Installer.sh remove

Which was not so good, because it removed many other Files, used by other application.

However it was on my test raspi, so not to much damage.

Then I did a fresh install of your software.

Still get this error: cp: Aufruf von stat für „/etc/raspimjpeg“ not posible: File or directory not found

I use two cam’s on two raspi since a while and hat no problem to install your software so far. The last time is about 2 weeks ago.

On this raspi’s I run also home automation software. The main part of this software looks like:

wheeze 05.05.2015

~ $ sudo apt-get update
~ $ sudo apt-get upgrade -y
~ $ sudo apt-get install git-core
~ $ sudo apt-get install php5-common php5-cgi php5-cli php-apc
~ $ sudo apt-get install apache2 libapache2-mod-php5

~ $ sudo apt-get install php5-dev
~ $ cd /opt
/opt $ sudo git clone https://github.com/phpredis/phpredis.git
/opt $ cd phpredis/
/opt/phpredis $ sudo phpize
/opt/phpredis $ sudo ./configure
/opt/phpredis $ sudo make && sudo make install
/opt/phpredis & sudo touch /etc/php5/conf.d/500-redis.ini
/opt/phpredis & sudo chmod 0777 /etc/php5/conf.d/500-redis.ini
/opt/phpredis & echo "extension=redis.so" >> /etc/php5/conf.d/500-redis.ini

cd /var/www
/var/www $ sudo git clone https://github.com/agent4788/SHC_Framework
/var/www $ sudo mv SHC_Framework/ shc/
/var/www $ sudo chmod 0777 -R shc/
/var/www $ cd shc/
/var/www/shc $ php shc_install.php
/var/www/shc $ sudo chmod 0777 -R ./

As I said before, I installed a couple times your software after the Home automation software without any problem. The last time was on May 26th.

Thank you for your attention.

Postanschrift:

Ronald Spahn

Rumpenheimer Str. 123 A

D-63075 Offenbach

Germany

Phone -49-69-175548470  

Fax	   -49-69-175548479

Mobile -49-172-6700764

Email: 

RONSPAHN@ICLOUD.COM

Skype Internet phone: rmjspa

roberttidey, 2015-06-09:
There was a mistake in the fix editing the apache config.

rmjspa, 2015-06-09:
Hello,

I got still one error on a fresh installation.

cp: Call of stat for „/etc/raspimjpeg“ not possible: File or Folder not found

Where do you want to install? Please enter subfolder name or press enter for www-root.
rcamTest
"Install directory is /var/www/rcamTest"
Currently webserver is using port "80"
Do you want to change it? <y/n> n

cp: Aufruf von stat für „/etc/raspimjpeg“ nicht möglich: Datei oder Verzeichnis nicht gefunden
Installer finished
pi@raspiTest ~/RPi_Cam_Web_Interface $ 

Postanschrift:

Ronald Spahn

Rumpenheimer Str. 123 A

D-63075 Offenbach

Germany

Phone -49-69-175548470  

Fax	   -49-69-175548479

Mobile -49-172-6700764

Email: 

RONSPAHN@ICLOUD.COM

Skype Internet phone: rmjspa

miraaz, 2015-06-19:
That fixed now if boss merge my patches. Please merge and close. 19.06.2015

closed 2015-06-19 by rmjspa