silvanmelchior/RPi_Cam_Web_Interface #98: Error in RaspiMJPEG popup when changing some settings.

Fritz4all, 2015-06-11:
Installed a new Raspberry Model B+ with a new Raspi-cam module powered with an 2A Adapter.

When installing the "Rpi_Cam_Web_Interface" like described on the Wiki: 

~~~ 
git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
chmod u+x RPi_Cam_Web_Interface_Installer.sh
./RPi_Cam_Web_Interface_Installer.sh install 
~~~
It will install, and work but Securing the page with username and password doesn't work.
adding "sudo" fixes the problem
~~~ 
git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
chmod u+x RPi_Cam_Web_Interface_Installer.sh
sudo ./RPi_Cam_Web_Interface_Installer.sh install 
~~~
Setting up autostart also results in errors : 
~~~
./RPi_Cam_Web_Interface_Installer.sh autostart_yes 
~~~
Same solution : 
~~~
sudo ./RPi_Cam_Web_Interface_Installer.sh autostart_yes
~~~

After this all seemed to work fine, but now when i go to "Download Videos and Images" and go back to the main screen or change "Camera Settings" and go back to the main screen, i get the following error : 

[image]

Tried it with Chrome, Firefox and Internet Explorer all give the same error.

After 10 complete system reboots and multiple: Install, Update and Start commands with : 
~~~ 
sudo ./RPi_Cam_Web_Interface_Installer.sh install
sudo ./RPi_Cam_Web_Interface_Installer.sh update
sudo ./RPi_Cam_Web_Interface_Installer.sh start
~~~

It now seems reasonably stable.

miraaz, 2015-06-11:
Thank you. I look what is wrong with security. I tested that using root and thats why bug not jumped out. I try with different user.

miraaz, 2015-06-15:
Fritz4all is it fixed now?

Fritz4all, 2015-06-16:
The installer is fixed, tnx.

But i'm still getting the following errors when strait after logon : 

[image]

[image]

When i enable the checkbox  : Prevent this page from....
the error will not popup until i close the browser and reopen/relogon again.

Fritz4all, 2015-06-16:
Solved the "Error in RaspiMJPEG" by removing the software, the a reboot and installing it again.

Tnx :+1:

closed 2015-06-16 by Fritz4all

miraaz, 2015-06-16:
Fine you shorted it out :)

JsBergbau, 2021-07-09:
If you stumble on this via Google: Please check that you have correctly attached the camera. Cable had no proper connection and also the error 
```
Error in RaspiMJPEG: 
Restart RaspiMJPEG (./RPi_Cam_Web_Interface_Installer.sh start) or the whole RPi.
```
occured.