silvanmelchior/RPi_Cam_Web_Interface #131: RPi Cam taking over the webserver

mosambers, 2015-06-30:
Hi,

i have installed the RPi Cam Web Interface and everything is working fine. It's installed in /var/www.
But I'm also running a Pi-Control web interface which is in the directory /var/www/pic.  

My problem is that I can't reach the directory pic where my control interface is located. I installed the RPi cam web interface in a custom directory, but it was still only reachable about the ip of the raspberry.

I would like to reach the RPi Cam web interface about /ip of raspberry/rpi. That means in the subfolder rpi.

What can I do?

Thank you

roberttidey, 2015-06-30:
You can install the webcam in a subfolder under www but we currently take over the DocumentRoot of whatever port chosen for the camera app. The intention here was that different web apps would use different ports. E.g. host:8080, host:8090 etc.

So if you are wanting to use the same port for both camera and your other apps and access them by including the subfolder in the url like host/cam host/pic then that is possible but the config needs to be adjusted a bit. In particular the DocumentRoot setting needs to be reset back to /var/www. The other change that is necessary is to then include the cam subfolder in the netcam_url setting in the motion.conf set up so that motion can get its feed of data correctly.

The setup did actually used to be like this and I am thinking of reverting it back.

miraaz, 2015-06-30:
Yes. I also agree revert it. People not like current situation like we see. I can program universal solution and let people choose what they want but easier if we revert it first.

roberttidey, 2015-06-30:
I started to do that but then I saw there was a conflict with the webport stuff. Have you got a version of that can include rpicamdir

mosambers, 2015-06-30:
Thank you. I'm new to the PI. For a beginner, like me, it is a little bit difficult.

Then i will change the ports for different apps. I think that is the easier way.

roberttidey, 2015-06-30:
I think we will be going ahead with this change but you can still use different ports as well.

miraaz, 2015-07-01:
roberttidey i have green light revert that weebroot stuff?

roberttidey, 2015-07-01:
Yes.

Also I did a quick test of the current installer and I couldn't seem to get the port into the motion.conf.

miraaz, 2015-07-01:
Ok i start testing and lure first bugs out! Tnx man.

miraaz, 2015-07-01:
roberttidey
Not understand where is problem!
I uninstalled all including packages and made new install. Install time i changed port 8080 to 8090 and i see in motion.conf now:
netcam_url http://localhost:8090/cam_pic.php
Its ok i think. I looking wrong place? 
Please help me out so i can see problem and fix it.

I make one reinstall more and delete config.txt also that time. Lets look where is that buggy!

roberttidey, 2015-07-01:
I retested the port piece again and it seems to be OK. Sorry for any confusion.