silvanmelchior/RPi_Cam_Web_Interface #144: Motion does not work

mosambers, 2015-07-06:
Hello, it's me again. I have two issues concerning the motion detection. 

1. After changing the video resolution the whole web interface gives an error to start RPi again. After a restart the new resolution is saved but motion doesn't work anymore. Somehow it can't handle the higher res. After changing back to the default res. motion works again.

After that I have installed the Web_Interface again. And everything worked perfect.

2. After a few days of running the RPi, I see motion doesn't work again. It should record but it doesn't. Everything is on default settings. I have no explanation.

Thank you.

roberttidey, 2015-07-06:
I'm not aware of any motion problem specific to resolution unless you increase the preview size substantially which can increase the CPU load. After you re-installed did you still have a resolution problem? Was it repeatable?

If motion is stopping after a few days then one should first check the schedule log to see exactly what is happening? It is worth looking at the trouble-shooting section on the eLinux wiki which has a step by step guide on checking motion detection out.

Also unreliability can be caused by less than perfect power supply arrangements particularly as the camera draws a bit more current. You need to be sure that the PS can supply the load and that the voltage drop in the power cable is small.

mosambers, 2015-07-10:
I couldn't fix it, but the internal motion detection works.