silvanmelchior/RPi_Cam_Web_Interface #153: Support for Jessie

bmedicke, 2015-07-14:
In Jessie www-data's shell switched to `/usr/sbin/nologin` which breaks the camera stream. I fixed it with `chsh www-data -s /bin/sh` but I don't think that's a clean fix.
As an aside: the new TUI is nice, but it makes it harder to catch the messages when running the debug mode.

Jessie also want's you to use  `/var/www` instead of `/var/www/html`. At least for Apache. Entering `html` as the subdirectory when prompted by the installer works.

Another Apache change is that `default.conf` has been renamed to `000-default.conf` which causes awk to fail. Again, it's kind of hard to read the error messages with the new TUI. Maybe the messages can be logged somewhere (if they are not already) or debug mode can use the old CLI.

roberttidey, 2015-07-14:
Thanks. I have not tried Jessie.
Messages are logged by default to the webfolder schedulerLog.txt This actually collects log messages from both the web side and raspimjpeg. It can be viewed from web under scheduler settings. Log messages from raspimjpeg have curly braces round time stamp to distinguish. One can set the log_file config in /etc/raspimjpeg to empty in which case raspimjpeg will revert to logging to stdout.

The other items are primarily install issues by the sound of it. We did put a change in the boot start up rc.local to deal with non-debian start up shell issues. That was looking for /etc/debian_version and changing the start up accordingly. To make the installer Jessie aware for sub-folder default and Apache we would need to have a Jessie detect mechanism.

miraaz, 2015-07-15:
Sorry my uncompetence but what is Jessie? If i even do not know what is it, then impossible think how support it...

miraaz, 2015-07-15:
is it Debian Jessie? Or some hardware?

miraaz, 2015-07-15:
#154 Jessie also wants you to use /var/www/html instead of /var/www. At least for Apache. Entering html as the subdirectory when prompted by the installer works.

That works like RedHat! There www root set /var/www/html

bmedicke, 2015-07-16:
Another small problem that I just encountered is that `avconv` was missing. Installing `libav-tools` fixed it. As far as I can tell the package name has not changed since Wheezy. 

I am using a minimal net install on all my Pis ([raspbian-ua-netinst](https://github.com/debian-pi/raspbian-ua-netinst)) so I'm not sure if it's installed by default or pulled in as a dependency. 

I did not know about schedulerLog.txt, that's very useful. Is there a log for the installation itself? Failing awk/sed/apt-get messages, etc. I'm going to set up a new Pi to take a look at what else might be failing. (If not I'm just going to run the install in tmux and scroll up, that should work too.)

`cat /etc/debian_version` prints `8.0` for Jessie.

roberttidey, 2015-07-16:
It should be harmless for us to just include libav-tools anyway. Raspbian will effectively just ignore it.

The installer doesn't log although that could be useful in future. Originally the installer was fairly minimal and required the user to patch up config files if not wanting defaults (ports, security etc). miraaz has done a lot to add that in.

Currently standard raspbian says 7.8 for debian_version. I'm not sure how those numbers work and whether raspbian becomes 8.0 in the future and Jessie moves on.