silvanmelchior/RPi_Cam_Web_Interface #169: Timelapse: return to normal framerate on motion detected?

watou, 2015-07-19:
Would this option be possible, especially with the newer GPU-based motion detection?

roberttidey, 2015-07-19:
Could you explain a bit more? I am not sure from the title what you want to achieve.

watou, 2015-07-19:
Let's say I'm recording one timelapse frame every 3 seconds.  Then, motion is detected, thereby starting to record what I've set for normal recording (say 25 frames per second).  When motion is no longer detected, return to capturing one frame every 3 seconds.  [This video](https://youtu.be/gyeTsl8ghcQ?t=47s) gives an example.  Notice the rate of the clouds moving when there is vs. isn't motion detected.

roberttidey, 2015-07-19:
OK I understand now. You have a time lapse going, then a motion triggers a video capture interrupting the time lapse operation. You want it to resume the time lapse operation after the video capture is completed.

In principle, that is do-able. Let me have a think about it.

watou, 2015-07-19:
Thanks, Robert!  If I could set this to run like the Youtube example shows and end up producing a single video, I could have an entire day's interesting events in one concise package.  Thanks for considering it.