silvanmelchior/RPi_Cam_Web_Interface #172: Missing extrastyle.css in the /css folder

juggledad, 2015-07-22:
Line 130 of motion.php is
      <link rel="stylesheet" href="css/extrastyle.css" />
however there is no 'extra style.css there. There is a extrastyle.txt that contains 'es_Default.css'. Becasue of the missing file, an error is showwing in the error.log:
[Wed Jul 22 10:46:48 2015] [error] [client 192.168.48.99] File does not exist: /var/www/picam/css/extrastyle.css, referer: http://192.168.48.190/picam/motion.php

juggledad, 2015-07-22:
Also an issue on 
line 311 of preview.php and
line 312 of schedule.php

roberttidey, 2015-07-22:
The extrastyle,css is for user overrides of styling. User css can be put in there and then selected from main page. This is how the night mode is done. 

We changed how that was selected on index page but not on schedule and preview.  I'll fix that.

Thanks