silvanmelchior/RPi_Cam_Web_Interface #173: index.php (line 191) using wrong 'src=...' reference form causing 'file does not exist' message

juggledad, 2015-07-22:
an error message is showing in the error.log:
[Wed Jul 22 10:46:49 2015] [error] [client 192.168.48.99] File does not exist: /var/www/loading.jpg, referer: http://192.168.48.190/picam/

This is caused because line 191 of index.php references  the jpg using:
   src="/loading.jpg"
when it should be
  src="./loading.jpg"
(note the period before the slash)

roberttidey, 2015-07-22:
Thanks. Fixed now.

juggledad, 2015-07-23:
btw - I couldn't get anything in teh web brwser to work till I ran it in debug mode.

roberttidey, 2015-07-23:
That's a bit more unusual. Those errors are normally benign. What browser are you using? Maybe it has some stricter options on that type of error.

juggledad, 2015-07-23:
Safari, FireFox, Chrome - none of them will take changes. For example if I change the Annotation and press 'Ok' nothing changes - run in debug and what ever I put in there changes.

Shall I make this a seperate issue? Any debugging info you want
running on Pi2 model B

juggledad, 2015-07-23:
I just ran it in FireFox with FireBug and when I press the OK on annotation I see
GET http://192.168.48.190:8080/cmd_pipe.php?cmd=an%20hello
'hello' is the text in teh annotation

roberttidey, 2015-07-23:
That sounds like the FIFO pipe input into raspimjpeg is not working. That would also mean that although the live preview is updating the main buttons to capture an image or start and stop video would also not work.

There was a version of raspimjpeg up for about 12 hours a couple of days ago that could get into the state where the FIFO wasn't working. I thought I reverted that back out and the current one should be OK. If you got the bad one then update and re-install. I am also checking the current version just in case the revert wasn't successful.

juggledad, 2015-07-23:
that was it. I reinstalled and now it works. Thanks