silvanmelchior/RPi_Cam_Web_Interface #180: Internal Motion

noquarter12, 2015-07-30:
Is there any other Documentation on the internal Motion feature? Like any more info on the setting and perhaps why the noise level is set to 1010 default when on the web page it says 0-255 range.

I was having more success with the internal then the external motion. The internal motion would at least start detecting motion and take a image but having trouble getting it to stop detecting motion.

roberttidey, 2015-07-30:
The wiki has some more info. There are actually two algorithms for the internal detect. The first did have a noise level of 0-255. When the second method was added 'noise level' was the selector and anything above 999 meant use second method and controlled the frame averaging used by the second method. So 1010 actually means rolling average over 10 frames.

As the second method is better and will be enhanced more then we will drop the first method and rationalise the parameters.

To make it stop quicker reduce the still frame to stop parameter.

It is also worth putting %f %c at the front of annotation string whilst setting this stuff up. The first number is cumulative frame count to trigger and the second is the vector sum towards the threshold.

noquarter12, 2015-08-01:
OK Cool, thanks for the help. After playing around with the threshold and the noise values I can get it to start and stop capture. Verified because I think it sends send smd 1 to the log when it detects and send smd 0 with motion is stopped. 

How ever I am now having trouble with the schedule. I have it set to Record 30 sec of video when motion is detected. When the scheduler is stopped and restarted this works. However it only works once. The next time motion is detected the following is shown in the log. 

Any thoughts?

{2015/07/31 23:57:52} Internal motion detection started
{2015/07/31 23:57:52} recommended video buffers 1
{2015/07/31 23:57:52} h264 buffers set to recommended 1
{2015/07/31 23:57:52} Set up internal detect width=121 height=68
{2015/07/31 23:58:24} send smd 1
[2015/07/31 23:58:24] Start capture requested from Pipe
[2015/07/31 23:58:24] Send ca 1 30
{2015/07/31 23:58:24} Capturing 30 seconds
{2015/07/31 23:58:24} Capturing started
{2015/07/31 23:58:54} Stopping video from timer
{2015/07/31 23:58:54} Capturing stopped
{2015/07/31 23:58:54} Add /var/www/media/vi_0000_20150731_235824.mp4 to Box Queue at pos 0
{2015/07/31 23:58:55} Start boxing /var/www/media/vi_0000_20150731_235824.mp4.h264 to /var/www/media/vi_0000_20150731_235824.mp4 Queue pos 0
{2015/07/31 23:59:17} Finished boxing /var/www/media/vi_0000_20150731_235824.mp4 from Box Queue at pos 0
{2015/07/31 23:59:53} Change: motion_noise = 1020
{2015/07/31 23:59:57} Change: motion_noise = 1020
{2015/08/01 00:00:24} Change: motion_threshold = 400
{2015/08/01 00:00:35} Change: motion_threshold = 500
{2015/08/01 00:00:39} Internal motion detection stopped
{2015/08/01 00:00:39} recommended video buffers 1
{2015/08/01 00:00:39} h264 buffers set to recommended 1
{2015/08/01 00:00:39} Set up internal detect width=121 height=68
{2015/08/01 00:00:41} Internal motion detection started
{2015/08/01 00:00:41} recommended video buffers 1
{2015/08/01 00:00:41} h264 buffers set to recommended 1
{2015/08/01 00:00:42} Set up internal detect width=121 height=68
{2015/08/01 00:01:01} send smd 1
[2015/08/01 00:01:01] Start capture request ignored, already started
{2015/08/01 00:01:08} send smd 0
[2015/08/01 00:01:08] Stop capture requested

roberttidey, 2015-08-01:
Ok. That is a sort of a bug triggered by your use of ca 1 30 in the start command and I'm not seeing a motion stop smd 0 from the first one.

Scheduler keeps track of normal motion start and stop triggers and will ignore a start if already in a started state. Similarly it will ignore a stop if already stopped. By using the ca 1 30 to do a 30 second recording and not getting a stop request then the scheduler thinks it is still in a started state even after the 30 second has elapsed. 

I'll see if that can be improved but it's a little tricky as scheduler is just sending the ca 1 30 command to raspimjpeg and doesn't really know when it has stopped.

Another way of achieving fixed 30 second recordings would be to set Max_Capture to 30 seconds and set motion stop to do nothing. That way starts would initiate recordings, scheduler would stop them after 30 seconds but would know they had stopped so would be ready for the next start.

roberttidey, 2015-08-01:
I have moved the ignore start when already started from scheduler to raspimjpeg. This means that a new start command will be handled OK if the video has stopped (by timed recording) and no intermediate stop command has been issued by the scheduler.