silvanmelchior/RPi_Cam_Web_Interface #182: No unattended installation possible anymore due to heavy usage of dialogs

konstantinj, 2015-08-09:
Hi there,

I wanted to create a docker image for the rpicam. Unfortunately an unattended installation is not possible anymore since you're using dialog quite heavily and it's not possible to bypass that with command line options. Only possibility would be using except but that's quite unfortunate. Could you add support for an unattended installation again?

roberttidey, 2015-08-10:
miraaz has done quite a bit of work making it more 'friendly' for the average interactive install as before it required quite a bit of post-install work if wanting to customise the install a bit. An unattended install option was going to be added at the end of this work but hasn't happened yet.

I'll have a look to see if that can be done sooner.

miraaz, 2015-08-10:
Yea sorry. I have currently very little time because i changed my work in real life... I need now learn new work. So i away little time.

miraaz, 2015-08-10:
I look on it when i find little more time...

konstantinj, 2015-08-10:
Hi, no problem. It's working for me with expect. Unfortunately expect is not running in Docker because of the dumb term. But I won't give up. Since docker is working great on rpi now I think that might me appreciated by others as well.

konstantinj, 2017-05-30:
Hi, finally "some" month later I have a working docker version now: https://github.com/konstantinj/RPi_Cam_Web_Interface

Just wanted to let you know.

closed 2017-05-30 by konstantinj