silvanmelchior/RPi_Cam_Web_Interface #184: webpage not always showing live stream

sh2515, 2015-08-17:
Is anyone else having the issue of the live stream not showing on the webpage all the time? I have it running on a model A and the ram left is 15-20mb, due to 3 second buffer and half memory on gpu and it is not always showing, unfortunately I can't figure out if it is when it is boxing in the background or not yet. However, if I pick up the cam.jpg image from the web directory (/var/www/cam.jpg) it is still running.
Anyone elses experience?

roberttidey, 2015-08-18:
Model A memory is quite tight. One can reduce GPU memory a bit to say 112MB to give a bit more headroom, particularly if using less than maximum resolution video. 

You should be able to see in the log under scheduler the boxing operation.

If the cam.jpg is updating then it should be updating live preview. Try both Default or MJPEG streaming options selected (main page - system) to see if that makes a difference.

sh2515, 2015-08-18:
Current I have "scheduled management tasks" as the last log when the live view isn't streaming to webpage.

roberttidey, 2015-08-18:
You say it is not showing all the time. Does it start showing and then stop after a while?

If the cam.jpg is updating then the issue is likely to be on the web side. You can turn on development tools in most browsers to see what is going on. For example in Chrome the network tab in development will show the cam.jpg being continuously fetched.