silvanmelchior/RPi_Cam_Web_Interface #185: Video not at correct speed

sh2515, 2015-08-17:
upon motion detection the video records, however on playback it shows a speeded up version.  I have a model A with 3 second buffer.

roberttidey, 2015-08-18:
There are two fps settings; one for the video recording and one for the boxing process. Normally and by default they should be the same (e.g. 25 fps). The boxing fps controls the playback speed so if different then the recording can be slower or faster. Some use this to record at a low fps (e.g. 5fps) so they can playback fast to check more quickly.

sh2515, 2015-08-18:
Speeds are set at the same fps.

roberttidey, 2015-08-18:
Try downloading a video and playing locally in a media player. This will help determine if your problem is with the recording or the playback.

What browser are you using?

yannickasselin, 2022-06-14:
I have the same issue. I am running RPI-CAM-WEB-INTERFACE on a Raspberry Pi Compute Module 4. With the recording fps set at 25, I need to set the boxing at 21 to get the right playback speed or else it will play faster than it should. What could cause this?

yannickasselin, 2022-06-15:
I tried the %k trick. I found some weird behavior. It sometimes seem to skip a frame but then, 1 or 2 frames later, I get the same frame number twice. So it seems to be an issue with the frame numbers not being updated correctly.

But still, I noticed that I never get the last 2 or 3 frames. I never get passed the frame 22. So it seems to be skipping frames.

I will try to find how to do the boxing mode setting you talked about.