silvanmelchior/RPi_Cam_Web_Interface #187: Stability issues in 6.0.3

a-l-e-x-d-s-9, 2015-08-21:
I upgraded from 6.0.2 to 6.0.3. 
1. Motion detection completely unfunctional, maybe related to: https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/186
2. I couldn't record video, it just didn't work at all.
3. The video preview was very laggy and froze after couple minutes. 
4. Taking pictures, made the web interface to blink like it the button was pressed over and over, and hundreds of pictures appeared in library. Even if I close the web page and reopen it, it still seem to take more pictures. After uninstall of 6.0.3 and reinstall 6.0.2, the problem still exist. But it worked ok before I installed 6.0.3.

roberttidey, 2015-08-21:
Strange, I am running 6.0.3 without a problem and the only change was to add a video start macro facility in. It shouldn't have affected the basic operation. Can you check log, particularly around the multiple picture issue? What sort of raspberry is this?

a-l-e-x-d-s-9, 2015-08-22:
I use: Raspberry Pi 2 Model B V1.1 2014 (tested on 2 Raspberry Pi with 2 separate cameras)
I've tested the 6.0.2 in the last 24H, with motion detection, it worked smoothly.
Now I installed 6.0.4, that's what I experience now:

1) I can't reproduce the problem I had with capturing pictures anymore, it seems to work fine now.

2) Camera preview seems to work fine now, and it doesn't freeze after short time.

3)  Motion detection mode causing RaspiMJPEG error after a few seconds: https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/188

4) Video recording not working. Only a thumbnail appears with status "Busy":
[image]

That's what I have in the  /var/www/media/ folder:
pi@raspberrypi ~ $ ls -la /var/www/media/
total 2884
drwxr-xr-x 2 www-data www-data    4096 Aug 22 18:53 .
drwxr-xr-x 7 www-data www-data    4096 Aug 22 17:47 ..
-rw-r--r-- 1 www-data www-data 2364410 Aug 22 18:04 im_0000_20150822_180431.jpg
-rw-r--r-- 1 www-data www-data   49114 Aug 22 18:04 im_0000_20150822_180431.jpg.i0000.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 17:48 vi_0000_20150822_174826.mp4.h264
-rw-r--r-- 1 www-data www-data   34244 Aug 22 17:48 vi_0000_20150822_174826.mp4.v0000.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 17:50 vi_0001_20150822_175007.mp4.h264
-rw-r--r-- 1 www-data www-data   35993 Aug 22 17:50 vi_0001_20150822_175007.mp4.v0001.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 17:50 vi_0002_20150822_175058.mp4.h264
-rw-r--r-- 1 www-data www-data   48628 Aug 22 17:50 vi_0002_20150822_175058.mp4.v0002.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 17:55 vi_0003_20150822_175541.mp4.h264
-rw-r--r-- 1 www-data www-data   51028 Aug 22 17:55 vi_0003_20150822_175541.mp4.v0003.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:03 vi_0004_20150822_180353.mp4.h264
-rw-r--r-- 1 www-data www-data   49196 Aug 22 18:03 vi_0004_20150822_180353.mp4.v0004.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:04 vi_0005_20150822_180423.mp4.h264
-rw-r--r-- 1 www-data www-data   48810 Aug 22 18:04 vi_0005_20150822_180423.mp4.v0005.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:05 vi_0006_20150822_180514.mp4.h264
-rw-r--r-- 1 www-data www-data   49019 Aug 22 18:05 vi_0006_20150822_180514.mp4.v0006.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:05 vi_0007_20150822_180544.mp4.h264
-rw-r--r-- 1 www-data www-data   44138 Aug 22 18:05 vi_0007_20150822_180544.mp4.v0007.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:44 vi_0008_20150822_184429.mp4.h264
-rw-r--r-- 1 www-data www-data   41150 Aug 22 18:44 vi_0008_20150822_184429.mp4.v0008.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:45 vi_0009_20150822_184526.mp4.h264
-rw-r--r-- 1 www-data www-data   46919 Aug 22 18:45 vi_0009_20150822_184526.mp4.v0009.th.jpg
-rw-r--r-- 1 www-data www-data       0 Aug 22 18:53 vi_0010_20150822_185327.mp4.h264
-rw-r--r-- 1 www-data www-data   57507 Aug 22 18:53 vi_0010_20150822_185327.mp4.v0010.th.jpg

a-l-e-x-d-s-9, 2015-08-22:
There was a missing line in my /opt/vc/bin/raspimjpeg: "start_vid &start_vid.sh".
Everything is working ok now.
Thank you very much for your help. :)

closed 2015-08-22 by a-l-e-x-d-s-9