silvanmelchior/RPi_Cam_Web_Interface #188: Error in RaspiMJPEG in 6.0.2 keep appearing

a-l-e-x-d-s-9, 2015-08-21:
In motion detection mode, maximum that I could run the system without this error is 24H. Most of the times it happens after just a few hours. 
Anything I can to understand what's wrong?
Maybe somehow it possible to recover from this error, with automatic restart of RaspiMJPEG?

[image]

roberttidey, 2015-08-21:
Anything appearing in log (under scheduler) when that happens?

roberttidey, 2015-08-21:
Another thing to check is the ribbon camera connections and power supply. The camera does take quite a bit more current and if the voltage at the raspberry is not maintained above 4.75V then that can give stability issues. Note that it is often the USB cable that causes a problem even with a good PSU. With 1A going down the cable it is easy to drop too much voltage. It needs to be short and low resistance.

a-l-e-x-d-s-9, 2015-08-22:
1) The 6.0.2 version is much more stable than 6.0.4. 
I just tested 6.0.2 with motion detection for 24H and it worked fine, 6.0.4 shows RaspiMJPEG error after a few seconds of run.

2) I connected the device through usb voltmeter: it shows stable 5.14~5.15V with 0.23~0.056A whole time.
[image]
I haven't changed the ribbon cable connector. 
My PSU is 30W 6A with only one Raspberry Pi connected to it.
[image]
Because of  6.0.2 smooth run for 24H, and almost immediate failure of 6.0.4, I assume it's not electrical problem, but software issue.

3) I'm not familiar with the schedule log, so I'm not sure if there is something wrong there. It's attached below.

4) After occurrence of RaspiMJPEG error, I keep getting this message in terminal:
  mmal: mmal_vc_component_enable: failed to enable component: ENOSPC

scheduleLog.txt:

{2015/08/22 17:17:28} RaspiMJPEG Version 5.2.6
{2015/08/22 17:17:29} recommended video buffers 1
{2015/08/22 17:17:29} h264 buffers set to recommended 1
{2015/08/22 17:17:29} MJPEG streaming, ready to receive commands
{2015/08/22 17:17:29} send smd 9
[2015/08/22 17:17:30] RaspiCam support started
[2015/08/22 17:17:30] Capture Pipe already exists /var/www/FIFO1
[2015/08/22 17:17:30] Scheduler loop is started
[2015/08/22 17:17:30] Reload parameters command requested
[2015/08/22 17:17:30] Scheduler loop is started
[2015/08/22 17:17:31] New period detected 0
[2015/08/22 17:17:31] Scheduled management tasks. Next at 1440256651
{2015/08/22 17:45:47} External motion detection started
{2015/08/22 17:45:47} recommended video buffers 1
{2015/08/22 17:45:47} h264 buffers set to recommended 1
{2015/08/22 17:46:45} SIGINT/SIGTERM received, stopping
{2015/08/22 17:47:15} RaspiMJPEG Version 5.2.6
{2015/08/22 17:47:15} recommended video buffers 1
{2015/08/22 17:47:15} h264 buffers set to recommended 1
{2015/08/22 17:47:15} MJPEG streaming, ready to receive commands
{2015/08/22 17:47:15} send smd 9
[2015/08/22 17:47:16] RaspiCam support started
[2015/08/22 17:47:16] Capture Pipe already exists /var/www/FIFO1
[2015/08/22 17:47:16] Scheduler loop is started
[2015/08/22 17:47:16] Reload parameters command requested
[2015/08/22 17:47:16] Scheduler loop is started
[2015/08/22 17:47:17] Stop capture request ignored, already stopped
[2015/08/22 17:47:17] New period detected 0
[2015/08/22 17:47:17] Scheduled management tasks. Next at 1440258437
{2015/08/22 17:47:19} External motion detection started
{2015/08/22 17:47:19} recommended video buffers 1
{2015/08/22 17:47:19} h264 buffers set to recommended 1
[2015/08/22 17:48:26] Start capture requested from Pipe
[2015/08/22 17:48:26] Send ca 1
{2015/08/22 17:48:31} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:31} Error: Could not enable camera
{2015/08/22 17:48:36} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:36} Error: Could not enable camera
[2015/08/22 17:48:37] Stop capture requested
[2015/08/22 17:48:37] Send ca 0
{2015/08/22 17:48:41} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:41} Error: Could not enable camera
{2015/08/22 17:48:46} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:46} Error: Could not enable camera
{2015/08/22 17:48:51} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:51} Error: Could not enable camera
{2015/08/22 17:48:56} RaspiMJPEG Version 5.2.6
{2015/08/22 17:48:56} Error: Could not enable camera
{2015/08/22 17:49:01} RaspiMJPEG Version 5.2.6
{2015/08/22 17:49:01} Error: Could not enable camera
{2015/08/22 17:49:51} RaspiMJPEG Version 5.2.6
{2015/08/22 17:49:51} recommended video buffers 1
{2015/08/22 17:49:51} h264 buffers set to recommended 1
[... 347 lines cut ...]
{2015/08/22 18:05:49} Error: Could not enable camera
{2015/08/22 18:05:54} RaspiMJPEG Version 5.2.6
{2015/08/22 18:05:54} Error: Could not enable camera
[2015/08/22 18:05:58] Stop capture requested
[2015/08/22 18:05:58] Send ca 0
{2015/08/22 18:05:59} RaspiMJPEG Version 5.2.6
{2015/08/22 18:05:59} Error: Could not enable camera
{2015/08/22 18:06:04} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:04} Error: Could not enable camera
{2015/08/22 18:06:09} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:09} Error: Could not enable camera
{2015/08/22 18:06:14} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:14} Error: Could not enable camera
{2015/08/22 18:06:19} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:19} Error: Could not enable camera
{2015/08/22 18:06:24} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:24} Error: Could not enable camera
{2015/08/22 18:06:29} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:29} Error: Could not enable camera
{2015/08/22 18:06:34} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:34} Error: Could not enable camera
{2015/08/22 18:06:39} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:39} Error: Could not enable camera
{2015/08/22 18:06:44} RaspiMJPEG Version 5.2.6
{2015/08/22 18:06:44} Error: Could not enable camera

a-l-e-x-d-s-9, 2015-08-22:
I use: Raspberry Pi 2 Model B V1.1 2014.
I have 2 Raspberry Pi, with 2 separate camera connected to them.
Both of them show RaspiMJPEG error, in motion mode with 6.0.4 in just a few seconds of run.
So I think that it's unlikely to be problem related to Raspberry Pi, ribbon cable or camera.
Although I haven't tested the PSU with scope, given the results from USB voltmeter, and reasonable stability of 6.0.2, I think that it's unlikely to be PSU or USB cable related issue.

roberttidey, 2015-08-22:
The media folder has .h264 files but of 0 length so it creates the file but nothing gets streamed into it.

That is consistent with the log as it errors with cannot enable camera so can't set up the stream video connection for some reason. It errors out because of that but then a watchdog restarts it 5 seconds later but the cycle then repeats.

It is definitely not a solid fault with 6.0.4 as I have that on 3 cameras (2 B's and a 2) running OK. So we need to find why it doesn't work on your systems.

When working correctly the scheduler log should look like this (start up and 1 video capture)
{2015/08/22 18:52:33} RaspiMJPEG Version 5.2.6
{2015/08/22 18:52:33} recommended video buffers 1
{2015/08/22 18:52:33} h264 buffers set to recommended 1
{2015/08/22 18:52:33} Set up internal detect width=82 height=61
{2015/08/22 18:52:33} MJPEG streaming, ready to receive commands
{2015/08/22 18:52:33} send smd 9
[2015/08/22 18:52:34] RaspiCam support started
[2015/08/22 18:52:34] Capture Pipe already exists /var/www/FIFO1
[2015/08/22 18:52:34] Scheduler loop is started
[2015/08/22 18:52:34] Reload parameters command requested
[2015/08/22 18:52:34] Scheduler loop is started
[2015/08/22 18:52:35] New period detected 0
[2015/08/22 18:52:35] Scheduled management tasks. Next at 1440269555
{2015/08/22 18:53:11} Executing macro /var/www/macros/start_vid.sh "/var/www/media/vi_0021_20150822_185311.mp4" &
{2015/08/22 18:53:11} Capturing started
{2015/08/22 18:53:15} Capturing stopped
{2015/08/22 18:53:15} Add /var/www/media/vi_0021_20150822_185311.mp4 to Box Queue at pos 0
{2015/08/22 18:53:16} Start boxing /var/www/media/vi_0021_20150822_185311.mp4.h264 to /var/www/media/vi_0021_20150822_185311.mp4 Queue pos 0
{2015/08/22 18:53:17} Finished boxing /var/www/media/vi_0021_20150822_185311.mp4 from Box Queue at pos 0

Three things to check / try.

a) What kernel version are your Pi's running? A command uname -a will give that

b) If you get the raspimjpeg binary from 4.0.2 then you can directly replace that in a 4.0.4 install by
1) Stop the system ( ./RPi_Cam_Web_Interface_Installer.sh stop ) in the install folder
2) sudo cp pathto4.0.2/raspimjpeg /opt/vc/bin/raspimjpeg
3) Start the system ( ./RPi_Cam_Web_Interface_Installer.sh start ) 
This would tell us for certain whether it is a 4.0.2 / difference causing the problem

c) How much GPU memory is allocated  (vcgencmd get_mem gpu) ?

a-l-e-x-d-s-9, 2015-08-22:
a) uname -a:
Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux

c) vcgencmd get_mem gpu:
gpu=128M

b) I copied raspimjpeg 5.2.4 from 6.0.2 folder, and motion detection working fine with it.
Start of log is 6.0.4 & RaspiMJPEG 5.2.6 and than 6.0.4 & RaspiMJPEG 5.2.4:

{2015/08/22 21:51:31} RaspiMJPEG Version 5.2.6
{2015/08/22 21:51:31} recommended video buffers 1
{2015/08/22 21:51:31} h264 buffers set to recommended 1
{2015/08/22 21:51:31} MJPEG streaming, ready to receive commands
{2015/08/22 21:51:31} send smd 9
[2015/08/22 21:51:32] RaspiCam support started
[2015/08/22 21:51:32] Capture Pipe already exists /var/www/FIFO1
[2015/08/22 21:51:32] Scheduler loop is started
[2015/08/22 21:51:32] Reload parameters command requested
[2015/08/22 21:51:32] Scheduler loop is started
[2015/08/22 21:51:33] New period detected 0
[2015/08/22 21:51:33] Scheduled management tasks. Next at 1440273093
{2015/08/22 21:51:35} External motion detection started
{2015/08/22 21:51:35} recommended video buffers 1
{2015/08/22 21:51:35} h264 buffers set to recommended 1
[2015/08/22 21:53:17] Start capture requested from Pipe
[2015/08/22 21:53:17] Send ca 1
[2015/08/22 21:53:20] Stop capture requested
[2015/08/22 21:53:20] Send ca 0
{2015/08/22 21:53:22} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:22} Error: Could not enable camera
{2015/08/22 21:53:27} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:27} Error: Could not enable camera
{2015/08/22 21:53:32} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:32} Error: Could not enable camera
{2015/08/22 21:53:37} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:37} Error: Could not enable camera
{2015/08/22 21:53:42} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:42} Error: Could not enable camera
{2015/08/22 21:53:47} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:47} Error: Could not enable camera
{2015/08/22 21:53:52} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:52} Error: Could not enable camera
{2015/08/22 21:53:58} RaspiMJPEG Version 5.2.6
{2015/08/22 21:53:58} Error: Could not enable camera
{2015/08/22 21:54:03} RaspiMJPEG Version 5.2.6
{2015/08/22 21:54:03} Error: Could not enable camera
{2015/08/22 21:54:08} RaspiMJPEG Version 5.2.6
{2015/08/22 21:54:08} Error: Could not enable camera
{2015/08/22 21:54:13} RaspiMJPEG Version 5.2.6
{2015/08/22 21:54:13} Error: Could not enable camera
{2015/08/22 22:00:40} RaspiMJPEG Version 5.2.4
{2015/08/22 22:00:41} recommended video buffers 1
{2015/08/22 22:00:41} h264 buffers set to recommended 1
{2015/08/22 22:00:41} MJPEG streaming, ready to receive commands
{2015/08/22 22:00:41} send smd 9
[2015/08/22 22:00:42] RaspiCam support started
[2015/08/22 22:00:42] Capture Pipe already exists /var/www/FIFO1
[2015/08/22 22:00:42] Scheduler loop is started
[2015/08/22 22:00:42] Reload parameters command requested
[2015/08/22 22:00:42] Scheduler loop is started
[2015/08/22 22:00:42] Stop capture request ignored, already stopped
[2015/08/22 22:00:43] New period detected 0
[2015/08/22 22:00:43] Scheduled management tasks. Next at 1440273643
{2015/08/22 22:00:44} External motion detection started
{2015/08/22 22:00:44} recommended video buffers 1
{2015/08/22 22:00:44} h264 buffers set to recommended 1
[2015/08/22 22:01:21] Start capture requested from Pipe
[2015/08/22 22:01:21] Send ca 1
{2015/08/22 22:01:22} Capturing started
[2015/08/22 22:01:24] Stop capture requested
[2015/08/22 22:01:24] Send ca 0
{2015/08/22 22:01:24} Capturing stopped
{2015/08/22 22:01:24} Add /var/www/media/vi_0010_20150822_220122.mp4 to Box Queue at pos 0
{2015/08/22 22:01:25} Start boxing /var/www/media/vi_0010_20150822_220122.mp4.h264 to /var/www/media/vi_0010_20150822_220122.mp4 Queue pos 0
{2015/08/22 22:01:26} Finished boxing /var/www/media/vi_0010_20150822_220122.mp4 from Box Queue at pos 0

a-l-e-x-d-s-9, 2015-08-22:
I use the configuration file from 6.0.2 & raspimjpeg 5.2.4, possibly some inconsistency between raspimjpeg 5.2.4 and 5.2.6 in configurations?

roberttidey, 2015-08-22:
a) I am using 4.0.8-v7+ #805 at the moment. I haven't done an update recently but will try that out.

b) The config file may explain it. 6.0.4 add as start macro setting start_vid into the config file. It is possible that if that is missing then it is throwing an error.

a-l-e-x-d-s-9, 2015-08-22:
Indeed, there was a missing line in my /opt/vc/bin/raspimjpeg: "start_vid &start_vid.sh".
Everything is working ok now.
Thank you very much for your help. :)

roberttidey, 2015-08-23:
Good to hear it's working OK. 

I have made a small change to avoid the crash if those definitions are missing.

a-l-e-x-d-s-9, 2015-08-23:
Unfortunately, from my last test on 6.0.4 & raspimjpeg 5.2.6, after 14 hours of working motion detection, I got the same RaspiMJPEG error:
[image]

I tested it with increased 196MB GPU memory, and it didn't help.
So it makes problematic the whole motion detection feature or general stability for long runs. Maybe it's possible to add recovery step? In case of error, it will be printed to log file, but it will also restart raspimjpeg, or it can restart the whole software. Other option is to add a field to web interface, user could write a bash command that will be executed on occurrence of this error, instead of popup error message. Possibly with: "RPi_Cam_Web_Interface_Installer.sh start". So the system would be reliable for long runs. 

scheduleLog.txt just before the error: 

{2015/08/23 13:19:14} Add /var/www/media/vi_0502_20150823_131907.mp4 to Box Queue at pos 30
{2015/08/23 13:19:14} Start boxing /var/www/media/vi_0502_20150823_131907.mp4.h264 to /var/www/media/vi_0502_20150823_131907.mp4 Queue pos 30
{2015/08/23 13:19:15} Finished boxing /var/www/media/vi_0502_20150823_131907.mp4 from Box Queue at pos 30
[2015/08/23 13:19:48] Start capture requested from Pipe
[2015/08/23 13:19:48] Send ca 1
{2015/08/23 13:19:48} Capturing started
[2015/08/23 13:19:54] Stop capture requested
[2015/08/23 13:19:54] Send ca 0
{2015/08/23 13:19:54} Capturing stopped
{2015/08/23 13:19:54} Add /var/www/media/vi_0503_20150823_131948.mp4 to Box Queue at pos 31
{2015/08/23 13:19:54} Start boxing /var/www/media/vi_0503_20150823_131948.mp4.h264 to /var/www/media/vi_0503_20150823_131948.mp4 Queue pos 31
{2015/08/23 13:19:55} Finished boxing /var/www/media/vi_0503_20150823_131948.mp4 from Box Queue at pos 31
[2015/08/23 13:22:21] Start capture requested from Pipe
[2015/08/23 13:22:21] Send ca 1
{2015/08/23 13:22:21} Capturing started
[2015/08/23 13:22:25] Stop capture requested
[2015/08/23 13:22:25] Send ca 0
{2015/08/23 13:22:25} Error: Could not send buffers to port
{2015/08/23 13:22:30} RaspiMJPEG Version 5.2.6
{2015/08/23 13:22:30} Error: Could not enable camera
{2015/08/23 13:22:35} RaspiMJPEG Version 5.2.6
{2015/08/23 13:22:35} Error: Could not enable camera
{2015/08/23 13:22:40} RaspiMJPEG Version 5.2.6
{2015/08/23 13:22:40} Error: Could not enable camera
{2015/08/23 13:22:45} RaspiMJPEG Version 5.2.6
{2015/08/23 13:22:45} Error: Could not enable camera
{2015/08/23 13:22:50} RaspiMJPEG Version 5.2.6

roberttidey, 2015-08-24:
This is a slightly different error. Could not send buffers to port indicates a breakdown in the camera interface. I'll keep checking to see if I can find what might cause that. My 3 cameras seem rock solid and never error even after very extended use.

In the meantime I have added a couple of macro hooks (error_hard and error_soft). These call error_hard.sh and error_soft.sh in the macros folder if present. The hard error is the most problematic one as that tends to go with a camera interface breakdown, I have put an example macro in the folder which just does a reboot. The examples do not have execution rights as installed. This is deliberate to make it a user action to activate these. So one needs to do a chmod to turn a macro on.

a-l-e-x-d-s-9, 2015-08-24:
Thanks for the macro hooks, they will be very useful for me if the error continue appearing, and a good workaround for all users, in case of future bugs...
Are there some debug flags that I can turn on in my system, or something else I can do to understand what's causing this error?

roberttidey, 2015-08-24:
It is a call to the camera interface that is failing so there is not a lot of debug that can be added. However, there are 3 places where this error could occur so I've changed it to identify that which will help narrow it down.

a-l-e-x-d-s-9, 2015-08-24:
I updated the sw, and after a few hours, the same error. :(

scheduleLog.txt:

{2015/08/24 19:39:04} Capturing started
{2015/08/24 19:39:04} Finished boxing /var/www/media/vi_1518_20150824_193855.mp4 from Box Queue at pos 29
[2015/08/24 19:39:20] Stop capture requested
[2015/08/24 19:39:20] Send ca 0
{2015/08/24 19:39:20} Capturing stopped
{2015/08/24 19:39:20} Add /var/www/media/vi_1519_20150824_193904.mp4 to Box Queue at pos 30
{2015/08/24 19:39:21} Start boxing /var/www/media/vi_1519_20150824_193904.mp4.h264 to /var/www/media/vi_1519_20150824_193904.mp4 Queue pos 30
{2015/08/24 19:39:23} Finished boxing /var/www/media/vi_1519_20150824_193904.mp4 from Box Queue at pos 30
[2015/08/24 19:39:31] Start capture requested from Pipe
[2015/08/24 19:39:31] Send ca 1
{2015/08/24 19:39:31} Capturing started
[2015/08/24 19:39:34] Stop capture requested
[2015/08/24 19:39:34] Send ca 0
{2015/08/24 19:39:34} Error: Could not send buffers to port
{2015/08/24 19:39:34} Executing macro /var/www/macros/error_hard.sh "Could not send buffers to port"  
{2015/08/24 19:39:39} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:39} Error: Could not enable camera
{2015/08/24 19:39:39} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:44} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:44} Error: Could not enable camera
{2015/08/24 19:39:44} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:49} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:49} Error: Could not enable camera
{2015/08/24 19:39:49} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:54} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:54} Error: Could not enable camera
{2015/08/24 19:39:54} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:59} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:59} Error: Could not enable camera
{2015/08/24 19:39:59} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:04} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:04} Error: Could not enable camera
{2015/08/24 19:40:04} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:09} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:09} Error: Could not enable camera
{2015/08/24 19:40:09} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:14} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:14} Error: Could not enable camera
{2015/08/24 19:40:14} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:19} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:19} Error: Could not enable camera
{2015/08/24 19:40:19} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:24} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:24} Error: Could not enable camera
{2015/08/24 19:40:24} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"

a-l-e-x-d-s-9, 2015-08-24:
After the error, I tried just to run: ./RPi_Cam_Web_Interface_Installer.sh start
It worked for 10 minutes, and then the same error appeared.

Here is the log:

{2015/08/24 19:39:02} Add /var/www/media/vi_1518_20150824_193855.mp4 to Box Queue at pos 29
{2015/08/24 19:39:02} Start boxing /var/www/media/vi_1518_20150824_193855.mp4.h264 to /var/www/media/vi_1518_20150824_193855.mp4 Queue pos 29
[2015/08/24 19:39:04] Start capture requested from Pipe
[2015/08/24 19:39:04] Send ca 1
{2015/08/24 19:39:04} Capturing started
{2015/08/24 19:39:04} Finished boxing /var/www/media/vi_1518_20150824_193855.mp4 from Box Queue at pos 29
[2015/08/24 19:39:20] Stop capture requested
[2015/08/24 19:39:20] Send ca 0
{2015/08/24 19:39:20} Capturing stopped
{2015/08/24 19:39:20} Add /var/www/media/vi_1519_20150824_193904.mp4 to Box Queue at pos 30
{2015/08/24 19:39:21} Start boxing /var/www/media/vi_1519_20150824_193904.mp4.h264 to /var/www/media/vi_1519_20150824_193904.mp4 Queue pos 30
{2015/08/24 19:39:23} Finished boxing /var/www/media/vi_1519_20150824_193904.mp4 from Box Queue at pos 30
[2015/08/24 19:39:31] Start capture requested from Pipe
[2015/08/24 19:39:31] Send ca 1
{2015/08/24 19:39:31} Capturing started
[2015/08/24 19:39:34] Stop capture requested
[2015/08/24 19:39:34] Send ca 0
{2015/08/24 19:39:34} Error: Could not send buffers to port
{2015/08/24 19:39:34} Executing macro /var/www/macros/error_hard.sh "Could not send buffers to port"  
{2015/08/24 19:39:39} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:39} Error: Could not enable camera
{2015/08/24 19:39:39} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:44} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:44} Error: Could not enable camera
{2015/08/24 19:39:44} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:49} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:49} Error: Could not enable camera
{2015/08/24 19:39:49} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:54} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:54} Error: Could not enable camera
{2015/08/24 19:39:54} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:39:59} RaspiMJPEG Version 5.2.8
{2015/08/24 19:39:59} Error: Could not enable camera
{2015/08/24 19:39:59} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:04} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:04} Error: Could not enable camera
{2015/08/24 19:40:04} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:09} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:09} Error: Could not enable camera
{2015/08/24 19:40:09} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:14} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:14} Error: Could not enable camera
{2015/08/24 19:40:14} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:19} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:19} Error: Could not enable camera
{2015/08/24 19:40:19} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:24} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:24} Error: Could not enable camera
{2015/08/24 19:40:24} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:29} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:30} Error: Could not enable camera
{2015/08/24 19:40:30} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:35} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:35} Error: Could not enable camera
{2015/08/24 19:40:35} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:40} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:40} Error: Could not enable camera
{2015/08/24 19:40:40} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:45} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:45} Error: Could not enable camera
{2015/08/24 19:40:45} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:50} RaspiMJPEG Version 5.2.8
{2015/08/24 19:40:50} Error: Could not enable camera
{2015/08/24 19:40:50} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 19:40:55} RaspiMJPEG Version 5.2.8
[... 2931 lines cut ...]
{2015/08/24 21:09:37} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:09:42} RaspiMJPEG Version 5.2.8
{2015/08/24 21:09:42} Error: Could not enable camera
{2015/08/24 21:09:42} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:09:47} RaspiMJPEG Version 5.2.8
{2015/08/24 21:09:47} Error: Could not enable camera
{2015/08/24 21:09:47} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:09:52} RaspiMJPEG Version 5.2.8
{2015/08/24 21:09:52} Error: Could not enable camera
{2015/08/24 21:09:52} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:09:57} RaspiMJPEG Version 5.2.8
{2015/08/24 21:09:57} Error: Could not enable camera
{2015/08/24 21:09:57} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:10:02} RaspiMJPEG Version 5.2.8
{2015/08/24 21:10:02} Error: Could not enable camera
[... cut ...]
{2015/08/24 21:10:07} RaspiMJPEG Version 5.2.8
{2015/08/24 21:10:07} Error: Could not enable camera
{2015/08/24 21:10:07} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:10:12} RaspiMJPEG Version 5.2.8
{2015/08/24 21:10:12} Error: Could not enable camera
{2015/08/24 21:10:12} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/24 21:10:17} RaspiMJPEG Version 5.2.8
{2015/08/24 21:10:17} Error: Could not enable camera
{2015/08/24

roberttidey, 2015-08-25:
The software version 5.2.9 should have different messages to "Could not send buffers to port" to help distinguish which of the three places had the problem.

The common thread seems to be it gets that error immediately after a ca 0 which is the stopping video recording command. It is possible there could be a conflict if a video buffer callback occurs during a critical part of that sequence. I'll check further on that. Why it particularly affects your system I don't know.

a-l-e-x-d-s-9, 2015-08-25:
I installed RaspiMJPEG 5.2.9, received the error after one hour.

scheduleLog.txt:

{2015/08/25 13:34:45} RaspiMJPEG Version 5.2.9
{2015/08/25 13:34:45} recommended video buffers 1
{2015/08/25 13:34:45} h264 buffers set to recommended 1
{2015/08/25 13:34:45} MJPEG streaming, ready to receive commands
{2015/08/25 13:34:45} send smd 9
[2015/08/25 13:34:46] RaspiCam support started
[2015/08/25 13:34:46] Capture Pipe already exists /var/www/FIFO1
[2015/08/25 13:34:46] Scheduler loop is started
[2015/08/25 13:34:47] Reload parameters command requested
[2015/08/25 13:34:47] Scheduler loop is started
[2015/08/25 13:34:48] New period detected 0
[2015/08/25 13:34:48] Scheduled management tasks. Next at 1440502488
{2015/08/25 13:35:02} External motion detection started
{2015/08/25 13:35:02} recommended video buffers 1
{2015/08/25 13:35:02} h264 buffers set to recommended 1
{2015/08/25 13:39:34} External motion detection stopped
[2015/08/25 13:39:34] Stop capture request ignored, already stopped
{2015/08/25 13:39:34} recommended video buffers 1
{2015/08/25 13:39:34} h264 buffers set to recommended 1
{2015/08/25 13:39:35} Stream halted
{2015/08/25 13:40:31} SIGINT/SIGTERM received, stopping
{2015/08/25 13:40:57} RaspiMJPEG Version 5.2.9
{2015/08/25 13:40:58} recommended video buffers 1
{2015/08/25 13:40:58} h264 buffers set to recommended 1
{2015/08/25 13:40:58} MJPEG streaming, ready to receive commands
{2015/08/25 13:40:58} send smd 9
[2015/08/25 13:40:59] RaspiCam support started
[2015/08/25 13:40:59] Capture Pipe already exists /var/www/FIFO1
[2015/08/25 13:40:59] Scheduler loop is started
[2015/08/25 13:40:59] Reload parameters command requested
[2015/08/25 13:40:59] Scheduler loop is started
[2015/08/25 13:41:00] New period detected 0
[2015/08/25 13:41:00] Scheduled management tasks. Next at 1440502860
{2015/08/25 13:41:14} External motion detection started
{2015/08/25 13:41:14} recommended video buffers 1
{2015/08/25 13:41:14} h264 buffers set to recommended 1
{2015/08/25 13:42:35} External motion detection stopped
{2015/08/25 13:42:36} recommended video buffers 1
{2015/08/25 13:42:36} h264 buffers set to recommended 1
[2015/08/25 13:42:36] Stop capture request ignored, already stopped
{2015/08/25 13:42:37} External motion detection started
{2015/08/25 13:42:38} recommended video buffers 1
{2015/08/25 13:42:38} h264 buffers set to recommended 1
[2015/08/25 13:42:57] Stop capture request ignored, already stopped
{2015/08/25 13:44:20} External motion detection stopped
{2015/08/25 13:44:20} recommended video buffers 1
{2015/08/25 13:44:20} h264 buffers set to recommended 1
[2015/08/25 13:44:20] Stop capture request ignored, already stopped
{2015/08/25 13:44:22} Stream halted
{2015/08/25 13:44:23} recommended video buffers 1
{2015/08/25 13:44:23} h264 buffers set to recommended 1
{2015/08/25 13:44:23} Stream continued
{2015/08/25 13:44:25} External motion detection started
{2015/08/25 13:44:25} recommended video buffers 1
{2015/08/25 13:44:25} h264 buffers set to recommended 1
{2015/08/25 13:54:06} External motion detection stopped
[2015/08/25 13:54:06] Stop capture request ignored, already stopped
{2015/08/25 13:54:06} recommended video buffers 1
{2015/08/25 13:54:06} h264 buffers set to recommended 1
{2015/08/25 13:54:09} Capturing image
{2015/08/25 13:54:17} Capturing started
{2015/08/25 13:54:24} Capturing stopped
{2015/08/25 13:54:24} Add /var/www/media/vi_1780_20150825_135417.mp4 to Box Queue at pos 0
{2015/08/25 13:54:24} Start boxing /var/www/media/vi_1780_20150825_135417.mp4.h264 to /var/www/media/vi_1780_20150825_135417.mp4 Queue pos 0
{2015/08/25 13:54:25} Finished boxing /var/www/media/vi_1780_20150825_135417.mp4 from Box Queue at pos 0
{2015/08/25 13:54:58} External motion detection started
[... 1304 lines cut ...]
{2015/08/25 14:48:33} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:33} Error: Could not enable camera
{2015/08/25 14:48:33} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:48:38} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:38} Error: Could not enable camera
{2015/08/25 14:48:38} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:48:43} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:43} Error: Could not enable camera
{2015/08/25 14:48:43} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:48:48} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:48} Error: Could not enable camera
{2015/08/25 14:48:48} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:48:53} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:53} Error: Could not enable camera
{2015/08/25 14:48:53} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:48:58} RaspiMJPEG Version 5.2.9
{2015/08/25 14:48:58} Error: Could not enable camera
{2015/08/25 14:48:58} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:49:03} RaspiMJPEG Version 5.2.9
{2015/08/25 14:49:04} Error: Could not enable camera
{2015/08/25 14:49:04} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:49:09} RaspiMJPEG Version 5.2.9
{2015/08/25 14:49:09} Error: Could not enable camera
{2015/08/25 14:49:09} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 14:49:14} RaspiMJPEG Version 5.2.9

roberttidey, 2015-08-25:
Thanks. That confirms my suspicion that this is being caused by a conflict between the buffer stream callback and the stop recording operation.

a-l-e-x-d-s-9, 2015-08-25:
I think that the new prints to log, causing the error to appear much faster.

Here is a log from 2 more runs:

{2015/08/25 16:16:40} RaspiMJPEG Version 5.2.9
{2015/08/25 16:16:40} Error: Could not enable camera
{2015/08/25 16:16:40} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 16:17:10} RaspiMJPEG Version 5.2.9
{2015/08/25 16:17:10} recommended video buffers 1
{2015/08/25 16:17:10} h264 buffers set to recommended 1
{2015/08/25 16:17:10} MJPEG streaming, ready to receive commands
{2015/08/25 16:17:10} send smd 9
[2015/08/25 16:17:11] RaspiCam support started
[2015/08/25 16:17:11] Capture Pipe already exists /var/www/FIFO1
[2015/08/25 16:17:11] Scheduler loop is started
[2015/08/25 16:17:11] Reload parameters command requested
[2015/08/25 16:17:11] Scheduler loop is started
[2015/08/25 16:17:12] New period detected 0
[2015/08/25 16:17:12] Scheduled management tasks. Next at 1440512232
{2015/08/25 16:17:13} External motion detection started
{2015/08/25 16:17:14} recommended video buffers 1
{2015/08/25 16:17:14} h264 buffers set to recommended 1
[2015/08/25 16:17:26] Start capture requested from Pipe
[2015/08/25 16:17:26] Send ca 1
{2015/08/25 16:17:26} Capturing started
[2015/08/25 16:17:29] Stop capture requested
[2015/08/25 16:17:29] Send ca 0
{2015/08/25 16:17:29} Capturing stopped
{2015/08/25 16:17:29} Add /var/www/media/vi_1856_20150825_161726.mp4 to Box Queue at pos 0
{2015/08/25 16:17:30} Start boxing /var/www/media/vi_1856_20150825_161726.mp4.h264 to /var/www/media/vi_1856_20150825_161726.mp4 Queue pos 0
{2015/08/25 16:17:31} Finished boxing /var/www/media/vi_1856_20150825_161726.mp4 from Box Queue at pos 0
[2015/08/25 16:17:35] Start capture requested from Pipe
[2015/08/25 16:17:35] Send ca 1
{2015/08/25 16:17:35} Capturing started
[2015/08/25 16:17:49] Stop capture requested
[2015/08/25 16:17:49] Send ca 0
{2015/08/25 16:17:49} Capturing stopped
{2015/08/25 16:17:49} Add /var/www/media/vi_1857_20150825_161735.mp4 to Box Queue at pos 1
{2015/08/25 16:17:49} Start boxing /var/www/media/vi_1857_20150825_161735.mp4.h264 to /var/www/media/vi_1857_20150825_161735.mp4 Queue pos 1
{2015/08/25 16:17:51} Finished boxing /var/www/media/vi_1857_20150825_161735.mp4 from Box Queue at pos 1
[2015/08/25 16:18:50] Start capture requested from Pipe
[2015/08/25 16:18:50] Send ca 1
{2015/08/25 16:18:50} Capturing started
[2015/08/25 16:18:57] Stop capture requested
[2015/08/25 16:18:57] Send ca 0
{2015/08/25 16:18:57} Capturing stopped
{2015/08/25 16:18:57} Add /var/www/media/vi_1858_20150825_161850.mp4 to Box Queue at pos 2
{2015/08/25 16:18:58} Start boxing /var/www/media/vi_1858_20150825_161850.mp4.h264 to /var/www/media/vi_1858_20150825_161850.mp4 Queue pos 2
{2015/08/25 16:18:59} Finished boxing /var/www/media/vi_1858_20150825_161850.mp4 from Box Queue at pos 2
[2015/08/25 16:19:07] Start capture requested from Pipe
[2015/08/25 16:19:07] Send ca 1
{2015/08/25 16:19:07} Capturing started
[2015/08/25 16:19:10] Stop capture requested
[2015/08/25 16:19:10] Send ca 0
{2015/08/25 16:19:10} Capturing stopped
{2015/08/25 16:19:10} Add /var/www/media/vi_1859_20150825_161907.mp4 to Box Queue at pos 3
{2015/08/25 16:19:10} Start boxing /var/www/media/vi_1859_20150825_161907.mp4.h264 to /var/www/media/vi_1859_20150825_161907.mp4 Queue pos 3
{2015/08/25 16:19:11} Finished boxing /var/www/media/vi_1859_20150825_161907.mp4 from Box Queue at pos 3
[2015/08/25 16:19:13] Start capture requested from Pipe
[2015/08/25 16:19:13] Send ca 1
{2015/08/25 16:19:13} Capturing started
[2015/08/25 16:19:16] Stop capture requested
[2015/08/25 16:19:16] Send ca 0
{2015/08/25 16:19:16} Capturing stopped
{2015/08/25 16:19:16} Add /var/www/media/vi_1860_20150825_161913.mp4 to Box Queue at pos 4
{2015/08/25 16:19:16} Start boxing /var/www/media/vi_1860_20150825_161913.mp4.h264 to /var/www/media/vi_1860_20150825_161913.mp4 Queue pos 4
{2015/08/25 16:19:17} Finished boxing /var/www/media/vi_1860_20150825_161913.mp4 from Box Queue at pos 4
[2015/08/25 16:19:26] Start capture requested from Pipe
[2015/08/25 16:19:26] Send ca 1
{2015/08/25 16:19:26} Capturing started
[... 1729 lines cut ...]
{2015/08/25 17:43:37} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:43:42} RaspiMJPEG Version 5.2.9
{2015/08/25 17:43:42} Error: Could not enable camera
{2015/08/25 17:43:42} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:43:47} RaspiMJPEG Version 5.2.9
{2015/08/25 17:43:47} Error: Could not enable camera
{2015/08/25 17:43:47} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:43:52} RaspiMJPEG Version 5.2.9
{2015/08/25 17:43:53} Error: Could not enable camera
{2015/08/25 17:43:53} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:43:58} RaspiMJPEG Version 5.2.9
{2015/08/25 17:43:58} Error: Could not enable camera
{2015/08/25 17:43:58} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:44:03} RaspiMJPEG Version 5.2.9
{2015/08/25 17:44:03} Error: Could not enable camera
{2015/08/25 17:44:03} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:44:08} RaspiMJPEG Version 5.2.9
{2015/08/25 17:44:08} Error: Could not enable camera
{2015/08/25 17:44:08} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:44:13} RaspiMJPEG Version 5.2.9
{2015/08/25 17:44:13} Error: Could not enable camera
{2015/08/25 17:44:13} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"  
{2015/08/25 17:44:18} RaspiMJPEG Version 5.2.9
{2015/08/25 17:44:18} Error: Could not enable camera
{2015/08/25 17:44:18} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"

a-l-e-x-d-s-9, 2015-08-26:
By the way, even if I make the script: error_hard.sh executable, it won't be executed on error. 
Despite: "Executing macro /var/www/macros/error_hard.sh "Could not enable camera" ".
I tried to append text with error_hard.sh, to some log file, and to reboot the system, but it's doing nothing.

roberttidey, 2015-08-26:
I'm fairly certain it is attempting to execute that script if it logs that line and I checked general macro execution which uses exactly the same routine.. The script can fail to execute if the ownership or permissions are not right. It should be owned by www-data (change with chown if not) or it doesn't have execute permissions. It can also fail if there is something wrong with the script. If you have edited the script on a PC for example make sure that it only has LF  endings not CRLF.

There is a script there called test.sh which just writes a date stamp to the testmacro.txt file in the /var/www folder. You can rename that for testing. It might be worth testing by calling it say end_vid.sh then it will get called at the end of each recording.

a-l-e-x-d-s-9, 2015-08-26:
You are right, error_hard.sh indeed executed, but www-data user doesn't have a root permissions, so he can't execute: "sudo reboot" as in your error_hard.sh example, but he can run "sudo shutdown -r now".

roberttidey, 2015-08-26:
Yes, the web interface uses the shutdown command rather than reboot and now I check I can see the install (sudoers.d) gives special permission for www-data to use the shutdown command but not reboot. 

I have changed the example error_hard.sh script.

Thanks

a-l-e-x-d-s-9, 2015-09-05:
In the last 10 days, I run the system with automatic reboot on HW error, and it happens pretty often, especially when there is a frequent activity. I'm pretty sure that last version increases the chance of HW error very drastically.
 
Here is a reboots log caused by HW errors on my system:
Wed Aug 26 16:08:31 IDT 2015
Wed Aug 26 16:23:20 IDT 2015
Wed Aug 26 21:25:48 IDT 2015
Wed Aug 26 23:46:41 IDT 2015
Thu Aug 27 11:11:51 IDT 2015
Thu Aug 27 15:06:48 IDT 2015
Thu Aug 27 19:49:23 IDT 2015
Thu Aug 27 20:50:49 IDT 2015
Thu Aug 27 22:32:15 IDT 2015
Thu Aug 27 23:47:45 IDT 2015
Fri Aug 28 00:53:21 IDT 2015
Fri Aug 28 00:57:48 IDT 2015
Fri Aug 28 01:04:22 IDT 2015
Fri Aug 28 02:28:22 IDT 2015
Fri Aug 28 03:19:57 IDT 2015
Fri Aug 28 03:43:03 IDT 2015
Fri Aug 28 03:53:13 IDT 2015
Fri Aug 28 06:36:50 IDT 2015
Fri Aug 28 14:05:42 IDT 2015
Fri Aug 28 16:29:46 IDT 2015
Fri Aug 28 17:37:57 IDT 2015
Fri Aug 28 18:36:37 IDT 2015
Sat Aug 29 03:24:31 IDT 2015
Sat Aug 29 08:54:16 IDT 2015
Sat Aug 29 13:26:43 IDT 2015
Sat Aug 29 13:56:35 IDT 2015
Sat Aug 29 14:25:04 IDT 2015
Sat Aug 29 15:09:28 IDT 2015
Sat Aug 29 16:21:58 IDT 2015
Sat Aug 29 16:49:55 IDT 2015
Sat Aug 29 17:04:40 IDT 2015
Sat Aug 29 18:09:40 IDT 2015
Sat Aug 29 18:18:05 IDT 2015
Sat Aug 29 19:21:51 IDT 2015
Sat Aug 29 19:42:36 IDT 2015
Sat Aug 29 23:14:58 IDT 2015
Sun Aug 30 01:51:07 IDT 2015
Sun Aug 30 03:28:48 IDT 2015
Sun Aug 30 21:22:47 IDT 2015
Sun Aug 30 21:47:15 IDT 2015
Sun Aug 30 21:51:20 IDT 2015
Sun Aug 30 22:43:45 IDT 2015
Sun Aug 30 23:20:17 IDT 2015
Sun Aug 30 23:32:24 IDT 2015
Mon Aug 31 00:43:53 IDT 2015
Mon Aug 31 14:03:05 IDT 2015
Mon Aug 31 14:23:44 IDT 2015
Mon Aug 31 16:22:15 IDT 2015
Mon Aug 31 19:40:57 IDT 2015
Mon Aug 31 19:54:35 IDT 2015
Mon Aug 31 20:37:42 IDT 2015
Mon Aug 31 20:49:32 IDT 2015
Mon Aug 31 20:56:30 IDT 2015
Mon Aug 31 22:17:41 IDT 2015
Mon Aug 31 22:57:13 IDT 2015
Tue Sep  1 06:43:44 IDT 2015
Tue Sep  1 18:39:46 IDT 2015
Tue Sep  1 20:30:31 IDT 2015
Tue Sep  1 23:28:34 IDT 2015
Wed Sep  2 02:41:06 IDT 2015
Wed Sep  2 05:02:26 IDT 2015
Wed Sep  2 06:35:11 IDT 2015
Wed Sep  2 07:00:32 IDT 2015
Thu Sep  3 03:23:28 IDT 2015
Thu Sep  3 03:29:05 IDT 2015
Thu Sep  3 15:33:07 IDT 2015
Thu Sep  3 20:43:58 IDT 2015
Thu Sep  3 21:37:32 IDT 2015
Thu Sep  3 21:49:30 IDT 2015
Fri Sep  4 01:09:46 IDT 2015
Fri Sep  4 01:57:11 IDT 2015
Fri Sep  4 02:15:25 IDT 2015
Fri Sep  4 02:21:59 IDT 2015
Fri Sep  4 03:11:38 IDT 2015
Fri Sep  4 03:26:31 IDT 2015
Fri Sep  4 06:58:51 IDT 2015
Sat Sep  5 14:23:25 IDT 2015
Sat Sep  5 14:52:16 IDT 2015
Sat Sep  5 16:00:21 IDT 2015
Sat Sep  5 17:10:38 IDT 2015
Sat Sep  5 19:03:55 IDT 2015
Sat Sep  5 19:27:38 IDT 2015
Sat Sep  5 20:06:37 IDT 2015
Sat Sep  5 20:38:07 IDT 2015
Sat Sep  5 21:07:35 IDT 2015
Sat Sep  5 23:53:21 IDT 2015
Sat Sep  5 23:56:56 IDT 2015
Sun Sep  6 00:11:19 IDT 2015
Sun Sep  6 01:49:41 IDT 2015
Sun Sep  6 01:55:26 IDT 2015

NicolasRam, 2015-10-19:
Sorry to ask, but I have the same problem here and I would to know if you have found a fix for this recurrent problem ?
Thanks for any answer you can provide.

a-l-e-x-d-s-9, 2015-11-07:
The problem exist even on updated system with Jessie, v6.0.12, RaspiMJPEG Version 5.2.11.
When using motion detection internal mode.
``` log
{2015/11/07 10:11:37} RaspiMJPEG Version 5.2.11
{2015/11/07 10:11:37} recommended video buffers 1
{2015/11/07 10:11:37} h264 buffers set to recommended 1
{2015/11/07 10:11:37} Set up internal detect width=121 height=68
{2015/11/07 10:11:37} MJPEG streaming, ready to receive commands
{2015/11/07 10:11:37} send smd 9
[2015/11/07 10:11:37] Reload parameters command requested
[2015/11/07 10:11:37] Scheduler loop is started
[2015/11/07 10:11:38] Scheduled management tasks. Next at 1446885098
[2015/11/07 10:11:38] Purged 40 Files
[2015/11/07 10:31:39] Scheduled management tasks. Next at 1446886299
[2015/11/07 10:31:39] Purged 2 Files
{2015/11/07 10:50:24} Internal motion detection started
{2015/11/07 10:50:24} recommended video buffers 1
{2015/11/07 10:50:24} h264 buffers set to recommended 1
{2015/11/07 10:50:24} send smd 1
[2015/11/07 10:50:24] Start capture requested from Pipe
[2015/11/07 10:50:24] Send ca 1
{2015/11/07 10:50:24} Set up internal detect width=121 height=68
{2015/11/07 10:50:24} Capturing started
{2015/11/07 10:50:25} send smd 1
[2015/11/07 10:50:26] Start capture requested from Pipe
[2015/11/07 10:50:26] Send ca 1
{2015/11/07 10:50:26} Already capturing. Ignore 
[2015/11/07 10:51:40] Scheduled management tasks. Next at 1446887500
[2015/11/07 10:51:40] Purged 14 Files
[2015/11/07 11:11:41] Scheduled management tasks. Next at 1446888701
[2015/11/07 11:11:44] Purged 144 Files
[2015/11/07 11:31:42] Scheduled management tasks. Next at 1446889902
[2015/11/07 11:31:42] Purged 128 Files
[2015/11/07 11:51:43] Scheduled management tasks. Next at 1446891103
{2015/11/07 12:04:38} send smd 0
[2015/11/07 12:04:38] Stop capture requested
[2015/11/07 12:04:38] Send ca 0
{2015/11/07 12:04:39} Capturing stopped
{2015/11/07 12:04:39} Add /var/www/html/media/vi_0462_20151107_105024.mp4 to Box Queue at pos 0
{2015/11/07 12:04:39} Start boxing /var/www/html/media/vi_0462_20151107_105024.mp4.h264 to /var/www/html/media/vi_0462_20151107_105024.mp4 Queue pos 0
{2015/11/07 12:05:01} send smd 1
[2015/11/07 12:05:01] Start capture requested from Pipe
[2015/11/07 12:05:01] Send ca 1
{2015/11/07 12:05:02} Capturing started
{2015/11/07 12:05:06} send smd 0
[2015/11/07 12:05:06] Stop capture requested
[2015/11/07 12:05:06] Send ca 0
{2015/11/07 12:05:06} Capturing stopped
{2015/11/07 12:05:06} Add /var/www/html/media/vi_0463_20151107_120501.mp4 to Box Queue at pos 1
{2015/11/07 12:05:11} send smd 1
[2015/11/07 12:05:15] Start capture requested from Pipe
[2015/11/07 12:05:15] Send ca 1
{2015/11/07 12:05:15} Capturing started
{2015/11/07 12:06:33} send smd 0
[2015/11/07 12:06:33] Stop capture requested
[2015/11/07 12:06:33] Send ca 0
{2015/11/07 12:06:33} Capturing stopped
{2015/11/07 12:06:33} Add /var/www/html/media/vi_0464_20151107_120515.mp4 to Box Queue at pos 2
{2015/11/07 12:06:34} send smd 1
[2015/11/07 12:06:35] Start capture requested from Pipe
[2015/11/07 12:06:35] Send ca 1
{2015/11/07 12:06:39} Capturing started
{2015/11/07 12:06:42} send smd 0
[2015/11/07 12:06:42] Stop capture requested
[2015/11/07 12:06:42] Send ca 0
{2015/11/07 12:06:43} Capturing stopped
{2015/11/07 12:06:43} Add /var/www/html/media/vi_0465_20151107_120635.mp4 to Box Queue at pos 3
{2015/11/07 12:06:52} send smd 1
[2015/11/07 12:06:52] Start capture requested from Pipe
[2015/11/07 12:06:52] Send ca 1
[... 3215 lines cut ...]
{2015/11/07 15:18:47} Capturing stopped
{2015/11/07 15:18:47} Add /var/www/html/media/vi_0761_20151107_151844.mp4 to Box Queue at pos 6
{2015/11/07 15:18:48} Start boxing /var/www/html/media/vi_0761_20151107_151844.mp4.h264 to /var/www/html/media/vi_0761_20151107_151844.mp4 Queue pos 6
{2015/11/07 15:18:49} Finished boxing /var/www/html/media/vi_0761_20151107_151844.mp4 from Box Queue at pos 6
{2015/11/07 15:21:10} send smd 1
[2015/11/07 15:21:10] Start capture requested from Pipe
[2015/11/07 15:21:10] Send ca 1
{2015/11/07 15:21:10} Capturing started
{2015/11/07 15:21:15} send smd 0
[2015/11/07 15:21:15] Stop capture requested
[2015/11/07 15:21:15] Send ca 0
{2015/11/07 15:21:16} Capturing stopped
{2015/11/07 15:21:16} Add /var/www/html/media/vi_0762_20151107_152110.mp4 to Box Queue at pos 7
{2015/11/07 15:21:16} Start boxing /var/www/html/media/vi_0762_20151107_152110.mp4.h264 to /var/www/html/media/vi_0762_20151107_152110.mp4 Queue pos 7
{2015/11/07 15:21:17} Finished boxing /var/www/html/media/vi_0762_20151107_152110.mp4 from Box Queue at pos 7
{2015/11/07 15:21:46} send smd 1
[2015/11/07 15:21:46] Start capture requested from Pipe
[2015/11/07 15:21:46] Send ca 1
{2015/11/07 15:21:46} Capturing started
{2015/11/07 15:21:51} send smd 0
[2015/11/07 15:21:51] Stop capture requested
[2015/11/07 15:21:51] Send ca 0
{2015/11/07 15:21:51} Capturing stopped
{2015/11/07 15:21:51} Add /var/www/html/media/vi_0763_20151107_152146.mp4 to Box Queue at pos 8
{2015/11/07 15:21:51} Start boxing /var/www/html/media/vi_0763_20151107_152146.mp4.h264 to /var/www/html/media