silvanmelchior/RPi_Cam_Web_Interface #189: Button to save settings

danst0, 2015-08-21:
I cannot find a way to save the settings (my picture has to be rotated). How does that work?

Daniel

roberttidey, 2015-08-21:
If you make a change to camera settings from the web interface it is automatically saved to the uconfig file which is located in the web install folder.

The way this works is that raspimjpeg first loads the main default config from /etc/raspimjpeg. It then loads the user config file which can override the default settings. The user config file itself is a config variable in /etc/raspimjpeg but should be the uconfig file.

If the settings are not persisting then check to see if the uconfig file is being created and settings stored in there.

danst0, 2015-08-22:
Thanks for the quick reply. I cannot find any uconfig file. Neither in /var/www nor in my ~/RPi Cam Web Interface folder.

roberttidey, 2015-08-22:
That sounds like a permissions problem. Is /var/www where you have installed to (default). Have you  mounted it to a different volume? That should allow writing by user www-data which is what all the programs / web php runs under.

This normally is all set up during the install by default.

One thing you could try is to manually create a uconfig file there and chown (change owner) to www-data and make sure it has read / write permissions.

danst0, 2015-08-22:
I did not change anything for /var/www
These are the permissions:

drwxr-xr-x 7 www-data www-data 4096 Aug 21 19:27 /var/www/

I manually added a file called uconfig

-rw-r--r-- 1 www-data www-data 0 Aug 22 08:21 uconfig

Sadly no effect.

So I deleted /var/www, did a "RPi Cam Installer.sh install“. Picture is still showing, however options cannot be changed, uconfig does not exist.

danst0, 2015-08-22:
Update: I completely reinstalled Raspbian and RPi Cam now it works. Must have been something I did before.

Thanks for the help.