silvanmelchior/RPi_Cam_Web_Interface #192: RPi Cam web page showing "Loading" message continuously

Micchiu, 2015-09-05:
Hello.
Despite I followed 2 times the instructions in the page for installation, after a recent update (August) when I go to the RPi Cam web page on my Raspberrypi, it shows all the buttons, but not all working, and most of all it has the wording "Loading" in red all the times in place of the webcam images. I reinstalled all, updated my Raspbian, but nothing.
I searched the web for a solution, but I am unable to find.
Before that update, it was not perfect (motion detection never worked) but at least the camera image was perfect.
Could someone help me?
Thank you in advance.
[image]

a-l-e-x-d-s-9, 2015-09-06:
I noticed this behaviour sometimes when I use Chrome on Android, it stuck on "Loading", but if I connect from Chrome on my laptop, at same time, the video preview is working fine. If I use Opera on Android, it's working fine too. So it may be a compatibility issue in some edge case, or maybe just a problem with the browser itself.

Micchiu, 2015-09-06:
Thank you, but, as specified, the sofware used to work perfectly before the last update.

lindsaymacvean, 2015-09-16:
I have the same issue. Does the red light on your camera come on?

rrudolph, 2015-09-18:
Hi, I also have this same issue. Both firefox and chrome give the same problem.  Other controls in the interface like system reboot work fine, which is kind of weird because shouldn't that be a root privilege?? When I run ./RPi_Cam_Web_Interface_Installer.sh update it says RPi Cam is stopped.  I try to start it and at the bottom of the screen it flashes "This account is currently not available." and then it stays stopped.  Is it a permissions thing?  my whole /var/www/rcam folder is owned by pi:www-data.  Many thanks to anyone who can help, this is a great little tool and I hope I can get it working.

roberttidey, 2015-09-18:
System reboot works because the shutdown command has been added as available for www-data user to the /etc/sudoers.d/RPI_Cam_Web_interface file.

I haven't seen the This account is currently not available error before, but we have to get past that so that raspimjpeg starts up otherwise Loading is what you will get.

The start command in the script doen't do much so one can execute it one step at a time to see where it goes wrong.

       sudo mkdir -p /dev/shm/mjpeg
       sudo chown www-data:www-data /dev/shm/mjpeg
       sudo chmod 777 /dev/shm/mjpeg
       sleep 1;sudo su -c 'raspimjpeg > /dev/null &' www-data
       sleep 1;sudo su -c "php /var/www/$rpicamdir/schedule.php > /dev/null &" www-data

where you will need to change the $rpicamdir to the subfolder if you have not installed direct into /var/www

If it is one of the last 2 then it sounds like the www-data user is not fully operational for some reason which is a bit strange as it is what Apache uses.

rrudolph, 2015-09-18:
Hey thanks for the reply.  This command fails:
sleep 1;sudo su -c 'raspimjpeg > /dev/null &' www-data
by giving me this
This account is currently not available.
But now I know why, I turned off shell for www-data as a security measure.
Guess I'll have to turn it back on to /bin/bash if I want to use this software?  Or is there another way?
Thanks

rrudolph, 2015-09-18:
that was it. changed it back to /bin/sh. Thanks! Hope I don't get hacked.

roberttidey, 2015-09-18:
www-data has very few rights so the security is pretty good anyway. It can only do stuff under /var/www apart from the sudoers shutdown for reboot. The user macro scripts are kept there and you have to explicitly give them execution rights.

Plus one should be using Apache authentication passwords to give access to the web site in the first place.

Korri, 2015-10-04:
Hey, thanks for the issue, seemd to be the default on the last raspbian image, user www-data has the `/usr/sbin/nologin` after following the installation tutorial on a blank `RASPBIAN JESSIE` image. Might want to include a warning in the wiki.

jpreetham, 2015-10-04:
I have installed RASPBIAN JESSIE... and entry in /etc/passwd shows 
"www-data:x:33:33:www-data:/var/www:/bin/sh". and my RPi cam strating properly.

But still i am getting the "Apache2 Debian Default Page". Looks like RPi cam is started but not able to view it from browser. 
This has worked perfectly in wheezy.I am i missing anything?

jwcd3m0cr3w, 2018-01-27:
Can someone help me out plz - is anyone still around ?

samKolan, 2018-10-10:
check your ribbon cable for cracks at either end, They can be small.
and replace the cable if needed.

or you could try reinstalling.
First remove the whole installation

cd /home/pi/RPi_Cam_Web_Interface
sudo ./remove.sh

just to be sure delete the folder in /home/pi/RPi_Cam_Web_Interface, if its there.

Restart system

then in terminal type
sudo apt-get update && sudo apt-get dist-upgrade -y && sudo rpi-update

//Now re-install//

open web browser
goto https://github.com/silvanmelchior/RPi_Cam_Web_Interface
	click "Clone or download"
		"Download ZIP" // shouldn't take long.. This will be saved to your download folder.

OR Install from WEB.. More info here 
https://help.github.com/articles/which-remote-url-should-i-use/ 

in terminal

git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
sudo ./install.sh