silvanmelchior/RPi_Cam_Web_Interface #193: Purge space mechanism not working

a-l-e-x-d-s-9, 2015-09-06:
I use the system with motion detection mode, with purge space set to: 
   PurgeSpace_ModeEx = "Max Usage GB"
   PurgeSpace_Level = "160"
   Management_Interval = "1200"
Rest of the settings:
[image]
Currently my /var/www/media/ is already 300GB and keep growing, the purge mechanism seems to be unfunctional.
I looked at the code, from what I see deleteFile() uses filesize(), and from http://php.net/manual/en/function.filesize.php:
"Note: Because PHP's integer type is signed and many platforms use 32bit integers, some filesystem functions may return unexpected results for files which are larger than 2GB.".
I checked if I have files over 2GB, and indeed I have a few files that larger than 2GB. From my php test script, I get a negative number from filesize(), so I think that it's the source of the problem.
In the purgeFiles(): 
               foreach ($pFiles as $pFile) {
                  $del = ($level <= 0);
                  $level -= deleteFile($pFile, $del);
                  if ($del) $purgeCount++;
               }
* Actually I don't understand why it's not:  "$del = ($level > 0);"...

roberttidey, 2015-09-14:
I'll check out the filesize > 2GB issue which I can see would be problematic.

The delete logic is because the files are sorted latest to earliest so as it starts on the list the $del is false (no file delete, just return size). After the level has been reduced to the requested limit then the flag goes true and subsequent files get deleted. This means latest files are kept.

a-l-e-x-d-s-9, 2015-09-14:
Can't you use "shell_exec" with "du" on files that return negative values on filesize()?

roberttidey, 2015-09-14:
I am testing something similar using ls and awk. I am not sure about using just negative for this though as I suspect that files bigger than 4GB may be seen as positive again but with an incorrect filesize. So I am only handling the actual video files this way as these are the only ones that could become large.

roberttidey, 2015-09-15:
Using ls, stat or du slows things down quite a bit which can cause a delay of a few seconds. So I have put a config constant in config.php. As standard it will now cope with sizes up to 4GB. By changing the FILESIZE_METHOD to 1 it will handle sizes beyond that.

a-l-e-x-d-s-9, 2015-09-18:
OK, so I updated to latest version 6.0.8, edited /var/www/config.php to: "define('FILESIZE_METHOD', '1');",
rebooted the system.
After 20H, I don't see that it's working, currently /var/www/media size is ~300GB, it should purge to 160G:
[image]

Purge by hours seems to work perfectly fine.

Log:
{2015/09/17 23:27:04} RaspiMJPEG Version 5.2.9
{2015/09/17 23:27:05} recommended video buffers 1
{2015/09/17 23:27:05} h264 buffers set to recommended 1
{2015/09/17 23:27:05} MJPEG streaming, ready to receive commands
{2015/09/17 23:27:05} send smd 9
[2015/09/17 23:27:05] RaspiCam support started
[2015/09/17 23:27:06] Capture Pipe already exists /var/www/FIFO1
[2015/09/17 23:27:06] Scheduler loop is started
[2015/09/17 23:27:06] Reload parameters command requested
[2015/09/17 23:27:06] Scheduler loop is started
[2015/09/17 23:27:21] New period detected 0
[2015/09/17 23:27:21] Scheduled management tasks. Next at 1442522841
{2015/09/17 23:27:23} External motion detection started
{2015/09/17 23:27:23} recommended video buffers 1
{2015/09/17 23:27:23} h264 buffers set to recommended 1
[2015/09/17 23:27:24] Purged 31 Files
[2015/09/17 23:30:53] Stop capture request ignored, already stopped
[2015/09/17 23:32:59] Start capture requested from Pipe
[2015/09/17 23:32:59] Send ca 1
{2015/09/17 23:33:00} Capturing started
[2015/09/17 23:33:08] Stop capture requested
[2015/09/17 23:33:08] Send ca 0
{2015/09/17 23:33:08} Capturing stopped
{2015/09/17 23:33:08} Add /var/www/media/vi_43007_20150917_233259.mp4 to Box Queue at pos 0
{2015/09/17 23:33:08} Start boxing /var/www/media/vi_43007_20150917_233259.mp4.h264 to /var/www/media/vi_43007_20150917_233259.mp4 Queue pos 0
{2015/09/17 23:33:09} Finished boxing /var/www/media/vi_43007_20150917_233259.mp4 from Box Queue at pos 0
[2015/09/17 23:34:45] Start capture requested from Pipe
[2015/09/17 23:34:45] Send ca 1
{2015/09/17 23:34:45} Capturing started
[2015/09/17 23:34:50] Stop capture requested
[2015/09/17 23:34:50] Send ca 0
{2015/09/17 23:34:50} Capturing stopped
{2015/09/17 23:34:50} Add /var/www/media/vi_43008_20150917_233445.mp4 to Box Queue at pos 1
{2015/09/17 23:34:51} Start boxing /var/www/media/vi_43008_20150917_233445.mp4.h264 to /var/www/media/vi_43008_20150917_233445.mp4 Queue pos 1
{2015/09/17 23:34:52} Finished boxing /var/www/media/vi_43008_20150917_233445.mp4 from Box Queue at pos 1
[2015/09/17 23:35:03] Start capture requested from Pipe
[2015/09/17 23:35:03] Send ca 1
{2015/09/17 23:35:03} Capturing started
[2015/09/17 23:35:21] Stop capture requested
[2015/09/17 23:35:21] Send ca 0
{2015/09/17 23:35:21} Capturing stopped
{2015/09/17 23:35:21} Add /var/www/media/vi_43009_20150917_233503.mp4 to Box Queue at pos 2
{2015/09/17 23:35:21} Start boxing /var/www/media/vi_43009_20150917_233503.mp4.h264 to /var/www/media/vi_43009_20150917_233503.mp4 Queue pos 2
[2015/09/17 23:35:23] Start capture requested from Pipe
[2015/09/17 23:35:23] Send ca 1
{2015/09/17 23:35:24} Capturing started
{2015/09/17 23:35:24} Finished boxing /var/www/media/vi_43009_20150917_233503.mp4 from Box Queue at pos 2
[2015/09/17 23:35:25] Stop capture requested
[2015/09/17 23:35:25] Send ca 0
{2015/09/17 23:35:25} Capturing stopped
{2015/09/17 23:35:25} Add /var/www/media/vi_43010_20150917_233523.mp4 to Box Queue at pos 3
{2015/09/17 23:35:25} Start boxing /var/www/media/vi_43010_20150917_233523.mp4.h264 to /var/www/media/vi_43010_20150917_233523.mp4 Queue pos 3
{2015/09/17 23:35:26} Finished boxing /var/www/media/vi_43010_20150917_233523.mp4 from Box Queue at pos 3
[2015/09/17 23:35:27] Start capture requested from Pipe
[2015/09/17 23:35:27] Send ca 1
{2015/09/17 23:35:27} Capturing started
[2015/09/17 23:35:33] Stop capture requested
[2015/09/17 23:35:33] Send ca 0
{2015/09/17 23:35:33} Capturing stopped
{2015/09/17 23:35:33} Add /var/www/media/vi_43011_20150917_233527.mp4 to Box Queue at pos 4
{2015/09/17 23:35:33} Start boxing /var/www/media/vi_43011_20150917_233527.mp4.h264 to /var/www/media/vi_43011_20150917_233527.mp4 Queue pos 4
{2015/09/17 23:35:34} Finished boxing /var/www/media/vi_43011_20150917_233527.mp4 from Box Queue at pos 4
[... 2938 lines cut ...]
[2015/09/18 01:44:14] Start capture requested from Pipe
[2015/09/18 01:44:14] Send ca 1
{2015/09/18 01:44:14} Capturing started
[2015/09/18 01:44:19] Stop capture requested
[2015/09/18 01:44:19] Send ca 0
{2015/09/18 01:44:19} Capturing stopped
{2015/09/18 01:44:19} Add /var/www/media/vi_43334_20150918_014414.mp4 to Box Queue at pos 10
{2015/09/18 01:44:20} Start boxing /var/www/media/vi_43334_20150918_014414.mp4.h264 to /var/www/media/vi_43334_20150918_014414.mp4 Queue pos 10
{2015/09/18 01:44:21} Finished boxing /var/www/media/vi_43334_20150918_014414.mp4 from Box Queue at pos 10
[2015/09/18 01:44:21] Start capture requested from Pipe
[2015/09/18 01:44:21] Send ca 1
{2015/09/18 01:44:21} Capturing started
[2015/09/18 01:44:27] Stop capture requested
[2015/09/18 01:44:27] Send ca 0
{2015/09/18 01:44:27} Capturing stopped
{2015/09/18 01:44:27} Add /var/www/media/vi_43335_20150918_014421.mp4 to Box Queue at pos 11
{2015/09/18 01:44:27} Start boxing /var/www/media/vi_43335_20150918_014421.mp4.h264 to /var/www/media/vi_43335_20150918_014421.mp4 Queue pos 11
{2015/09/18 01:44:28} Finished boxing /var/www/media/vi_43335_20150918_014421.mp4 from Box Queue at pos 11
[2015/09/18 01:44:29] Start capture requested from Pipe
[2015/09/18 01:44:29] Send ca 1
{2015/09/18 01:44:29} Capturing started
[2015/09/18 01:44:31] Stop capture requested
[2015/09/18 01:44:31] Send ca 0
{2015/09/18 01:44:31} Capturing stopped
{2015/09/18 01:44:31} Add /var/www/media/vi_43336_20150918_014429.mp4 to

roberttidey, 2015-09-18:
I put in a bit of debug an did a test on my system with FILESIZE_METHOD = 1 and it worked for me but with much smaller volumes of data.

The debug I put in to schedule.php was around line 616 which then looked like
            case 4:
               $pFiles = getSortedFiles(false); //files in latest to earliest order
               writeLog(" Max space purge max: $level");
               foreach ($pFiles as $pFile) {
                  $del = ($level <= 0);
                  $level -= deleteFile($pFile, $del);
                  writeLog("lev $level file $pFile");
                  if ($del) $purgeCount++;
               }
               break;

The only difference is the first WriteLog is uncommented and a new one is added to report the level after each file is examined

I then got
[2015/09/18 18:46:32] Max space purge max: 629145.6
[2015/09/18 18:46:32] lev 615399.13515625 file vi_2238_20150918_171553.mp4.v2238.th.jpg
[2015/09/18 18:46:32] lev 599940.13027344 file vi_2237_20150918_171500.mp4.v2237.th.jpg
[2015/09/18 18:46:32] lev 587634.03847656 file vi_2236_20150918_160205.mp4.v2236.th.jpg
[2015/09/18 18:46:32] lev 571938.57070312 file vi_2235_20150918_144539.mp4.v2235.th.jpg
[2015/09/18 18:46:32] lev 560319.89101562 file vi_2234_20150918_123647.mp4.v2234.th.jpg
[2015/09/18 18:46:32] lev 548665.35292969 file vi_2233_20150918_111020.mp4.v2233.th.jpg
[2015/09/18 18:46:32] lev 534694.80800781 file vi_2232_20150917_151900.mp4.v2232.th.jpg
[2015/09/18 18:46:32] lev 522335.56191406 file vi_2231_20150917_141112.mp4.v2231.th.jpg
[2015/09/18 18:46:32] lev 508639.50136719 file vi_2230_20150917_141030.mp4.v2230.th.jpg
[2015/09/18 18:46:32] lev 492341.771875 file vi_2229_20150917_135839.mp4.v2229.th.jpg
[2015/09/18 18:46:32] lev 478083.30214844 file vi_2228_20150917_135828.mp4.v2228.th.jpg
[2015/09/18 18:46:32] lev 460609.74160156 file vi_2227_20150917_123330.mp4.v2227.th.jpg
[2015/09/18 18:46:32] lev 442001.68984375 file vi_2226_20150917_114958.mp4.v2226.th.jpg
[2015/09/18 18:46:32] lev 432914.64296875 file vi_2225_20150917_114509.mp4.v2225.th.jpg
[2015/09/18 18:46:32] lev 418251.31289062 file vi_2224_20150917_114458.mp4.v2224.th.jpg
[2015/09/18 18:46:32] lev 409446.25136719 file vi_2223_20150917_113552.mp4.v2223.th.jpg
[2015/09/18 18:46:33] lev 405753.80019531 file im_0000_20150917_113538.jpg.i0000.th.jpg
[2015/09/18 18:46:33] lev 392296.74941406 file vi_2222_20150917_111454.mp4.v2222.th.jpg
[2015/09/18 18:46:33] lev 379423.37246094 file vi_2221_20150917_111440.mp4.v2221.th.jpg
[2015/09/18 18:46:33] lev 357326.43886719 file vi_2220_20150917_102941.mp4.v2220.th.jpg
[2015/09/18 18:46:33] lev 346572.84121094 file vi_2219_20150917_084820.mp4.v2219.th.jpg
[2015/09/18 18:46:33] lev 335865.47011719 file vi_2218_20150917_083718.mp4.v2218.th.jpg
[2015/09/18 18:46:33] lev 324651.62441406 file vi_2217_20150917_083020.mp4.v2217.th.jpg
[2015/09/18 18:46:33] lev 311764.4359375 file vi_2216_20150917_082332.mp4.v2216.th.jpg
[2015/09/18 18:46:33] lev 295203.97597656 file vi_2215_20150917_081453.mp4.v2215.th.jpg
[2015/09/18 18:46:33] lev 284067.74355469 file vi_2214_20150917_081146.mp4.v2214.th.jpg
[2015/09/18 18:46:33] lev 272731.71132812 file vi_2213_20150917_081003.mp4.v2213.th.jpg
[2015/09/18 18:46:33] lev 262074.77285156 file vi_2212_20150917_080942.mp4.v2212.th.jpg
[2015/09/18 18:46:33] lev 255651.11269531 file vi_2211_20150917_080847.mp4.v2211.th.jpg
[2015/09/18 18:46:33] lev 244544.54921875 file vi_2210_20150917_080838.mp4.v2210.th.jpg
[2015/09/18 18:46:33] lev 233857.88417969 file vi_2209_20150917_080738.mp4.v2209.th.jpg
[2015/09/18 18:46:33] lev 223061.10585937 file vi_2208_20150917_080606.mp4.v2208.th.jpg
[2015/09/18 18:46:33] lev 212658.1 file vi_2207_20150917_080524.mp4.v2207.th.jpg
[2015/09/18 18:46:33] lev 200644.26894531 file vi_2206_20150917_080128.mp4.v2206.th.jpg
[2015/09/18 18:46:34] lev 190978.84414062 file vi_2205_20150917_075600.mp4.v2205.th.jpg
[2015/09/18 18:46:34] lev 178842.1703125 file vi_2204_20150917_075518.mp4.v2204.th.jpg
[2015/09/18 18:46:34] lev 168588.90273437 file vi_2203_20150917_075503.mp4.v2203.th.jpg
[2015/09/18 18:46:34] lev 158469.41738281 file vi_2202_20150917_075444.mp4.v2202.th.jpg
[2015/09/18 18:46:34] lev 147751.44863281 file vi_2201_20150917_075407.mp4.v2201.th.jpg
[2015/09/18 18:46:34] lev 136534.51503906 file vi_2200_20150916_174744.mp4.v2200.th.jpg
[2015/09/18 18:46:34] lev 126086.57949219 file vi_2199_20150916_172403.mp4.v2199.th.jpg
[2015/09/18 18:46:34] lev 115074.14785156 file vi_2198_20150916_162935.mp4.v2198.th.jpg
[2015/09/18 18:46:34] lev 104501.62148437 file vi_2197_20150916_161542.mp4.v2197.th.jpg
[2015/09/18 18:46:34] lev 90517.011132812 file vi_2196_20150916_161349.mp4.v2196.th.jpg
[2015/09/18 18:46:34] lev 77983.337304687 file vi_2195_20150916_160146.mp4.v2195.th.jpg
[2015/09/18 18:46:34] lev 66058.277734375 file vi_2194_20150916_145840.mp4.v2194.th.jpg
[2015/09/18 18:46:34] lev 54977.070703125 file vi_2193_20150916_144100.mp4.v2193.th.jpg
[2015/09/18 18:46:34] lev 39721.847070312 file vi_2192_20150916_142530.mp4.v2192.th.jpg
[2015/09/18 18:46:34] lev 27102.866601562 file vi_2191_20150916_124539.mp4.v2191.th.jpg
[2015/09/18 18:46:34] lev 12817.120507812 file vi_2190_20150915_180500.mp4.v2190.th.jpg
[2015/09/18 18:46:34] lev -3751.6265625 file vi_2189_20150915_180406.mp4.v2189.th.jpg
[2015/09/18 18:46:35] lev -16135.70859375 file vi_2188_20150915_155503.mp4.v2188.th.jpg
[2015/09/18 18:46:35] lev -31948.611914063 file vi_2187_20150915_151952.mp4.v2187.th.jpg
[... cut ...]
[2015/09/18 18:46:35] Purged 8 Files

Where you can see level was 0.629GB. The file sizes were subtracted, the last 8 had negative level and got purged.

As it is not working for you then 
Either the filing system is still not returning the correct sizes or there is some overflow somewhere due to large sizes. PHP normally switches to floating to avoid this.

If you can repeat the test then it will be easier to discover the cause. Note this generates a lot of log data so one doesn't want to leave this in.

a-l-e-x-d-s-9, 2015-09-18:
I wanted only to test purge based on folder size, so what I did:
1. Added these 2 prints to schedule.php.
2. Stopped motion detection. 
3. Turned off purged based on time.
4. Cleared the log.
5. Restarted the system, apparently it just used the old file without restarting.
I see the prints, but they all only related to video thumbnail, exactly as in your log. 
From what I see in the function getSortedFiles(), the condition "if(($file != '.') && ($file != '..') && isThumbnail($file))" allowing to add the file to list, only if it's a thumbnail. So that's why videos themselves not being deleted. Or maybe I misunderstood something?!

Log:
https://raw.githubusercontent.com/a-l-e-x-d-s-9/share/master/scheduleLog.txt_2015.09.19_00.21.46

roberttidey, 2015-09-18:
Everything here is based around thumbnail names which act as a key to find the actual data files. So the purge logic starts by only checking the thumbnails. When the log indicates a thumbnail it is getting back the total size consumed by the thumbnail and any associated data files. With lapse recording there may be many data files associated with 1 thumbnail. When it decides to purge then the delete function actually deletes the thumbnail and all associated data files as well.

From the logs it looks like the level check is 167772160. It then works its way through the list of files and gets down to 87257015 when all the files are checked indicating there is 80GB of file data altogether less than the 160GB purge limit About 5300 files.The running total never goes negative so no purging occurs.

There may be a misunderstanding here on the purge set up. There are two basic purge techniques based on disc usage as well as the maximum hourly use. The minimum space settings work to maintain a certain amount of free space on the disc (% of volume or Gigabyte). The maximum use settings work to maintain a maximum space occupied by the media files themselves, not a maximum use of all files on the disc. This is deliberate decision for maximum space so you can decide how much space to allow for media and the purge is not dependent on other files, code, data etc on the disc.

So say your disc had approximately 100GB of data already on it before any media is stored. With a 160GB limit it would need to get up to about 260GB of total data before the media purging started taking effect.

a-l-e-x-d-s-9, 2015-09-18:
I see what's the problem now. In the folder I have a few files with h264 extension, they are about 200G+, and sum up the total folder size to ~300G, but they don't have a thumbnail files, so they never would be purged.
Here are the biggest h264 files in the folder:
2.4G	/var/www/media/vi_11819_20150831_022321.mp4.h264
61G	/var/www/media/vi_13982_20150901_083650.mp4.h264
58G	/var/www/media/vi_15415_20150902_170824.mp4.h264
91G	/var/www/media/vi_43006_20150917_100837.mp4.h264

I assume they are temporary files used to capture the videos, and probably was never completed due to HW error.
The simplest solution is just to delete them, because they are slightly corrupted. If HW error problem will be solved sometime in the future, they wouldn't appear so much. Unless there was a sudden power failure, and maybe the video is important to the user, probably there is a way to recover most of the video somehow, so it's not a good idea just to delete it. I'm not sure...

a-l-e-x-d-s-9, 2015-10-12:
@roberttidey
Is there some way to convert h264 file into mp4 in error_hard.sh?

roberttidey, 2015-10-12:
h264 are converted to MP4 using the MP4Box tool. If configured to produce mp4 files then raspimjpeg just spawns a separate process to do this using a cmd like
      MP4Box -fps 25 -add h264filename mp4filename 

You can execute this type of command from a macro file but you need to know the in and out filenames. A suitable filename is passed in when using the job macros, but would not be available from the error_hard.sh macro as this can be triggered outside of job contexts. One could of course scan folders in the error_hard.sh looking for h264 files to convert.

a-l-e-x-d-s-9, 2015-10-12:
@roberttidey
Yeah, I'll scan the folder for h264 files as you suggested.
Thank you very much!

a-l-e-x-d-s-9, 2015-10-13:
OK, so I made the following python script for error_hard.sh. It converts h264 to mp4, and preventing too frequent reboots. 

```python
#!/usr/bin/env python
import sys
import os
import re
import time
from datetime import datetime
import string

COOLDOWN_REBOOTS_MINUTES = 5

def tail( f, n, offset=0 ):
    stdin,stdout = os.popen2( "tail -n " + str(n + offset) + " " + f )
    stdin.close()
    lines = stdout.readlines(); stdout.close()
    return lines[ offset : n ]

def main():
    MEDIA_FOLDER = "/var/www/media/"
    re_h264 = re.compile(r".*\.mp4\.h264")

    for root, dirs, filenames in os.walk(MEDIA_FOLDER):
        for FILE_NAME_SRC in filenames:
            if re_h264.match( FILE_NAME_SRC ):
                FILE_NAME_DEST = re.sub( r"(.*)\.mp4\.h264", r"\1.mp4", FILE_NAME_SRC )
                print( "MP4Box -fps 25 -add \"" + MEDIA_FOLDER + FILE_NAME_SRC + "\" \"" + MEDIA_FOLDER + FILE_NAME_DEST + "\"" )
                os.system( "MP4Box -fps 25 -add \"" + MEDIA_FOLDER + FILE_NAME_SRC + "\" \"" + MEDIA_FOLDER + FILE_NAME_DEST + "\"" )
                print( "os.remove( \"" + MEDIA_FOLDER + FILE_NAME_SRC + "\" )" )
                os.remove( MEDIA_FOLDER + FILE_NAME_SRC )

    current_time_stamp = time.strftime("%Y.%m.%d %H:%M:%S")
    # print( "current time: " + current_time_stamp )

    REBOOTS_LOG = "/var/www/reboots.log"
    reboot      = False

    if True == os.path.isfile(REBOOTS_LOG):
        
        last_reboot_string = tail( REBOOTS_LOG, 1 )[0]
        # print( "last_reboot_string: " + last_reboot_string )

        last_rebooted_time = datetime.strptime( last_reboot_string, "%Y.%m.%d %H:%M:%S\n" )
        current_reboot_time = datetime.strptime( current_time_stamp, "%Y.%m.%d %H:%M:%S" )
        # print(last_rebooted_time)
        # print(current_reboot_time)

        time_diff = current_reboot_time - last_rebooted_time

        seconds_in_one_day = 24 * 60 * 60
        time_diff_in_min_and_sec = divmod( time_diff.days * seconds_in_one_day + time_diff.seconds, 60 )
        
        reboot = time_diff_in_min_and_sec[0] > COOLDOWN_REBOOTS_MINUTES
        
    else:
        reboot = True

    if True == reboot:
        print( "echo \"" + current_time_stamp + "\" >> " + REBOOTS_LOG )
        os.system( "echo \"" + current_time_stamp + "\" >> " + REBOOTS_LOG )
        print( "sudo shutdown -r now" )
        os.system( "sudo shutdown -r now" )

main()

```

closed 2015-10-13 by a-l-e-x-d-s-9