silvanmelchior/RPi_Cam_Web_Interface #194: Time-lapse stop capturing pictures spontaneously

a-l-e-x-d-s-9, 2015-09-06:
I try to do a time-lapse for 24 hours, with 5 sec interval. It's working fine for a few hours, but currently it stop capturing pictures after 10+ hours by itself, and I don't know why. Obviously it's not a free space issue, I have 20GB+:
[image]

Purge not activated:
[image]

The web interface seem to be disabled, like this:
[image]
 
So I had to reboot the system, to use the web interface again.

The log doesn't say much. 
Test 1, log end (02:53-13:15: 10 hours, 21 minutes and 54 seconds):
{2015/09/04 13:15:35} Capturing image
{2015/09/04 13:15:40} Capturing image
[2015/09/04 13:46:58] Scheduled management tasks. Next at 1441367218
[2015/09/04 14:46:59] Scheduled management tasks. Next at 1441370819

Test 2, log end (19:13-08:22 : 13 hours, 8 minutes and 45 seconds):
{2015/09/06 08:21:45} Capturing image
{2015/09/06 08:21:50} Capturing image
{2015/09/06 08:21:55} Capturing image
{2015/09/06 08:22:00} Capturing image
[2015/09/06 09:10:35] Scheduled management tasks. Next at 1441523435
[2015/09/06 10:10:36] Scheduled management tasks. Next at 1441527036
[2015/09/06 11:10:37] Scheduled management tasks. Next at 1441530637
[2015/09/06 12:10:38] Scheduled management tasks. Next at 1441534238
[2015/09/06 13:10:39] Scheduled management tasks. Next at 1441537839

a-l-e-x-d-s-9, 2015-09-14:
By the way, by overclocking my Raspberry Pi 2, I was able to run the time lapse for ~21H, haven't tested longer periods.

bmedicke, 2015-09-14:
Same issue here with 3 Pies (2 model B and one model 2). I wrote a short shell function that sends me an alert when one of my Pies stops taking pictures. The model 2 allows for a shorter interval and as far as I can tell so do faster SD cards.

There's plenty of space left by the time the camera becomes unresponsive. It's easy to replicate by using a very short time-lapse interval on a slower (1 B) model.