silvanmelchior/RPi_Cam_Web_Interface #195: Motion detection and web UI stuck

a-l-e-x-d-s-9, 2015-09-11:
Motion detection suddenly stop working and web UI stuck:

[image]

Motion detection settings:

[image]

It's already happened twice on this RPi in the last 12 hours.

The end of the Log:

[2015/09/11 11:32:27] Start capture requested from Pipe
[2015/09/11 11:32:27] Send ca 1
{2015/09/11 11:32:27} Capturing started
[2015/09/11 11:32:31] Stop capture requested
[2015/09/11 11:32:31] Send ca 0
{2015/09/11 11:32:31} Capturing stopped
{2015/09/11 11:32:31} Add /var/www/media/vi_1350_20150911_113227.mp4 to Box Queue at pos 28
{2015/09/11 11:32:31} Start boxing /var/www/media/vi_1350_20150911_113227.mp4.h264 to /var/www/media/vi_1350_20150911_113227.mp4 Queue pos 28
{2015/09/11 11:32:32} Finished boxing /var/www/media/vi_1350_20150911_113227.mp4 from Box Queue at pos 28
[2015/09/11 11:32:38] Start capture requested from Pipe
[2015/09/11 11:32:38] Send ca 1
{2015/09/11 11:32:38} Capturing started
[2015/09/11 11:33:05] Stop capture requested
[2015/09/11 11:33:05] Send ca 0
{2015/09/11 11:33:05} Capturing stopped
{2015/09/11 11:33:05} Add /var/www/media/vi_1351_20150911_113238.mp4 to Box Queue at pos 29
{2015/09/11 11:33:06} Start boxing /var/www/media/vi_1351_20150911_113238.mp4.h264 to /var/www/media/vi_1351_20150911_113238.mp4 Queue pos 29
[2015/09/11 11:33:07] Start capture requested from Pipe
[2015/09/11 11:33:07] Send ca 1
{2015/09/11 11:33:16} Capturing started
{2015/09/11 11:33:24} Finished boxing /var/www/media/vi_1351_20150911_113238.mp4 from Box Queue at pos 29
[2015/09/11 11:33:56] Stop capture requested
[2015/09/11 11:33:56] Send ca 0

a-l-e-x-d-s-9, 2015-09-12:
Another log with the same problem:

[2015/09/11 16:02:06] Send ca 1
{2015/09/11 16:02:06} Capturing started
[2015/09/11 16:02:10] Stop capture requested
[2015/09/11 16:02:10] Send ca 0
{2015/09/11 16:02:10} Capturing stopped
{2015/09/11 16:02:10} Add /var/www/media/vi_1496_20150911_160206.mp4 to Box Queue at pos 11
{2015/09/11 16:02:11} Start boxing /var/www/media/vi_1496_20150911_160206.mp4.h264 to /var/www/media/vi_1496_20150911_160206.mp4 Queue pos 11
{2015/09/11 16:02:12} Finished boxing /var/www/media/vi_1496_20150911_160206.mp4 from Box Queue at pos 11
[2015/09/11 16:02:27] Start capture requested from Pipe
[2015/09/11 16:02:27] Send ca 1
{2015/09/11 16:02:28} Capturing started
[2015/09/11 16:02:31] Stop capture requested
[2015/09/11 16:02:31] Send ca 0
{2015/09/11 16:02:31} Capturing stopped
{2015/09/11 16:02:31} Add /var/www/media/vi_1497_20150911_160228.mp4 to Box Queue at pos 12
{2015/09/11 16:02:31} Start boxing /var/www/media/vi_1497_20150911_160228.mp4.h264 to /var/www/media/vi_1497_20150911_160228.mp4 Queue pos 12
{2015/09/11 16:02:32} Finished boxing /var/www/media/vi_1497_20150911_160228.mp4 from Box Queue at pos 12
[2015/09/11 16:02:49] Start capture requested from Pipe
[2015/09/11 16:02:49] Send ca 1
{2015/09/11 16:02:49} Capturing started
[2015/09/11 16:03:01] Stop capture requested
[2015/09/11 16:03:02] Send ca 0
{2015/09/11 16:03:02} Capturing stopped
{2015/09/11 16:03:02} Add /var/www/media/vi_1498_20150911_160249.mp4 to Box Queue at pos 13
{2015/09/11 16:03:03} Start boxing /var/www/media/vi_1498_20150911_160249.mp4.h264 to /var/www/media/vi_1498_20150911_160249.mp4 Queue pos 13
{2015/09/11 16:03:12} Finished boxing /var/www/media/vi_1498_20150911_160249.mp4 from Box Queue at pos 13
[2015/09/11 16:03:37] Start capture requested from Pipe
[2015/09/11 16:03:37] Send ca 1
{2015/09/11 16:03:37} Capturing started
[2015/09/11 16:03:54] Stop capture requested
[2015/09/11 16:03:54] Send ca 0
{2015/09/11 16:03:54} Capturing stopped
{2015/09/11 16:03:54} Add /var/www/media/vi_1499_20150911_160337.mp4 to Box Queue at pos 14
{2015/09/11 16:03:54} Start boxing /var/www/media/vi_1499_20150911_160337.mp4.h264 to /var/www/media/vi_1499_20150911_160337.mp4 Queue pos 14
{2015/09/11 16:03:58} Finished boxing /var/www/media/vi_1499_20150911_160337.mp4 from Box Queue at pos 14
[2015/09/11 16:04:22] Start capture requested from Pipe
[2015/09/11 16:04:22] Send ca 1
{2015/09/11 16:04:23} Capturing started
[2015/09/11 16:04:46] Stop capture requested
[2015/09/11 16:04:46] Send ca 0
{2015/09/11 16:04:46} Capturing stopped
{2015/09/11 16:04:46} Add /var/www/media/vi_1500_20150911_160423.mp4 to Box Queue at pos 15
{2015/09/11 16:04:47} Start boxing /var/www/media/vi_1500_20150911_160423.mp4.h264 to /var/www/media/vi_1500_20150911_160423.mp4 Queue pos 15
{2015/09/11 16:04:51} Finished boxing /var/www/media/vi_1500_20150911_160423.mp4 from Box Queue at pos 15
[2015/09/11 16:11:01] Start capture requested from Pipe
[2015/09/11 16:11:01] Send ca 1
{2015/09/11 16:11:01} Capturing started
[2015/09/11 16:11:08] Stop capture requested
[2015/09/11 16:11:08] Send ca 0
{2015/09/11 16:11:08} Capturing stopped
{2015/09/11 16:11:08} Add /var/www/media/vi_1501_20150911_161101.mp4 to Box Queue at pos 16
{2015/09/11 16:11:09} Start boxing /var/www/media/vi_1501_20150911_161101.mp4.h264 to /var/www/media/vi_1501_20150911_161101.mp4 Queue pos 16
{2015/09/11 16:11:10} Finished boxing /var/www/media/vi_1501_20150911_161101.mp4 from Box Queue at pos 16
[2015/09/11 16:11:22] Start capture requested from Pipe
[2015/09/11 16:11:22] Send ca 1
{2015/09/11 16:11:22} Capturing started
[2015/09/11 16:11:26] Stop capture requested
[2015/09/11 16:11:26] Send ca 0
{2015/09/11 16:11:26} Capturing stopped
{2015/09/11 16:11:26} Add /var/www/media/vi_1502_20150911_161122.mp4 to Box Queue at pos 17
{2015/09/11 16:11:26} Start boxing /var/www/media/vi_1502_20150911_161122.mp4.h264 to /var/www/media/vi_1502_20150911_161122.mp4 Queue pos 17
{2015/09/11 16:11:27} Finished boxing /var/www/media/vi_1502_20150911_161122.mp4 from Box Queue at pos 17
[2015/09/11 16:25:43] Scheduled management tasks. Next at 1441981543
[2015/09/11 16:28:45] Start capture requested from Pipe
[2015/09/11 16:28:45] Send ca 1
{2015/09/11 16:28:45} Capturing started
[2015/09/11 16:28:49] Stop capture requested
[2015/09/11 16:28:49] Send ca 0
[... 183 lines cut ...]
{2015/09/11 16:42:00} Finished boxing /var/www/media/vi_1523_20150911_164154.mp4 from Box Queue at pos 6
[2015/09/11 16:42:18] Start capture requested from Pipe
[2015/09/11 16:42:19] Send ca 1
{2015/09/11 16:42:19} Capturing started
[2015/09/11 16:44:13] Stop capture requested
[2015/09/11 16:44:13] Send ca 0
{2015/09/11 16:44:13} Capturing stopped
{2015/09/11 16:44:13} Add /var/www/media/vi_1524_20150911_164219.mp4 to Box Queue at pos 7
{2015/09/11 16:44:14} Start boxing /var/www/media/vi_1524_20150911_164219.mp4.h264 to /var/www/media/vi_1524_20150911_164219.mp4 Queue pos 7
[2015/09/11 16:44:15] Start capture requested from Pipe
[2015/09/11 16:44:15] Send ca 1
{2015/09/11 16:44:15} Capturing started
[2015/09/11 16:44:28] Stop capture requested
[2015/09/11 16:44:29] Send ca 0
{2015/09/11 16:44:29} Capturing stopped
{2015/09/11 16:44:29} Add /var/www/media/vi_1525_20150911_164415.mp4 to Box Queue at pos 8
[2015/09/11 16:44:32] Start capture requested from Pipe
[2015/09/11 16:44:32] Send ca 1
{2015/09/11 16:44:32} Capturing started
[2015/09/11 16:44:55] Stop capture requested
[2015/09/11 16:44:55] Send ca 0
[2015/09/11 17:25:44] Scheduled management tasks. Next at 1441985144
[2015/09/11 18:25:45] Scheduled management tasks. Next at 1441988745
[2015/09/11 18:27:08] New period detected 4
[2015/09/11 18:27:08] Send em night

Friendbg, 2015-10-08:
Hey,

I've got the same problem like @a-l-e-x-d-s-9 . I use raspberry pi 2 and the latest version of rpi cam web interface and my log is : 

{2015/10/08 06:55:24} Add /var/www/media/vi_0709_20151008_065521.mp4 to Box Queue at pos 15
[2015/10/08 06:55:25] Start capture requested from Pipe
[2015/10/08 06:55:25] Send ca 1
{2015/10/08 06:55:26} Capturing started
[2015/10/08 06:55:28] Stop capture requested
[2015/10/08 06:55:28] Send ca 0
[2015/10/08 06:55:30] Start capture requested from Pipe
[2015/10/08 06:55:30] Send ca 1
{2015/10/08 06:55:36} Capturing stopped
{2015/10/08 06:55:36} Add /var/www/media/vi_0710_20151008_065525.mp4 to Box Queue at pos 16
{2015/10/08 06:55:37} Capturing started
[2015/10/08 06:55:55] Stop capture requested
[2015/10/08 06:55:55] Send ca 0
[2015/10/08 07:14:49] Scheduled management tasks. Next at 1444281289
[2015/10/08 08:14:50] Scheduled management tasks. Next at 1444284890
[2015/10/08 09:14:51] Scheduled management tasks. Next at 1444288491

I saw that the video was freezing and the last timestamp was 06:55 and it says the same like @a-l-e-x-d-s-9  "recording video..."

I've rebooted the pi and video started again.

hasenbolle, 2015-11-23:
I have the same issue. I've had this issue on Jessie and now also on Wheezy. Looks like suddenly the "Send ca 0" for stop recording video is not accepted anymore from raspimjpeg. Maybe raspimpjeg hangs, so i cannot accept any new commands?

Here is my log:
[2015/11/23 19:52:13] Start capture requested from Pipe
[2015/11/23 19:52:13] Send ca 1
{2015/11/23 19:52:13} Capturing started
[2015/11/23 19:52:22] Stop capture requested
[2015/11/23 19:52:22] Send ca 0
{2015/11/23 19:52:22} Capturing stopped
{2015/11/23 19:52:22} Add /var/www/media/vi_0024_20151123_195213.mp4 to Box Queue at pos 28
{2015/11/23 19:52:22} Finished boxing /var/www/media/vi_0023_20151123_195204.mp4 from Box Queue at pos 27
{2015/11/23 19:52:22} Start boxing /var/www/media/vi_0024_20151123_195213.mp4.h264 to /var/www/media/vi_0024_20151123_195213.mp4 Queue pos 28
{2015/11/23 19:52:37} Finished boxing /var/www/media/vi_0024_20151123_195213.mp4 from Box Queue at pos 28
[2015/11/23 19:52:40] Start capture requested from Pipe
[2015/11/23 19:52:40] Send ca 1
{2015/11/23 19:52:40} Capturing started
[2015/11/23 19:52:44] Stop capture requested
[2015/11/23 19:52:44] Send ca 0
{2015/11/23 19:52:44} Capturing stopped
{2015/11/23 19:52:44} Add /var/www/media/vi_0025_20151123_195240.mp4 to Box Queue at pos 29
{2015/11/23 19:52:44} Start boxing /var/www/media/vi_0025_20151123_195240.mp4.h264 to /var/www/media/vi_0025_20151123_195240.mp4 Queue pos 29
[2015/11/23 19:52:46] Start capture requested from Pipe
[2015/11/23 19:52:46] Send ca 1
{2015/11/23 19:52:46} Capturing started
[2015/11/23 19:52:50] Stop capture requested
[2015/11/23 19:52:50] Send ca 0
{2015/11/23 19:52:50} Capturing stopped
{2015/11/23 19:52:50} Add /var/www/media/vi_0026_20151123_195246.mp4 to Box Queue at pos 30
[2015/11/23 19:52:53] Start capture requested from Pipe
[2015/11/23 19:52:54] Send ca 1
{2015/11/23 19:52:54} Capturing started
{2015/11/23 19:52:56} Finished boxing /var/www/media/vi_0025_20151123_195240.mp4 from Box Queue at pos 29
{2015/11/23 19:52:56} Start boxing /var/www/media/vi_0026_20151123_195246.mp4.h264 to /var/www/media/vi_0026_20151123_195246.mp4 Queue pos 30
[2015/11/23 19:53:09] Stop capture requested
[2015/11/23 19:53:09] Send ca 0

roberttidey, 2015-11-23:
Not sure if this is the same issue but I started compiling using gcc 4.9 which comes with Jessie and I was seeing the input from FIFO stop after a boxing command, I hadn't seen that with a wheezy gcc 4.6 compiled version but its possible some systems exhibited it because the source code was exactly the same.

I'm not sure what is causing that yet but I re-arranged the code slightly and the Jessie one started working OK. This is in the latest version. I am still continuing to check out what the root cause is because the boxing and input FIFO aren't connected in any way.

hasenbolle, 2015-11-24:
Thanks for the fast response. I've updated my installation on wheezy and now the buttons on the web interface doesn't work anymore. Also send commands to FIFO1 doesn't work anymore. 

Here is the shedule log: 
{2015/11/24 07:40:01} RaspiMJPEG Version 5.2.13
{2015/11/24 07:40:02} recommended video buffers 1
{2015/11/24 07:40:02} h264 buffers set to recommended 1
{2015/11/24 07:40:02} MJPEG streaming, ready to receive commands
{2015/11/24 07:40:02} send smd 9
[2015/11/24 07:40:07] RaspiCam support started
[2015/11/24 07:40:08] Capture Pipe already exists /var/www/FIFO1
[2015/11/24 07:40:08] Scheduler loop is started
[2015/11/24 07:40:08] Reload parameters command requested
[2015/11/24 07:40:08] Scheduler loop is started
[2015/11/24 07:40:09] New period detected 0
[2015/11/24 07:40:09] Scheduled management tasks. Next at 1448350809

roberttidey, 2015-11-24:
I have changed the version on the repository to being compiled under Wheezy. It's the same code but for some 'unknown' reason when compiled under Jessie it works under Jessie but not under Wheezy.

hasenbolle, 2015-11-24:
Now the buttons working again. Will test the code changes and report back, if the changes worked or not. Thanks for the fast update.

hasenbolle, 2015-11-30:
Ok after the past 6 days, everyday raspimjpeg stopped working :( Exactly the way like before the update. Any suggestions? Do you need some logs?

Egregius, 2016-02-04:
Interesting, good that I found this. Having the same issue with a RPi2 and Jessie.
Will try to invistigate when/where it goes wrong.