silvanmelchior/RPi_Cam_Web_Interface #197: Any way to add frames to teh Annotated Time Code?

melloajello, 2015-09-17:
As far as I can tell, there is no way to have frames on the annotated time code.

Is there any way to do this?

roberttidey, 2015-09-17:
There is no config at the moment to do this. The way the annotation works at the moment is that every time a preview image is generated then an MMAL call is made to the redefine the annotation text to be added. The text string used is calculated from the annotation template which includes date time variables. The app gets the linux which is to the nearest second and substitutes these before making the call.

Getting a higher resolution annotation might be possible but has a couple of complications. The first is that the preview rate can be divided down from the video frame rate so the annotation text is only updated at this lower rate. That is not so bad if a low divider or a divider of 1 is used. The second issue is deciding whether to just allow say millisecond timing or whether to count real frames. The latter could get complicated if trying to do say a 0-24 frame as it would not naturally be aligned to the  tick over of seconds.

So is the rquirement here just to get higher resolution or is there some purpose for frames?

melloajello, 2015-09-18:
Hello Robert,

Thanks for the quick reply!

I know it is definitely possible to do, I did it this way using picamera:

        HMS = dt.datetime.now().strftime('%H:%M:%S')
        FPS = int(round(float(dt.datetime.now().strftime("%f")[:-4])*.6,0))
        camera.annotate_text = '%s:%s' % (HMS, FPS)

I'm not sure if its possible to add this feature or how long it would take,
but I would be very appreciative, even if you could just point me to the
place in the PHP code where I could add this.

My other question is can you record at 640 X 480 at 90 fps?

Thank you for your time,
-Nateon

roberttidey, 2015-09-18:
If I understand that then you are adding formatting on decimal fractions of a second to the annotation rather than the actual frame number of the video. As I indicated that is much more do-able than a physical frame counter. I have started checking out supporting an additional annotation variable to represent milliseconds. Any higher resolution than this is pointless as a frame is already 40mSec.

This is handled by the raspimjpeg program which does all the low level camera interfacing work, not by the php programs which just pass through the data to the web interface.

Although the camera is capable of recording 640 x 480 at 90 fps that is when it is just doing one pure video recording and nothing else. In the case of raspimjpeg there are several parallel activities going on with the camera; generating the lower resolution mjpeg stream for the live preview, data for motion detection, and the video recording itself. This does lower the maximum rate somewhat. I believe 60 to 70 fps at that resolution is possible. I have also done 90 at lower resolutions.

roberttidey, 2015-09-19:
Latest version now has a %u naming variable which gives a 3 digit millisecond string

melloajello, 2015-09-22:
Thank you for the quick turn around on this Robert, that's great!

I hate to ask you for more, but is there any way that you could change the
base to 60 fps, 30 fps, or 24 fps like we use in film? That would be the
most useful.  Maybe you could say like %60u, %30u, %24u, etc.

Thanks again,
-Nateon

melloajello, 2015-09-25:
Hey Robert,

Hate to bug you again, but is there any chance you'll have the time to add
the functionality to have a 60 fps, 30 fps, or any other time base for the
annotation? It would also have to update on every frame.

This is the last step for me to be able to sync my video with to other time
coded applications, I think other people may find it useful as well.

Thanks for the consideration, and I understand if its too much or you are
too busy.  If you could let me know where the code is located, I may be
able to do it myself.

All the best,
-Nateon

melloajello, 2015-11-04:
Hello again Robert.

My name is Nateon, you exposed the miliseconds in the timestamp for me a
few months back, which I really appreciate.

I understand you are probably very busy and can't get to every persons
issue with the project given its popularity, but do you think you could at
least direct me to part of the code that deals with the timestamp so I can
work with it myself?

Thanks, and thanks for your help so far.

-Nateon

roberttidey, 2015-11-06:
The annotation is done in the raspimjpeg binary. The code for this is at https://github.com/roberttidey/userland  under /host_applications/linux/apps/raspicam

In the raspimcam.c module the cam_set_annotation function is called for each preview image. This in turn calls the makeName function in raspimutils.c which interprets the annotation string and substitutes the variables.

The time is the system time and so the tick over of a second is not directly related to any frame counter one adds so the 0 of 0-24 sequence does not naturally fall on second boundaries. One would need to use code that saw the second tick over and use that to reset a frame counter to 0.

roberttidey, 2015-11-21:
I have added a %k frame counter to annotation scheme

melloajello, 2015-11-21:
This is great Robert, thank you so much!

Is there a way to control whether it is 15, 24, 30, or 60 FPS?

roberttidey, 2015-11-21:
You set the video frame rate under camera settings. The frame counter %k automatically follows this. E.g. if fps is 25 then %k goes 0-24, if 30 it goes 0-29

melloajello, 2015-11-21:
I tried that and it doesn't seem to work for me. It will change to the
camera setting for one second cycle, then go back to 15 fps

roberttidey, 2015-11-21:
You can't change fps via camera settings?

You should be able to do this just by changing the number and pressing OK.

SchedulerLog should show this change and it should also appear in uconfig in the web folder. You might want to check that uconfig belongs to www-data and has read/write permissions.

I just tried a range of values from 5 - 30 and it worked Ok for me.

melloajello, 2015-11-21:
Ah, I see.   It works when you actually start recording, but not in the
preview.  Seems to work fine up to 60 FPS.

Thank you so much for this feature guys, I really really appreciate it.
When I finish my project I will credit you for sure and show you how much
it has helped me.

Is there somewhere I can donate to the project?

All the best,
-Nateon