silvanmelchior/RPi_Cam_Web_Interface #198: When motion detected, how to capture still images only?

pybokeh, 2015-09-20:
Hello, I made changes per http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=63276&start=175#p491013

Basically, in the motion.config file, I made the following change:
output_normal best

The target_dir is already set to the media folder: /var/www/rpicam/media

But the RPi cam still captures videos.  I don't want videos being record, just want still images captured.  Thanks.

roberttidey, 2015-09-20:
The motion detection is only used to provide triggers when motion has been detected. These triggers are processed by the scheduler set up to determine what to do. They may do any action or sequence of actions depending on what commands are associated with the triggers.

By default a motion start trigger is associated with a ca 1 command (start video) and a motion stop trigger is associated with a ca 0 (stop video). 

If you want a single still image when motion start is detected then use an im command instead of the ca 1. No command is necessary for the motion stop.

pybokeh, 2015-09-20:
Thanks, but where do I set the value that you speak of?

I looked at motion.conf and I see 2 possible locations:
# Command to be executed when an event starts. (default: none)
# An event starts at first motion detected after a period of no motion defined by gap
on_event_start echo -n '1' > /var/www/rpicam/FIFO1

or

# Command to be executed when a motion frame is detected (default: none)
; on_motion_detected value

In the first example, where would I insert "im" command?  In the 2nd example, do I just replace "value" with "im"?

Also confused about the location of the semi-colons.  Are the lines that begin with semi-colon, continuation of the previous line and therefore, continuation of the comment and therefore the line isn't "executed"?

Sorry for the noob questions, but I couldn't find anything in the documentation that would elaborate on the things I have questions on.

Thanks!

pybokeh, 2015-09-20:
@roberttidey Sorry, I figured it out.  I looked at the scheduler.php and added the im command there.  Thanks!

closed 2015-10-27 by roberttidey