silvanmelchior/RPi_Cam_Web_Interface #200: Raspberry Pi B , Debian Jessie problem

Bonzadog, 2015-10-02:
RPi-Cam-Web-Interface does not work with Jessie on a RPi B.
The index.php is not started but the index.hmtl is. Apache 2 seems to have different
directory structure.

I did hear via an Email ( now unfortunately deleted) that incompatibility with Raspimjpeg (?) 
could be a problem.
I would really like to see RPi-Cam-Web-Interface running on Debian Jessie
and hope that it will be updated soon.

Any chance?

roberttidey, 2015-10-04:
Issue 133 had some pointers on Jessie back in July. There may be some other issues since then but try those first.

roberttidey, 2015-10-05:
Forum message posted about this. Jessie doesn't work out of the box primarily due to Apache changes 

I made it work by setting subdir to html in install to match Apache and giving www-data user bash rights in /etc/passwd. It also needs libav-tools installed.

Installer will need a bit of work to support these changes.

a-l-e-x-d-s-9, 2015-10-05:
Isn't working on Jessie for me either, even with the changes you suggested. I installed it into "html" folder, installed libav-tools, changed /etc/passwd: 
    www-data:x:33:33:www-data:/var/www:/usr/sbin/nologin:/bin/bash 

If I use raspistill, it takes pictures, so camera is working. But the web UI only showing "Loading", and the red led of the camera turned off.