silvanmelchior/RPi_Cam_Web_Interface #201: Scheduler take care of reboot

mistyk, 2015-10-07:
Hello, 

Thanks for your works !
I'm currently using your software to build timelapse, i've scheduled a fixed hours to start and stop byt  i've noticed that if my raspberry reboot,  the timelapse does not restart , it not taking care of the Schedule on restart.
Do you think it's possible to check the Schedule on application start and launch the command if the time is passed ?

Best regards,

a-l-e-x-d-s-9, 2015-10-07:
I use external bash script on startup, to send configuration command to software fifo, you can configure most of the things by similar way, "tl" for time lapse, 1 - turn on, 0 - turn off :
echo -n 'tl 1' > /var/www/FIFO

More commands in: "Edit schedule settings" > "Command reference" (on the bottom of the page)

mistyk, 2015-10-07:
Yes i understand , but your command doesn't take care of the Schedule setting, i dont want to turn on timelapse on boot, but i would want that my raspberry turn on timelapse if it's set up in the Schedule (because the schedule configuration can be changed whenever).

(maybe create a bash script to parse the Schedule.json and compare with current clock and execute command if its panned)

You know ?

a-l-e-x-d-s-9, 2015-10-07:
I didn't understand your question before.
The behaviour that you describe is probably an edge case that hasn't been handled yet. 
If you know how to parse the Schedule.json and other stuff, you better just fix and push it to the source code to...

mistyk, 2015-10-08:
I don't know if it's the better solutoin, but if i success i will post my source code, of course!

mistyk, 2015-10-08:
I've saw this case could be manged in Schedule.php in the mainCLI function but if anyone can help me ;) i remark that when i launch in CLI "php5 Schedule.php" it seems to recheck Schedule as well  the first launch but after , the loop not seems to check every time the state of the Schedule and picam current state.  Anyone could help me ? 

Thanks for all

closed 2015-10-08 by mistyk