silvanmelchior/RPi_Cam_Web_Interface #204: Disable camera when timelapsing ?

mistyk, 2015-10-12:
Hello,

Is it possible to disable picam when is in timelapse mode ? (just power-on on picture) Because i need to use battery and when the timelapse mode is enabled picam is always running.. Have you any idea how to do that ? Because i really need to save power...)

Thanks so much.

roberttidey, 2015-10-12:
The camera is normally being used to provide a continuous preview all the time even during time lapse. I assume you are using a large interval between pictures and are not interested in the preview. It may be easier just to use raspistill with a script.

If you want to use the web interface as well to set things up then it may be possible to trigger a script from the web interface that turns off the camera and then does periodic raspistill commands.

Turning the camera off and on is possible using the ru cmd (0 off, 1 on)

mistyk, 2015-10-13:
Hello,
I understand that is not the first goal of your application, but for timelapse it could be very intersting to use your scheduler, to have the possibility to record video/image. More, i think  the preview could be an option because no need to have a preview for timelapse no ?

You think if i start timelapse and send CMD to turn off the camera, rpicam will continue run in timelapse mode ?

roberttidey, 2015-10-13:
Turning off camera will stop the timelapse. It totally removes the connection to the camera hardware.

It is not quite as simple as Preview On, Preview Off because to achieve what you want the camera would be needed to be turned on just before each time lapse exposure and then turned back off again. Raspimjpeg is also maintaining a video buffer which is in effect recording all the time so that data can be pre-captured prior to motion triggers.

It is not impossible to do, but needs quite a bit of thinking through how to achieve it.

mistyk, 2015-10-13:
Ok i understand, I would be glad that this function was implemented, i think lot of people use timelapse mode and this feature to save energy would be so interesting on small device like rpi (for battery usage and for consumption)

Thanks you, tell me know if you planned to implement that !

torstenhaase, 2016-03-25:
Hi Robert, are there any news on this function? I am using a battery powered pi and the 10000mAH battery pack just lasts for less then a day :(
Can you turn the preview on and off with the scheduler? The camera doesn't really work when it's dark so we could save some energy by turning it off on nighttime.
Thanks for your help
Torsten

torstenhaase, 2016-03-25:
Hi Robert, I just read the documentation for the scheduler and I think it can do exactly what I want. Can you quickly check my parameters? I created 4 periods with the following commands:

**6:59 ru 0** _that should start the camera right? no preview before 06:59 "power saving mode"_
**7:00 tl 0 tv 300** _that should start the time-lapse, 1 picture every 30s_
**18:00 tl 1** _that should stop the time-lapse_
**18:01 ru 1** _that should stop the camera, no preview after 18:01 "power saving mode"_

Thanks for your help
Torsten

roberttidey, 2016-03-25:
ru 1 starts camera ru 0 stops camera so you need to flip those.
tl 1 starts timelapse tl 0 stops time lapse so you need to flip those.
tv 300 will be remembered when set. So you could just set it once manually in camera settings.
If you put multiple commands in scheduler then put a ; in between and don't leave spaces after the ;

Like tv 300;tl 1; would set interval and then start tht etime lapse going

torstenhaase, 2016-03-26:
Hi Robert,
thanks for you help. I changed it and will try afterwards.
Maybe change the scheduler documentation. I not quite clear which number is stop and start

> tl	0/1	start/stop timelapse

roberttidey, 2016-03-26:
Scheduler command reference updated to be clearer on number usage.

torstenhaase, 2016-03-30:
Hi Robert, I have version 6.0.37 installed on apache with the old installation.sh.

Unfortunately the start/stop camera function doesn't work anymore. It automatically restarts the camera every time I try to switch it off. Or does that happen automatically when you open the webpage? In the last version there was a picture saying no connection.

Thanks for your help
Torsten

roberttidey, 2016-03-30:
I don't have much to do with old installation method which is maintained by miraaz. I maintain and use the install.sh method. So it is possible the installation is not as needed.

To check on the ru command, open a terminal window and with camera operating normally issue the command
echo -n 'ru 0' >/var/www/FIFO

include any subfolder used in the install.
The preview should freeze and all buttons dim out.

If you now issue a 
echo -n 'ru 1' >/var/www/FIFO

then camera should restart and carry on normal operation.